pgid69/bcm63xx-phone #2: Error with huawei hg556a

dotmark, 2015-06-07:
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.356000] bcm63xx-phone pcm_stop() 2362: RX DMA halted
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.360000] bcm63xx-phone pcm_stop() 2373: Disabling all PCM channels
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.368000] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.372000] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.380000] bcm63xx-phone pcm_dma_tx_descs_reset() 1373: pcm_dma_tx_descs_reset()
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.388000] bcm63xx-phone pcm_pr_stats() 2053: ======= Stats =======
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.396000] bcm63xx-phone pcm_pr_stats() 2054: rx_errors            = 0
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.400000] bcm63xx-phone pcm_pr_stats() 2055: rx_length_errors     = 0
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.408000] bcm63xx-phone pcm_pr_stats() 2056: rx_empty_errors      = 0
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.416000] bcm63xx-phone pcm_pr_stats() 2057: rx_good              = 0
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.424000] bcm63xx-phone pcm_pr_stats() 2058: rx_bytes             = 0
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.428000] bcm63xx-phone pcm_pr_stats() 2059: tx_errors            = 0
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.436000] bcm63xx-phone pcm_pr_stats() 2060: tx_good              = 0
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.444000] bcm63xx-phone pcm_pr_stats() 2061: tx_bytes             = 0
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.452000] bcm63xx-phone pcm_pr_stats() 2062: cnt_irq_rx           = 0
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.456000] bcm63xx-phone pcm_pr_stats() 2063: cnt_irq_tx           = 0
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.464000] bcm63xx-phone pcm_dma_deinit() 1798: pcm_dma_deinit()
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.472000] bcm63xx-phone pcm_dma_deinit() 1820: Unmapping DMA addresses of TX descs
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.480000] bcm63xx-phone pcm_dma_deinit() 1825: Unmapping DMA addresses of RX descs
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.488000] bcm63xx-phone pcm_dma_deinit() 1872: Freeing memory of RX and TX data buffers
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.496000] bcm63xx-phone pcm_dma_deinit() 1876: Freeing memory for DMA adresses
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.504000] bcm63xx-phone pcm_dma_deinit() 1881: Freeing DMA memory of RX descs
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.512000] bcm63xx-phone pcm_dma_deinit() 1890: Freeing DMA memory of TX descs
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.520000] bcm63xx-phone pcm_regs_deinit() 1259: pcm_regs_deinit()
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.524000] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.532000] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.540000] bcm63xx-phone pcm6358_pll_deinit() 1037: pcm6358_pll_deinit()
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.636000] bcm63xx-phone pcm_release_shared_res() 2001: pcm_release_shared_res()
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.644000] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1a00, size=32)
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.652000] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1900, size=32)
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.660000] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1800, size=256)
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.672000] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1600, size=256)
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.680000] bcm63xx-phone bcm_drv_init() 1590: Unregistering device
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.688000] bcm63xx-phone bcm_wait_queue_deinit() 39: bcm_wait_queue_deinit()
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.696000] bcm63xx-phone bcm_wait_queue_deinit() 39: bcm_wait_queue_deinit()
Sun Jun  7 20:30:24 2015 kern.info kernel: [  366.704000] bcm63xx-phone board_deinit() 622: board_deinit()
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6034.976000] bcm63xx-phone: bcm63xx-phone: v0.1
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6034.984000] bcm63xx-phone bcm_drv_init() 1439: bcm_drv_init()
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6034.988000] bcm63xx-phone board_init() 561: board_init()
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6034.996000] bcm63xx-phone: CPU is Broadcom 6358
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.000000] bcm63xx-phone: Board is HW556
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.004000] bcm63xx-phone board_get_desc() 615: board_get_desc()
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.008000] bcm63xx-phone bcm_drv_init() 1454: Initializing mutex
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.016000] bcm63xx-phone bcm_drv_init() 1457: Initializing RX wait queue
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.024000] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.032000] bcm63xx-phone bcm_drv_init() 1459: Initializing TX wait queue
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.036000] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.044000] bcm63xx-phone bcm_drv_init() 1474: Registering device bcm63xx-phone
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.052000] bcm63xx-phone pcm_init() 2482: pcm_init()
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.056000] bcm63xx-phone pcm_init() 2550: Using 25 buffers of 256 bytes for RX
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.064000] bcm63xx-phone pcm_init() 2586: Using 25 buffers of 256 bytes for TX
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.072000] bcm63xx-phone pcm_get_shared_res() 1950: pcm_get_shared_res()
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.080000] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm, addr=0xfffe1600, size=256)
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.088000] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dma, addr=0xfffe1800, size=256)
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.100000] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmac, addr=0xfffe1900, size=32)
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.108000] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmas, addr=0xfffe1a00, size=32)
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.208000] bcm63xx-phone pcm6358_pll_init() 970: pcm6358_pll_init()
Sun Jun  7 22:04:52 2015 kern.info kernel: [ 6035.212000] bcm63xx-phone pcm6358_pll_init() 975:  pll_ctrl1=f0010001, pll_ctrl2=53d47, pll_ctrl3=a85
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.256000] bcm63xx-phone pcm6358_pll_init() 1028:  pll_ctrl1=20001, pll_ctrl2=8492b, pll_ctrl3=1e1c
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.264000] bcm63xx-phone pcm_regs_init() 1247: pcm_regs_init()
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.268000] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.276000] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.284000] bcm63xx-phone pcm_dma_init() 1502: pcm_dma_init()
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.288000] bcm63xx-phone pcm_dma_init() 1535: Allocating 200 bytes of DMA memory for RX descs
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.300000] bcm63xx-phone pcm_dma_init() 1546: 200 bytes of memory allocated, starting at address 0xa2e77000
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.308000] bcm63xx-phone pcm_dma_init() 1552: Allocating 200 bytes of DMA memory for TX descs
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.316000] bcm63xx-phone pcm_dma_init() 1563: 200 bytes of memory allocated, starting at address 0xa2e76000
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.328000] bcm63xx-phone pcm_dma_init() 1571: Allocating 12800 bytes of memory for RX and TX data buffers
[... cut ...]
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.900000] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1600, size=256)
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.908000] bcm63xx-phone bcm_drv_init() 1590: Unregistering device
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.916000] bcm63xx-phone bcm_wait_queue_deinit() 39: bcm_wait_queue_deinit()
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.924000] bcm63xx-phone bcm_wait_queue_deinit() 39: bcm_wait_queue_deinit()
Sun Jun  7 22:04:53 2015 kern.info kernel: [ 6035.932000] bcm63xx-phone board_deinit() 622: board_deinit()

pgid69, 2015-06-07:
From the traces, it seems that MPI submodule failed to initialize properly : line 916 of mpi.c, 
call to platform_driver_register() fails without even calling bcm63xx_spi_probe().
Frankly, I don't understand why the kernel does not print any message.
I suggest you put more trace messages in function bcm_mpi_init() like printing value of ret after the call to platform_driver_register.

Some questions : 
- do you compile bcm63xx-phone driver with default options or do you change something ?
- does the problem occurs every time you try to load bcm63xx-phone driver ? 
- are you sure driver bcm63xx-spi is not loaded ?

dotmark, 2015-06-08:
Thanks for your quick response！

1.  I compile bcm63xx-phone driver with default options.
2. When the openwrt booted.  
#modprobe bcm63xx-spi
failed to find a module named bcm63xx-spi
but 
#1 logread|grep bcm63xx-spi
Sun Jun  7 20:33:36 2015 kern.info kernel: [    0.428000] bcm63xx-spi bcm63xx-spi: at 0xfffe0800 (irq 9, FIFOs size 542)

the full log is:

Sun Jun  7 20:33:36 2015 kern.info kernel: [    0.588000] 8021q: 802.1Q VLAN Support v1.8
Sun Jun  7 20:33:36 2015 kern.info kernel: [    0.612000] VFS: Mounted root (squashfs filesystem) readonly on device 31:2.
Sun Jun  7 20:33:36 2015 kern.info kernel: [    0.636000] Freeing unused kernel memory: 1300K (8035b000 - 804a0000)
Sun Jun  7 20:33:36 2015 user.err kernel: [    2.456000] init: failed to symlink /tmp -> /var
Sun Jun  7 20:33:36 2015 user.info kernel: [    2.460000] init: Console is alive
Sun Jun  7 20:33:36 2015 user.info kernel: [    2.464000] init: - watchdog -
Sun Jun  7 20:33:36 2015 kern.info kernel: [    5.828000] usbcore: registered new interface driver usbfs
Sun Jun  7 20:33:36 2015 kern.info kernel: [    5.832000] usbcore: registered new interface driver hub
Sun Jun  7 20:33:36 2015 kern.info kernel: [    5.840000] usbcore: registered new device driver usb
Sun Jun  7 20:33:36 2015 kern.notice kernel: [    5.940000] SCSI subsystem initialized
Sun Jun  7 20:33:36 2015 kern.info kernel: [    5.960000] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Sun Jun  7 20:33:36 2015 kern.info kernel: [    5.972000] ehci-platform: EHCI generic platform driver
Sun Jun  7 20:33:36 2015 kern.info kernel: [    5.980000] ehci-platform ehci-platform: EHCI Host Controller
Sun Jun  7 20:33:36 2015 kern.info kernel: [    5.984000] ehci-platform ehci-platform: new USB bus registered, assigned bus number 1
Sun Jun  7 20:33:36 2015 kern.info kernel: [    5.992000] ehci-platform ehci-platform: irq 18, io mem 0xfffe1300
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.012000] ehci-platform ehci-platform: USB 2.0 started, EHCI 1.00, overcurrent ignored
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.020000] hub 1-0:1.0: USB hub found
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.024000] hub 1-0:1.0: 2 ports detected
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.044000] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.052000] ohci-platform: OHCI generic platform driver
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.056000] ohci-platform ohci-platform: Generic Platform OHCI controller
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.064000] ohci-platform ohci-platform: new USB bus registered, assigned bus number 2
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.072000] ohci-platform ohci-platform: irq 13, io mem 0xfffe1400
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.140000] hub 2-0:1.0: USB hub found
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.144000] hub 2-0:1.0: 2 ports detected
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.172000] usbcore: registered new interface driver usb-storage
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.592000] usb 2-2: new full-speed USB device number 2 using ohci-platform
Sun Jun  7 20:33:36 2015 user.info kernel: [    6.796000] init: - preinit -
Sun Jun  7 20:33:36 2015 kern.info kernel: [    6.804000] usb 2-2: not running at top speed; connect to a high speed hub
Sun Jun  7 20:33:36 2015 kern.info kernel: [    7.028000] hub 2-2:1.0: USB hub found
Sun Jun  7 20:33:36 2015 kern.info kernel: [    7.204000] hub 2-2:1.0: 4 ports detected
Sun Jun  7 20:33:36 2015 kern.notice kernel: [    7.968000] random: procd urandom read with 31 bits of entropy available
Sun Jun  7 20:33:36 2015 kern.info kernel: [    9.788000] eth0: link UP - 100/full - flow control off
Sun Jun  7 20:33:36 2015 user.info kernel: [   11.672000] mount_root: loading kmods from internal overlay
Sun Jun  7 20:33:36 2015 kern.notice kernel: [   12.112000] jffs2: notice: (286) jffs2_build_xattr_subsystem: complete building xattr subsystem, 1 of xdatum (0 unchecked, 0 orphan) and 1 of xref (0 dead, 0 orphan) found.
Sun Jun  7 20:33:36 2015 user.info kernel: [   12.128000] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
Sun Jun  7 20:33:36 2015 user.info kernel: [   12.140000] block: extroot: not configured
Sun Jun  7 20:33:36 2015 kern.notice kernel: [   12.192000] jffs2: notice: (283) jffs2_build_xattr_subsystem: complete building xattr subsystem, 1 of xdatum (0 unchecked, 0 orphan) and 1 of xref (0 dead, 0 orphan) found.
Sun Jun  7 20:33:36 2015 user.info kernel: [   12.208000] mount_root: loading kmods from internal overlay
Sun Jun  7 20:33:36 2015 user.info kernel: [   12.408000] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
Sun Jun  7 20:33:36 2015 user.info kernel: [   12.420000] block: extroot: not configured
Sun Jun  7 20:33:36 2015 user.info kernel: [   12.428000] mount_root: switching to jffs2 overlay
Sun Jun  7 20:33:36 2015 user.info kernel: [   12.500000] procd: - early -
Sun Jun  7 20:33:36 2015 user.info kernel: [   12.504000] procd: - watchdog -
Sun Jun  7 20:33:36 2015 kern.info kernel: [   12.932000] eth0: link DOWN
Sun Jun  7 20:33:36 2015 user.info kernel: [   13.620000] procd: - ubus -
Sun Jun  7 20:33:36 2015 user.info kernel: [   14.648000] procd: - init -
Sun Jun  7 20:33:36 2015 kern.info kernel: [   18.024000] NET: Registered protocol family 10
[... cut ...]
Sun Jun  7 20:49:57 2015 daemon.warn odhcpd[891]: DHCPV6 REQUEST IA_NA from 000100011b18e0c294de80bc7056 on br-lan: ok fd55:c38e:b22f::bcc/128 
Sun Jun  7 20:49:57 2015 daemon.info dnsmasq[1359]: read /etc/hosts - 1 addresses
Sun Jun  7 20:49:57 2015 daemon.info dnsmasq[1359]: read /tmp/hosts/odhcpd - 1 addresses
Sun Jun  7 20:49:57 2015 daemon.info dnsmasq[1359]: read /tmp/hosts/dhcp - 1 addresses
Sun Jun  7 20:49:57 2015 daemon.info dnsmasq-dhcp[1359]: read /etc/ethers - 0 addresses
Sun Jun  7 20:53:57 2015 daemon.info hostapd: wlan0: STA e0:f8:47:2d:28:08 WPA: group key handshake completed (RSN)

pgid69, 2015-06-08:
Ok, there's a conflict with bcm63xx-spi driver.
I make my tests with OpenWrt 12.09 where you can choose the way driver bcm63xx-spi is included in the kernel.
With OpenWrt 14.07 you have no choice : bcm63xx-spi is always included in the kernel.
So if it's possible, i will try to add an option to bcm63xx-phone, to tell it to unregister bcm63xx-spi driver.

pgid69, 2015-06-08:
I wrote a small patch to resolve your problem. I will test it this evening if i can.
If you want, here's the content of the patch.
If you want you can apply it, recompile bcm63xx-phone driver and load it with parameter unregister_bcm63xx_spi=y

```
From bc2e9d483c018b012d1cbf98f1ae20a5164772a2 Mon Sep 17 00:00:00 2001
From: Gilles Mazoyer <mazoyer.gilles@omega.ovh>
Date: Mon, 8 Jun 2015 13:11:44 +0200
Subject: [PATCH] Ajout d'un parametre pour retirer le driver bcm63xx-spi

---
 bcm63xx-phone/src/main.c    | 14 ++++++++++++++
 bcm63xx-phone/src/mpi/mpi.c | 20 +++++++++++++++++++-
 2 files changed, 33 insertions(+), 1 deletion(-)

diff --git a/bcm63xx-phone/src/main.c b/bcm63xx-phone/src/main.c
index 3a70b49..1a0c54e 100644
--- a/bcm63xx-phone/src/main.c
+++ b/bcm63xx-phone/src/main.c
@@ -68,6 +68,20 @@ module_param_named(channel_mapping, bcm_drv_param_channel_mapping, uint, 0);
 #endif // __KERNEL__
 #endif // !BCMPH_TEST_PCM
 
+#ifndef BCMPH_USE_SPI_DRIVER
+#ifdef __KERNEL__
+// This param is used to unregister driver named bcm63xx-spi that
+// conflicts with our own manager of SPI bus.
+// Beware that this operation can't be reversed.
+// It's a (quite dirty) workaround for the problem where bcm63xx-spi
+// driver is linked into the kernel (default with OpenWrt 14.07),
+// because in that case 'rmmod bcm63xx-spi' is not possible.
+bool bcm_drv_param_unregister_bcm63xx_spi = false;
+
+module_param_named(unregister_bcm63xx_spi, bcm_drv_param_unregister_bcm63xx_spi, bool, 0);
+#endif // __KERNEL__
+#endif // !BCMPH_USE_SPI_DRIVER
+
 // Callback called periodically by timer of phone_mgr
 static void bcm_drv_timer_cb(bcm_phone_mgr_t *phone_mgr)
 {
diff --git a/bcm63xx-phone/src/mpi/mpi.c b/bcm63xx-phone/src/mpi/mpi.c
index 1d39dd0..f1c5a6b 100644
--- a/bcm63xx-phone/src/mpi/mpi.c
+++ b/bcm63xx-phone/src/mpi/mpi.c
@@ -846,6 +846,10 @@ int bcm_mpi_read_write(bcm_mpi_t *t, __u8 *buf, __u8 buf_len)
    return (ret);
 }
 
+#ifndef BCMPH_USE_SPI_DRIVER
+extern bool bcm_drv_param_unregister_bcm63xx_spi;
+#endif // !BCMPH_USE_SPI_DRIVER
+
 int __init bcm_mpi_init(bcm_mpi_t *t, const bcm_mpi_params_t *params)
 {
    int ret = 0;
@@ -913,7 +917,21 @@ fail_master:
    t->wait_completion_with_irq = params->wait_completion_with_irq;
 
    if (bcm_mpi_dev_data.ref_count <= 0) {
-      ret = platform_driver_register(&(bcm63xx_spi_driver));
+      struct device_driver *spi_driver = driver_find(bcm63xx_spi_driver.driver.name, &platform_bus_type);
+      if (NULL != spi_driver) {
+         if (bcm_drv_param_unregister_bcm63xx_spi) {
+            driver_unregister(spi_driver);
+            spi_driver = NULL;
+         }
+         else {
+            bcm_pr_err("Error: Driver '%s' is already registered, aborting...\n",
+               bcm63xx_spi_driver.driver.name);
+            ret = -EBUSY;
+         }
+      }
+      if (NULL != spi_driver) {
+         ret = platform_driver_register(&(bcm63xx_spi_driver));
+      }
       bcm_assert(((ret) && (0 == bcm_mpi_dev_data.ref_count))
          || ((!ret) && (1 == bcm_mpi_dev_data.ref_count)));
    }
-- 
2.1.4
```

dotmark, 2015-06-09:
I used "make kernel_menuconfig " and deselect the bcm63xx_spi. 
now "bcm63xx-phone" succeeded load.

pgid69, 2015-06-09:
Thank you for your feedback.
When you have time, please let me know the results of your tests with Asterisk. I'd like to know if Asterisk channel and bcm63xx-phone works correctly on HW556.

pgid69, 2015-06-13:
I finally test the patch above and it does not work : unregistering the driver spi-bcm63xx causes a kernel oops, so i won't include this patch (moreover it was dirty solution).
I will just include a more explicit message that bcm63xx-spi is already registered.

closed 2015-06-13 by pgid69