pgid69/bcm63xx-phone #3: How to dial to  FXS lines

dotmark, 2015-06-09:
When i dial "100" from a SIP phone,  i want FXS line1 ring.
How to config extensions.conf ?

[bcmph-line-1] 
exten => 100,1,Dial(?)

I`m not familiar with the asterisk. Thank you.

pgid69, 2015-06-09:
I'm not familiar with Asterisk either.
But for your problem, section [bcmph-line-1] is for handling calls initiated from the FXS line.
In your case you should have a section dedicated to your SIP phone with the following line
exten => 100,1,Dial(Bcm63xxPhone/1)

pgid69, 2015-06-09:
One remarks : you can launch the asterisk console to watch what's happen when you dial 100 from your SIP phone : asterisk -rvvvvvv

dotmark, 2015-06-09:
Succeed！

pgid69, 2015-06-13:
I will investigate the problem of phone 1 that can't be enabled

pgid69, 2015-06-14:
On the HW553, FXS port "Phone 1" is line 2 and "Phone 2" is line 1. So i change file board.c to swap id of lines of HW553.
Else i can enable the two lines in Asterisk.

closed 2015-09-30 by pgid69