pgid69/bcm63xx-phone #4: suggestions

stesc, 2015-09-27:
I tested on a HG553 (Vodafone Station). It works very well but I think that 2 updates could be done and I could do it for community if you can give me some suggestions.

1) timeout before to send phone call to asterisk should be different compared ro the timeout of the 'first digit'. Infact usually when phone is took off-hook timeout should be longer compared to timeout between one digit and the next one
2) Also could be nice to speed-up the phone call, if is possible to add a '#' (for example) at the end of the  typed number so that phone call is imemeditaely transmitted to asterisk.

I tried to check source code where I could be these updates but I am not really sure.

pgid69, 2015-09-27:
Your suggestions are welcome.
I will try to check source code in the next days, to see what has to be modified.

pgid69, 2015-09-30:
Hello Stefano,

I made a branch that have modifications i made for your first suggestion.
It compiles, but i have not time now to test it.

I still have to think about your second suggestion, if i add a new parameter, if that parameter can be a sequence of characters or just a single character, if when searching for an extension, i must include the character(s), and if not what to do if no extension is found.
The code that needs to be modified is in function bcmph_handle_digits() at line 1588.

Gilles

stesc, 2015-09-30:
Hello,

first of allthank you for your fast response. Now I tried your new version and about 1st digit timeout it appears it is all OK, really good job.

About second request these are ansqers to your questions:

1) I think 1 charachter is enough. It is like so many ip phone (for example grandstream) that let user type '#' to speed-up passing call to asterisk.
2) When searching for the extensions, this characther must be removed
3) If extension is not found, asterisk will return some messages about it

NOw I modified your source code in the following manner (thank you for your suggestion). At line 1588:

            if ( ('#' == digits[i-1]) || (pvt->line_cfg->dialing_timeout <= 0)) {
               if ('#' == digits[i-1])
               {
                 pvt->ast_channel.digits_len--;
               }
               pvt->ast_channel.search_extension = true;
               bcmph_search_extension(pvt, monitor_prms, true);
               bcm_assert(!pvt->ast_channel.search_extension);
               if ((BCMPH_ST_OFF_DIALING != pvt->ast_channel.state) || (!monitor_prms->channel_is_locked)) {
                  break;
               }
            }

the correct one is digits[i-1] because some line before you already have added 1 to i. Then

pvt->ast_channel.digits_len--;

to remove the '#'.

With this modifications now works exactly as I wanted, that is, when I type '#', immediately asterisk try to connect to the extension I typed, even if timeout is larger.

Now I would like to have just one information why there is such OR condition:

pvt->line_cfg->dialing_timeout <= 0

when dialing_timeout is 0? I searched code but I didn't find the meaning of this.

Also I have still this effect:

1) When I dial the external extension (I use a Huawey HSDPA stick with a mobile SIM), I cannot hear the ring tone when the called extension is ringing, when other party answers I can correctly hear voice.
Asterisk, when call is passed to asterisk, I have so many mesages:

Only wrote 0 of 320 bytes of audio data to line 1

do you know why?

Again thank you, I wait for your next reply.

pgid69, 2015-10-02:
Hello

Thank you for the feedback.
I'm glad that you adapt the code to fit your needs.
If you allow me, i will integrate your modification in the code.
I agree with you that one digit is enough.
I will add a parameter to let the user choose that digit (if you have any idea for the name of the parameter, please let me know).
If Asterisk don't find an extension. Asterisk won't return any message. In this case the function bcmph_search_extension() returns without changing the state of the channel. Even in that case i think the digit '#' have to be discarded.

The OR condition (pvt->line_cfg->dialing_timeout <= 0) is to handle the case where the user set the parameter dialing_timeout  to 0, to search for an extension immediately after a digit is dialed.

For the ring tone i think that's because Asterisk don't call bcmph_chan_indicate() with parameter AST_CONTROL_RINGING. I don't know why.

For the messages, "Only wrote 0 of 320 bytes of audio data to line 1", i don't know why too. For me it means that Asterisk feeds data too fast, but not sure. Do you have theses messages during the whole call or only at the start ? 

For the parameters i agree with you, there's a need for default values.

And finally about Asterisk 1.8, the code won't compile. There's API changes between Asterisk 1.8 and Asterisk 11 (and between Asterisk 11 and Asterisk 13), for example for the function that handle formats. Why do you want to use Asterisk 1.8 instead of Asterisk 11 ? Does Asterisk 1.8 needs less memory ?
 
Thank you for the time you spent to test.

stesc, 2015-10-05:
Sorry for late reply. Here are answers and firther notes.
Yesm you can intagrate it with no problem to your code. About parameter I thought '#' because is the digit Grandstream phones (I have one of this) use. About the need to manage the asterisk reply. I thought that when your chan have the number (either extension exists and extensions doesn't exist), your chan ends, you are saying that you could manage some reply from Asterisk? When cahn passes number to asterisk have still not ended his work? Maybe I don't know something about it then I cannot understand this.
About RINGING tone let me do some more test.
Errors "Only wrote 0 of 320 bytes of audio data to line 1" are really many but they are just during ringing. I thought that these errors were the reason with I don't ehaer ring tone.

About Atserisk 1.8, no there is no reason why I should use it, I just wanted to do that your chan will be more compatible with all releases

pgid69, 2015-10-10:
Hello, just to mention i add a parameter 'search_extension_trigger'.
It's character and it's default value is 0 (meaning there's no special digit used to search for a valid extension immediately).
In your case you have to set it to '#'.
When you type that character, the channel calls immediately function bcmph_search_extension() but the character is not included in the list of digits dialed, used to search for an extension.
In bcmph_search_extension() the channel asks Asterisk if the digits form a valid extension.
If answer is yes, the channel stops waiting for new digits and starts the call.
If answer is no, the channel asks Asterisk if the digits form the start of an extension.
If answer is yes, we add the new digit and continue until dialing_timeout ellapses or a new digit is dialed.
If answer is no, we change the state of the channel to BCMPH_ST_OFF_NO_SERVICE.

Else, i think i found the problem causing the messages "Only wrote 0 of 320 bytes of audio data to line 1". I make the correction and commit the code when it's ok.

If i have time i will try to make the channel compile with Asterisk 1.8

stesc, 2015-10-19:
Hello, sorry for late reply. I just now had time for reply. Sorry again.

Thank you for your explanation, I didn't know that chan can 'ask' asterisk if extension is valid and then decide what to do. Now it is clearer. I think the behaviour you supposed is correct.

I think that the reason why I cannot hear the ringback tone when I dial chan_dongle is not a problem of your channel but it is a problem of chan_dongle. The warkaround is to have a 'r' parameter intoDial macro into the asterisk dialplan, even if this ringbak is not real.

I also did these trys:

1) I did a call from external (PSTN) to Vodafone Station to mobile SIM number that is into dongle.
As soon as analogue phone connected to port 1 FXS start to ring I have 1 error:

WARNING[6263][C-00000001]: channel.c:1002 channel_indicate: [Dongle/dongle0-0100000001] Don't know how to indicate condition 22

Now phone start to ring and no other error appears
As soon I pick up phone (off-hook) I have following errors:

WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 256 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 128 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 0 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 0 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 128 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 128 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 0 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 0 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 128 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 192 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 128 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 192 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[5308][C-00000000]: asterisk:2917 bcmph_chan_write: Only wrote 128 of 320 bytes of audio data to line 1
[... cut ...]
2) I call from analogue phone connected to port 1 FXS using the dongle. As soon I type '#' call is passed to Asterisk and it starts to compare these errors:

WARNING[7392][C-00000004]: asterisk:2917 bcmph_chan_write: Only wrote 0 of 320 bytes of audio data to line 1

They are about 40-50 errors of these. Then pause about 1 second and then other 40-50 errors, pause and so on, until I answer call, then no more errors. During these errors I should hear the ring tone into the analogue phone but I don't hear anything.

I look forward for other tests.

pgid69, 2015-10-20:
Hello, don't apologize for late reply.
I also miss time to test the modifications i made and that i don't commit yet in github.

The modifications are for being able to compile with Asterisk 1.8, Asterisk 11 and Asterisk 13.
And, at least i hope so, they resolve the issue you had with your second test.
The problem i discovered is that the channel can receive data before the call is answered. That data was queued but the driver was not in the correct mode and so did not consume that data. So the queue filled up. I think you're right and that data is for ring tone : the pause of about 1 second between series of messages confirms that hypothesis.

I suggest you to wait that i commit my modifications before making others tests.
But i don't want to commit before making my own tests. And as i already mention i miss time. So please be patient.

Thanks for your help

pgid69, 2015-10-24:
Hello

I just push modifications i made.
I test them, but unfortunately, i never get the messages you had 'bcmph_chan_write: Only wrote xxx of 320 bytes of audio data to line 1'.
I hope the modifications fix at least problem in the test #2 you made.

stesc, 2015-10-24:
Hello, thank you for your effort to update yourbreat chan. Here are the test repeated with the new version.

1) I did a call from external (PSTN) to Vodafone Station to mobile SIM number that is into dongle. As soon as analogue phone connected to port 1 FXS start to ring I have following:

    -- Executing [+1234567890@gsm-external:1] Dial("Dongle/dongle0-0100000003", "Bcm63xxPhone/1,30,t") in new stack
    -- Called Bcm63xxPhone/1
WARNING[3233][C-00000004]: channel.c:1002 channel_indicate: [Dongle/dongle0-0100000003] Don't know how to indicate condition 22
    -- Bcm63xxPhone/1 is ringing

It is normal except the 'condition 22' that I don't know what it is. Now I hear the phone ringing, then I pick up the phone and I soon have following:

    -- Bcm63xxPhone/1 answered Dongle/dongle0-0100000003
.....................
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 256 of 320 bytes of audio data to line 1
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 128 of 320 bytes of audio data to line 1
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 128 of 320 bytes of audio data to line 1
WARNING[3233][C-00000004]: asterisk:3509 bcmph_chan_write: Only wrote 64 of 320 bytes of audio data to line 1
.................................

I pasted here just a part of the errors, but these are about 70. After these errors I can hear other party and all is ok. When I hangup (I hangup using the phone connected to Vodafone Station) all is correct.

2) This time it appears your patch did a step forward but maybe something else should be patched. Here is my test: I call from analogue phone connected to port 1 FXS using the dongle. As soon I type '#' call is passed to Asterisk and Asterisk confirms that he is calling. After some seconds destination phone starts o ring and I hear the ringback in the analogue phone. The ringback tone is crackly. After some seconds I see following errors:

...........
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 256 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 64 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 0 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 128 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 64 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 64 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 64 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 64 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 128 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 64 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 0 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 128 of 320 bytes of audio data to line 1
WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 64 of 320 bytes of audio data to line 1
...........

I pasted here just a part of the errors, but these are about 239. After these errors, ringback tone is now clear, no more crackly.

I think that now is much better because this time I can finally hear the ringback tone during ring of the destination phone. There is just this strange thing I described.

I hope these new tests can by of some help for you.

pgid69, 2015-10-28:
Hello

Now, i don't understand what's wrong.

Do you have some messages in the system log from the module bcm63xx-phone.ko ?

I'm still unable to reproduce your problem as i can't test in the same condition as you.
For my tests i'm using a SIP account.
Whereas, from what i understand, you are using a SIM card inserted in a dongle plugged in the modem and managed by chan-dongle.
I also have a dongle (Sierra AirCard 312U) but from what i read it is not compatible with chan-dongle. Can you confirm me that ?

The other possibility is that the CPU of the modem is not powerfull enough and either chan-dongle or chan-bcm63xx-phone is eating too much CPU cycles at start of call.

What version of OpenWrt are you running (14.07 or 15.05) ?

In bcm63xx_phone.conf, are the two lines enabled or just one ?

Perhaps if you have time you can try to enable just one line, to detect dtmf only when dialing and to use only codec slin.
Another test would be to compile bcm63xx_phone.ko and chan-bcm63xx-phone without flag BCMPH_DEBUG to disable traces. 

Thanks

Gilles

stesc, 2015-11-05:
Hello,

sorry again for delay. In 'dmesg' I have many messages from bcm63xx-phone.ko but they appear to be normal debug messages (is there a way to disable them?).
Yes, I am using a SIM into a dongle using chan_dongle. To check if it is compatible, just plug-in into USB and check into 'dmesg' if ttyUSBx devices are created. If you can see them try to configure /etc/asterisk/dongle.conf,
dongle.conf, just configure [dongle0] section, the audio and data values (they should be ttyUSB0 and ttyUSB1. Restart asterisk ancheck with 'dongle show devices' (be sure module  is chan_dongle.so loaded by asterisk). If you can see your dongle details then you can configure extensions using, for example, Dial(Dongle/g0/${EXTEN},30,T) where g0 is the group number defined into dongle.conf.
I am using 12.09. I enabled both fxs lines.

In the next days I will follow your suggestions about trace debug and codec.
Thank for your support.
Stefano.

pgid69, 2015-11-08:
Hi Stefano

I tried asterisk-chan-dongle with the Sierra dongle and unfortunately,i have a message saying my dongle does not support voice. I also tried with a ZTE K3565-Z dongle, and it also does not support voice. )-:
So unfortunately, i can't reproduce your configuration.
Any idea is welcome.

Gilles

stesc, 2015-12-15:
> Data: 8-nov-2015 16.08
>
>
>
> Ogg: Re: [bcm63xx-phone] suggestions (#4)

Hi. I saw you closed and merged the updates. I am glad you followed my suggestions and applied them. I will try again it as soon as possible and I'll back to you reporting my tests.
Thank you.

----Messaggio originale----

pgid69, 2016-05-15:
Hello

I think i finally fix the issue.
If you have time can you make some more tests.
Thanks

stesc, 2016-05-16:
Hello. Now I will compile again and test soon. Thank you for your effort.

pgid69, 2016-06-01:
Hello 

Any (good) news ?

pgid69, 2016-06-17:
No news, good news.
I close the issue.

stesc, 2016-07-05:
Hello, sorry for the very late reply. I tested new source code but all 
appears it is like before. Exactly, what you expect to be fixed?

thank you

stesc, 2016-07-05:
I am talking about your last release, obviously :)

stesc, 2016-07-05:
I forgot to say that I tested both incoming call (GSM->DONGLE->FXS) and 
outgoung call (FXS->DONGLE->GSM)

stesc, 2016-07-05:
Also, thank you

pgid69, 2016-07-10:
Hello stesc,

I expected to fix the problem of the message 
"WARNING[4705][C-00000000]: asterisk:3509 bcmph_chan_write: OnlyWrote 256 of 320 bytes of audio data to line 1" you have when remote phone is ringing.
Are you sure you are testing the source code of my repository and not the one of your repository ?
Is the problem is still the same as the one you reported initially, with the same messages in asterisk console ?

Thank you for your help.

Gilles

pgid69, 2016-07-17:
It would help me if you also send me asterisk config : dialplan and any relevant config files you can think of.

closed 2016-10-25 by pgid69