pgid69/bcm63xx-phone #6: Compiled but not working

EniGMist4, 2016-05-03:
I compiled but in dmesg at boot I have:

**[    0.000000] bcm63xx_nvram: nvram checksum failed, contents may be invalid (expected 33303336, got fc05ed45)
[    0.380000] bcm63xx_uart.0: ttyS0 at MMIO 0xfffe0100 (irq = 10, base_baud = 1562500) is a bcm63xx_uart
[    0.432000] bcm63xxpart: CFE boot tag found with version 7 and board type HW553
[    0.440000] 5 bcm63xxpart partitions found on MTD device 1e000000.nor
[    0.512000] bcm63xx-spi bcm63xx-spi: at 0xfffe0800 (irq 9, FIFOs size 542)
[    0.552000] libphy: bcm63xx_enet MII bus: probed
[    0.632000] bcm63xx_enet bcm63xx_enet.1: attached PHY at address 0 [Broadcom B53 (1)]
[    0.640000] bcm63xx-wdt bcm63xx-wdt:  started, timer margin: 30 sec**

insmod bcm63xx-phone.ko gives me: _failed to insert bcm63xx-phone.ko_

with this log:
**[  356.672000] bcm63xx-phone: bcm63xx-phone: v0.1
[  356.676000] bcm63xx-phone bcm_drv_init() 1439: bcm_drv_init()
[  356.680000] bcm63xx-phone board_init() 706: board_init()
[  356.688000] bcm63xx-phone: CPU is Broadcom 6358
[  356.692000] bcm63xx-phone: Board is HW553
[  356.696000] bcm63xx-phone board_get_desc() 764: board_get_desc()
[  356.704000] bcm63xx-phone bcm_drv_init() 1454: Initializing mutex
[  356.708000] bcm63xx-phone bcm_drv_init() 1457: Initializing RX wait queue
[  356.716000] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
[  356.724000] bcm63xx-phone bcm_drv_init() 1459: Initializing TX wait queue
[  356.728000] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
[  356.736000] bcm63xx-phone bcm_drv_init() 1474: Registering device bcm63xx-phone
[  356.744000] bcm63xx-phone pcm_init() 2482: pcm_init()
[  356.752000] bcm63xx-phone pcm_init() 2550: Using 25 buffers of 256 bytes for RX
[  356.756000] bcm63xx-phone pcm_init() 2586: Using 25 buffers of 256 bytes for TX
[  356.764000] bcm63xx-phone pcm_get_shared_res() 1950: pcm_get_shared_res()
[  356.772000] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm, addr=0xfffe1600, size=256)
[  356.780000] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dma, addr=0xfffe1800, size=256)
[  356.792000] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmac, addr=0xfffe1900, size=32)
[  356.800000] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmas, addr=0xfffe1a00, size=32)
[  356.900000] bcm63xx-phone pcm6358_pll_init() 970: pcm6358_pll_init()
[  356.904000] bcm63xx-phone pcm6358_pll_init() 975:  pll_ctrl1=f0010001, pll_ctrl2=53d47, pll_ctrl3=a85
[  356.948000] bcm63xx-phone pcm6358_pll_init() 1028:  pll_ctrl1=20001, pll_ctrl2=8492b, pll_ctrl3=1e1c
[  356.956000] bcm63xx-phone pcm_regs_init() 1247: pcm_regs_init()
[  356.960000] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
[  356.968000] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
[  356.976000] bcm63xx-phone pcm_dma_init() 1502: pcm_dma_init()
[  356.980000] bcm63xx-phone pcm_dma_init() 1535: Allocating 200 bytes of DMA memory for RX descs
[  356.992000] bcm63xx-phone pcm_dma_init() 1546: 200 bytes of memory allocated, starting at address 0xa2477000
[  357.000000] bcm63xx-phone pcm_dma_init() 1552: Allocating 200 bytes of DMA memory for TX descs
[  357.008000] bcm63xx-phone pcm_dma_init() 1563: 200 bytes of memory allocated, starting at address 0xa2476000
[  357.020000] bcm63xx-phone pcm_dma_init() 1571: Allocating 12800 bytes of memory for RX and TX data buffers
[  357.028000] bcm63xx-phone pcm_dma_init() 1578: 12800 bytes of memory allocated, starting at address 0x8247c000
[  357.040000] bcm63xx-phone pcm_dma_init() 1604: Allocating 224 bytes of memory for DMA addresses
[  357.048000] bcm63xx-phone pcm_dma_init() 1611: 224 bytes of memory allocated, starting at address 0x83364a00
[  357.060000] bcm63xx-phone pcm_dma_init() 1618: Mapping DMA addresses of RX descs
[  357.068000] bcm63xx-phone pcm_dma_init() 1635: Mapping DMA addresses of TX descs
[  357.076000] bcm63xx-phone pcm_dma_rx_descs_reset() 1342: pcm_dma_rx_descs_reset()
[  357.084000] bcm63xx-phone pcm_dma_tx_descs_reset() 1373: pcm_dma_tx_descs_reset()
[  357.092000] bcm63xx-phone bcm_drv_init() 1524: t->mm_desc.ioctl_param_off = 0, t->mm_desc.ioctl_param_size = 208
[  357.100000] bcm63xx-phone bcm_drv_init() 1534: t->mm_desc.rbs[0].rx_ring_buf_desc_off = 208, t->mm_desc.rbs[0].rx_ring_buf_desc_size = 12
[  357.112000] bcm63xx-phone bcm_drv_init() 1542: t->mm_desc.rbs[0].tx_ring_buf_desc_off = 224, t->mm_desc.rbs[0].tx_ring_buf_desc_size = 12
[  357.128000] bcm63xx-phone bcm_drv_init() 1534: t->mm_desc.rbs[1].rx_ring_buf_desc_off = 240, t->mm_desc.rbs[1].rx_ring_buf_desc_size = 12
[  357.140000] bcm63xx-phone bcm_drv_init() 1542: t->mm_desc.rbs[1].tx_ring_buf_desc_off = 256, t->mm_desc.rbs[1].tx_ring_buf_desc_size = 12
[  357.152000] bcm63xx-phone bcm_drv_init() 1564: 8192 bytes of memory allocated, starting at address 0x82468000
[  357.164000] bcm63xx-phone bcm_phone_mgr_init() 850: bcm_phone_mgr_init()
[  357.168000] bcm63xx-phone bcm_phone_mgr_devs_init() 724: bcm_phone_mgr_devs_init()
[  357.176000] bcm63xx-phone phone_device_alloc() 55: phone_device_alloc()
[  357.184000] bcm63xx-phone phone_device_alloc() 66: 2020 bytes of memory allocated, starting at address 0x8337e800
[  357.196000] bcm63xx-phone phone_dev_le88221_init() 54: phone_dev_le88221_init()
[  357.204000] bcm63xx-phone phone_dev_zarlink_ve880_init() 1948: phone_dev_zarlink_ve880_init()
[  357.212000] bcm63xx-phone bcm_mpi_init() 862: bcm_mpi_init()
[  357.216000] bcm63xx-phone: Error: Driver 'bcm63xx-spi' is already registered, aborting...
[  357.224000] bcm63xx-phone bcm_drv_init() 1584: Freeing mmap buffer used for RX, TX...
[  357.232000] bcm63xx-phone pcm_deinit() 2684: pcm_deinit()
[  357.240000] bcm63xx-phone pcm_stop() 2321: pcm_stop()
[  357.244000] bcm63xx-phone pcm_dma_stop() 1470: pcm_dma_stop()
[  357.252000] bcm63xx-phone pcm_stop() 2333: Halting TX DMA
[  357.264000] bcm63xx-phone pcm_stop() 2340: TX DMA halted
[  357.268000] bcm63xx-phone pcm_stop() 2355: Halting RX DMA
[  357.280000] bcm63xx-phone pcm_stop() 2362: RX DMA halted
[  357.284000] bcm63xx-phone pcm_stop() 2373: Disabling all PCM channels
[  357.292000] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
[  357.296000] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
[  357.304000] bcm63xx-phone pcm_dma_tx_descs_reset() 1373: pcm_dma_tx_descs_reset()
[  357.312000] bcm63xx-phone pcm_pr_stats() 2053: ======= Stats =======
[  357.320000] bcm63xx-phone pcm_pr_stats() 2054: rx_errors            = 0
[  357.324000] bcm63xx-phone pcm_pr_stats() 2055: rx_length_errors     = 0
[  357.332000] bcm63xx-phone pcm_pr_stats() 2056: rx_empty_errors      = 0
[  357.340000] bcm63xx-phone pcm_pr_stats() 2057: rx_good              = 0
[  357.344000] bcm63xx-phone pcm_pr_stats() 2058: rx_bytes             = 0
[  357.352000] bcm63xx-phone pcm_pr_stats() 2059: tx_errors            = 0
[  357.360000] bcm63xx-phone pcm_pr_stats() 2060: tx_good              = 0
[  357.368000] bcm63xx-phone pcm_pr_stats() 2061: tx_bytes             = 0
[  357.372000] bcm63xx-phone pcm_pr_stats() 2062: cnt_irq_rx           = 0
[  357.380000] bcm63xx-phone pcm_pr_stats() 2063: cnt_irq_tx           = 0
[  357.388000] bcm63xx-phone pcm_dma_deinit() 1798: pcm_dma_deinit()
[  357.392000] bcm63xx-phone pcm_dma_deinit() 1820: Unmapping DMA addresses of TX descs
[  357.400000] bcm63xx-phone pcm_dma_deinit() 1825: Unmapping DMA addresses of RX descs
[  357.408000] bcm63xx-phone pcm_dma_deinit() 1872: Freeing memory of RX and TX data buffers
[  357.416000] bcm63xx-phone pcm_dma_deinit() 1876: Freeing memory for DMA adresses
[  357.424000] bcm63xx-phone pcm_dma_deinit() 1881: Freeing DMA memory of RX descs
[  357.432000] bcm63xx-phone pcm_dma_deinit() 1890: Freeing DMA memory of TX descs
[  357.440000] bcm63xx-phone pcm_regs_deinit() 1259: pcm_regs_deinit()
[  357.448000] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
[  357.452000] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
[  357.460000] bcm63xx-phone pcm6358_pll_deinit() 1037: pcm6358_pll_deinit()
[  357.556000] bcm63xx-phone pcm_release_shared_res() 2001: pcm_release_shared_res()
[  357.560000] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1a00, size=32)
[  357.572000] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1900, size=32)
[  357.580000] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1800, size=256)
[  357.592000] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1600, size=256)
[  357.600000] bcm63xx-phone bcm_drv_init() 1590: Unregistering device
[  357.608000] bcm63xx-phone bcm_wait_queue_deinit() 39: bcm_wait_queue_deinit()
[  357.612000] bcm63xx-phone bcm_wait_queue_deinit() 39: bcm_wait_queue_deinit()
[  357.620000] bcm63xx-phone board_deinit() 771: board_deinit()**

Kernel is 3.18.29 and Openwrt 15.05. What's wrong?

pgid69, 2016-05-03:
Hi

The problem is that the driver bcm63xx-spi is already included in the kernel. And so you get the message :
[ 357.216000] bcm63xx-phone: Error: Driver 'bcm63xx-spi' is already registered, aborting...
You have to compile the Linux kernel without including bcm63xx-spi.
I recently updated the README to explain what to do

EniGMist4, 2016-05-04:
Yeah It worked! Simple method:

_make kernel_menuconfig CONFIG_TARGET=subtarget_

and removed the driver. Thanks it work very well

closed 2016-05-04 by pgid69