pgid69/bcm63xx-phone #7: Regional tones

Leoz83, 2016-08-24:
Hi, i've try to set
`country = italy`
on bcm63xx_phone.conf

I've already set
`country=it`
on indications.conf, but when i pick up the phone  the configuration not work.
HW556 - asterisk11

Suggestions?
Thanks for all your work!

pgid69, 2016-08-25:
Hi,
The parameter country has no effect on dial tones.
It only has effects on voltages... It's used by the Voice Path API Lite SDK to select AC, DC profiles.
The dial tones are hard coded and do not depend upon the parameter country.
Is it really important ?

BR

Leoz83, 2016-08-25:
Nothing of important, just for habit

pgid69, 2017-12-27:
Hello

I finally add an option in bcm63xx-phone.conf to choose the zone, allowing to have tone and ring cadence specific to your country as defined in file indications.conf.
If option zone is not set it is derived from option country if possible.

BR

closed 2017-12-27 by pgid69