pgid69/bcm63xx-phone #8: bcm63xx-ast-chan not installing

flafyrex, 2016-10-30:
Hello! I'm flafyrex(also in OpenWRT Forums)  and I tried to compile the bcm63xx-phone. The kernel module and phone-test got installed, but ast-chan doesn't want to install(doesn't show installing...). I'm compiling on Debian Jessie 8. Please help me quickly. Thank you.

pgid69, 2016-10-30:
Hello,
First there's no need for package phone-test : as it name suggests it is for test only.
For your problem, there's no package named bcm63xx-ast-chan : there are 3 packages, asterisk18-chan-bcm63xx-phone for Asterisk 1.8, asterisk11-chan-bcm63xx-phone for Asterisk 11 and asterisk13-chan-bcm63xx-phone for Asterisk 13.
Of course to compile one of these packages, you must first select one version of Asterisk.
Give me more details about your problem (the commands you typed...) because i can't understand from your messages if you have a problem of compilation or installation on your router.
Thanks.

flafyrex, 2016-10-30:
root@flafyrex:~/openwrt# ./scripts/feeds install bcm63xx-ast-chan
root@flafyrex:~/openwrt# 

Seriously,that is what is happening. I want to test,that's why I also installed the phone test. Also I wroted to you on the forum,but i haven't got an answer from you. That error gets in randomly. Now I copied makefile from asterisk13 folder(chan-asterisk13-bcm63xx-phone). And now I get this. This is how I got so far.

flafyrex, 2016-10-30:
Got a progress! Thank you! You can close (and delete this,if you want).
root@SopIPTV:~/openwrt# ./scripts/feeds install asterisk13-chan-bcm63xx-phone
Installing package 'asterisk-13-chan-bcm63xx-phone'
root@SopIPTV:~/openwrt# 
(SopIPTV is the host name, it was a project I was working on my laptop)

flafyrex, 2016-10-30:
Nope! I'm back. Now It wouldn't compile. HMMM?!?

Log:

root@SopIPTV:~/openwrt# make V=99
tmp/.config-package.in:36:warning: ignoring type redefinition of 'PACKAGE_libc' from 'boolean' to 'tristate'
tmp/.config-package.in:64:warning: ignoring type redefinition of 'PACKAGE_libgcc' from 'boolean' to 'tristate'
tmp/.config-package.in:1281:warning: ignoring type redefinition of 'PACKAGE_kmod-crypto-aes' from 'boolean' to 'tristate'
tmp/.config-package.in:1290:warning: ignoring type redefinition of 'PACKAGE_kmod-crypto-arc4' from 'boolean' to 'tristate'
tmp/.config-package.in:1330:warning: ignoring type redefinition of 'PACKAGE_kmod-crypto-core' from 'boolean' to 'tristate'
tmp/.config-package.in:2642:warning: ignoring type redefinition of 'PACKAGE_kmod-lib-crc-ccitt' from 'boolean' to 'tristate'
tmp/.config-package.in:2962:warning: ignoring type redefinition of 'PACKAGE_kmod-ip6tables' from 'boolean' to 'tristate'
tmp/.config-package.in:3021:warning: ignoring type redefinition of 'PACKAGE_kmod-ipt-conntrack' from 'boolean' to 'tristate'
tmp/.config-package.in:3053:warning: ignoring type redefinition of 'PACKAGE_kmod-ipt-core' from 'boolean' to 'tristate'
tmp/.config-package.in:3182:warning: ignoring type redefinition of 'PACKAGE_kmod-ipt-nat' from 'boolean' to 'tristate'
tmp/.config-package.in:3287:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-conntrack' from 'boolean' to 'tristate'
tmp/.config-package.in:3306:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-conntrack6' from 'boolean' to 'tristate'
tmp/.config-package.in:3316:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-ipt' from 'boolean' to 'tristate'
tmp/.config-package.in:3324:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-ipt6' from 'boolean' to 'tristate'
tmp/.config-package.in:3334:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-nat' from 'boolean' to 'tristate'
tmp/.config-package.in:3355:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-nathelper' from 'boolean' to 'tristate'
tmp/.config-package.in:4156:warning: ignoring type redefinition of 'PACKAGE_kmod-ipv6' from 'boolean' to 'tristate'
tmp/.config-package.in:4249:warning: ignoring type redefinition of 'PACKAGE_kmod-ppp' from 'boolean' to 'tristate'
tmp/.config-package.in:4291:warning: ignoring type redefinition of 'PACKAGE_kmod-pppoe' from 'boolean' to 'tristate'
tmp/.config-package.in:4312:warning: ignoring type redefinition of 'PACKAGE_kmod-pppox' from 'boolean' to 'tristate'
tmp/.config-package.in:4406:warning: ignoring type redefinition of 'PACKAGE_kmod-slhc' from 'boolean' to 'tristate'
tmp/.config-package.in:4488:warning: ignoring type redefinition of 'PACKAGE_kmod-bcma' from 'boolean' to 'tristate'
Config-build.in:4619:warning: defaults for choice values not supported
#
# configuration written to .config
#
make[1]: Entering directory '/root/openwrt'
make[2]: Entering directory '/root/openwrt'
make[3]: Entering directory '/root/openwrt/feeds/base/package/libs/toolchain'
if [ -f /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libpthread" >> /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/toolchain.default.install
make[3]: Leaving directory '/root/openwrt/feeds/base/package/libs/toolchain'
make[3]: Entering directory '/root/openwrt/feeds/base/package/libs/polarssl'
echo '1.3.14-1' | cmp -s - /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/libpolarssl.version || echo '1.3.14-1' > /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/pkginfo/libpolarssl.version
make[3]: Leaving directory '/root/openwrt/feeds/base/package/libs/polarssl'
make[3]: Entering directory '/root/openwrt/feeds/base/package/network/utils/curl'
rm -f /root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/stamp/.curl_installed
Makefile:177: recipe for target '/root/openwrt/build_dir/target-mips_mips32_uClibc-0.9.33.2/curl-7.40.0/.configured_ynnnnynnnnnyynynnnnnnnynnnynnynnnnyynnnnn' failed
make[3]: [/root/openwrt/build_dir/target-mips_mips32_uClibc-0.9.33.2/curl-7.40.0/.configured_ynnnnynnnnnyynynnnnnnnynnnynnynnnnyynnnnn] Error 123 (ignored)
(cd /root/openwrt/build_dir/target-mips_mips32_uClibc-0.9.33.2/curl-7.40.0; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /root/openwrt/scripts [... cut ...]
package/Makefile:191: recipe for target 'package/feeds/base/curl/compile' failed
make[2]: *** [package/feeds/base/curl/compile] Error 2
make[2]: Leaving directory '/root/openwrt'
package/Makefile:188: recipe for target '/root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openwrt'
/root/openwrt/include/toplevel.mk:174: recipe for target 'world' failed
make: *** [world] Error 2

pgid69, 2016-10-30:
Hello,

- bcm-phone-test is useless for you : it's a test program that i use during my dev to test PCM bus, SPI bus... You don't need it.
- you say that you wrote to me on the forum. I 've just listed all the messages you have posted : for those under topic Voip HG556 sorry but i'm not subscribe to this topic, and for those under topic support for hg553, i only saw the first message and i don't know how it's possible to guess that you're talking about packages of bcm63xx-phone module. 
- in the log of compilation above, it's curl that don't compile : the following line "checking for ssl_init in -lpolarssl... no" means that there's a problem compiling curl with polarssl. You should disable polarssl and enable openssl instead as openssl is already selected by asterisk13.
- there are several ways to compile external OpenWrt packages.
My method is to add the following line in feeds.conf
src-link custom /usr/local/src/openwrt/extern/bcm63xx-phone
Then i run 
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
i choose asterisk13-chan-bcm63xx-phone under section asterisk13
and finally 
make V=sc -j 5

flafyrex, 2016-10-30:
Worked for a while, then:

checking for mode_t... yes
checking for sig_atomic_t... yes
checking for pid_t... yes
checking for eaccess... yes
checking for mbstate_t... yes
checking whether stat file-mode macros are broken... no
checking for nlink_t... yes
checking whether fchmodat is declared without a macro... yes
checking whether fstat is declared without a macro... yes
checking whether fstatat is declared without a macro... yes
checking whether futimens is declared without a macro... yes
checking whether lchmod is declared without a macro... yes
checking whether lstat is declared without a macro... yes
checking whether mkdirat is declared without a macro... yes
checking whether mkfifo is declared without a macro... yes
checking whether mkfifoat is declared without a macro... yes
checking whether mknod is declared without a macro... yes
checking whether mknodat is declared without a macro... yes
checking whether stat is declared without a macro... yes
checking whether utimensat is declared without a macro... yes
checking whether getdelim is declared... yes
checking whether getdtablesize is declared... yes
checking whether getline is declared... yes
checking for getopt.h... (cached) yes
checking for getopt_long_only... yes
checking whether getopt is POSIX compatible... yes
checking for working GNU getopt function... yes
checking for working GNU getopt_long function... yes
checking whether getenv is declared... yes
checking for C/C++ restrict keyword... __restrict
checking for struct timeval... yes
checking for wide-enough struct timeval.tv_sec member... yes
checking whether gettimeofday is declared without a macro... yes
checking whether iswcntrl works... yes
checking for towlower... yes
checking for wctype_t... yes
checking for wctrans_t... yes
checking whether wctype is declared without a macro... yes
checking whether iswctype is declared without a macro... yes
checking whether wctrans is declared without a macro... yes
checking whether towctrans is declared without a macro... yes
checking whether included glib is requested... no
checking for glib... no
checking whether to use the included glib... yes
checking for unistd.h... (cached) yes
checking whether included libunistring is requested... no
checking for libunistring... no, consider installing GNU libunistring
checking whether getc_unlocked is declared... (cached) yes
checking for max_align_t... no
checking whether NULL can be used in arbitrary expressions... yes
checking whether lstat correctly handles trailing slash... yes
checking whether malloc, realloc, calloc are POSIX compliant... yes
checking for stdlib.h... (cached) yes
checking for GNU libc compatible malloc... yes
checking for a traditional japanese locale... none
checking for a transitional chinese locale... none
checking for a french Unicode locale... none
checking for a traditional french locale... none
checking for mmap... yes
checking for MAP_ANONYMOUS... yes
checking whether memchr works... yes
checking whether <limits.h> defines MIN and MAX... no
checking whether <sys/param.h> defines MIN and MAX... yes
checking for promoted mode_t type... mode_t
checking for library containing posix_spawn... none required
checking for posix_spawn... yes
checking whether posix_spawn works... yes
checking whether posix_spawnattr_setschedpolicy is supported... yes
[... 903 lines cut ...]
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
Makefile:370: recipe for target 'aclocal.m4' failed
make[4]: *** [aclocal.m4] Error 127
make[4]: Leaving directory '/root/openwrt/build_dir/host/gettext-0.19.6'
Makefile:78: recipe for target '/root/openwrt/build_dir/host/gettext-0.19.6/.built' failed
make[3]: *** [/root/openwrt/build_dir/host/gettext-0.19.6/.built] Error 2
make[3]: Leaving directory '/root/openwrt/feeds/base/package/libs/gettext-full'
package/Makefile:191: recipe for target 'package/feeds/base/gettext-full/host/compile' failed
make[2]: *** [package/feeds/base/gettext-full/host/compile] Error 2
make[2]: Leaving directory '/root/openwrt'
package/Makefile:188: recipe for target '/root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openwrt'
/root/openwrt/include/toplevel.mk:174: recipe for target 'world' failed
make: *** [world] Error 2
root@SopIPTV:~/openwrt# 

Hmmm?!?

flafyrex, 2016-10-30:
https://forum.openwrt.org/viewtopic.php?id=30947&p=3
Your username was there,and so here.

pgid69, 2016-10-30:
The compilation log is explicit<br>
```
/root/openwrt/build_dir/host/gettext-0.19.6/build-aux/missing: line 81: aclocal-1.15: command not found
WARNING: 'aclocal-1.15' is missing on your system.
You should only need it if you modified 'acinclude.m4' or
'configure.ac' or m4 files included by 'configure.ac'.
The 'aclocal' program is part of the GNU Automake package:
http://www.gnu.org/software/automake
It also requires GNU Autoconf, GNU m4 and Perl in order to run:
http://www.gnu.org/software/autoconf
http://www.gnu.org/software/m4/
http://www.perl.org/
Makefile:370: recipe for target 'aclocal.m4' failed
make[4]: ** [aclocal.m4] Error 127
```
You need to install some missing packages, automake, autoconf, perl, m4

flafyrex, 2016-10-30:
Found it too,I am compiling automake 1.15. Then back to make V=sc -j2

flafyrex, 2016-10-30:
Now it won't even compile. Don't forget: I use Debian Jessie 8
What OS you used? I want to use your kmod. I want to put the phone ports in use.

```
root@SopIPTV:~/openwrt# make V=sc -j2
Checking 'xgettext'... ok.
tmp/.config-package.in:36:warning: ignoring type redefinition of 'PACKAGE_libc' from 'boolean' to 'tristate'
tmp/.config-package.in:64:warning: ignoring type redefinition of 'PACKAGE_libgcc' from 'boolean' to 'tristate'
tmp/.config-package.in:1281:warning: ignoring type redefinition of 'PACKAGE_kmod-crypto-aes' from 'boolean' to 'tristate'
tmp/.config-package.in:1290:warning: ignoring type redefinition of 'PACKAGE_kmod-crypto-arc4' from 'boolean' to 'tristate'
tmp/.config-package.in:1330:warning: ignoring type redefinition of 'PACKAGE_kmod-crypto-core' from 'boolean' to 'tristate'
tmp/.config-package.in:2642:warning: ignoring type redefinition of 'PACKAGE_kmod-lib-crc-ccitt' from 'boolean' to 'tristate'
tmp/.config-package.in:2962:warning: ignoring type redefinition of 'PACKAGE_kmod-ip6tables' from 'boolean' to 'tristate'
tmp/.config-package.in:3021:warning: ignoring type redefinition of 'PACKAGE_kmod-ipt-conntrack' from 'boolean' to 'tristate'
tmp/.config-package.in:3053:warning: ignoring type redefinition of 'PACKAGE_kmod-ipt-core' from 'boolean' to 'tristate'
tmp/.config-package.in:3182:warning: ignoring type redefinition of 'PACKAGE_kmod-ipt-nat' from 'boolean' to 'tristate'
tmp/.config-package.in:3287:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-conntrack' from 'boolean' to 'tristate'
tmp/.config-package.in:3306:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-conntrack6' from 'boolean' to 'tristate'
tmp/.config-package.in:3316:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-ipt' from 'boolean' to 'tristate'
tmp/.config-package.in:3324:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-ipt6' from 'boolean' to 'tristate'
tmp/.config-package.in:3334:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-nat' from 'boolean' to 'tristate'
tmp/.config-package.in:3355:warning: ignoring type redefinition of 'PACKAGE_kmod-nf-nathelper' from 'boolean' to 'tristate'
tmp/.config-package.in:4156:warning: ignoring type redefinition of 'PACKAGE_kmod-ipv6' from 'boolean' to 'tristate'
tmp/.config-package.in:4249:warning: ignoring type redefinition of 'PACKAGE_kmod-ppp' from 'boolean' to 'tristate'
tmp/.config-package.in:4291:warning: ignoring type redefinition of 'PACKAGE_kmod-pppoe' from 'boolean' to 'tristate'
tmp/.config-package.in:4312:warning: ignoring type redefinition of 'PACKAGE_kmod-pppox' from 'boolean' to 'tristate'
tmp/.config-package.in:4406:warning: ignoring type redefinition of 'PACKAGE_kmod-slhc' from 'boolean' to 'tristate'
tmp/.config-package.in:4488:warning: ignoring type redefinition of 'PACKAGE_kmod-bcma' from 'boolean' to 'tristate'
Config-build.in:4619:warning: defaults for choice values not supported
#
# configuration written to .config
#
make[1]: Entering directory '/root/openwrt'
make[2]: Entering directory '/root/openwrt'
make[3]: Entering directory '/root/openwrt/feeds/base/package/libs/toolchain'
make[3]: Entering directory '/root/openwrt/feeds/base/package/libs/gettext-full'
make -j1 -C /root/openwrt/build_dir/host/gettext-0.19.6  
make[4]: Entering directory '/root/openwrt/build_dir/host/gettext-0.19.6'
CDPATH="${ZSH_VERSION+.}:" && cd . && /usr/bin/env bash /root/openwrt/build_dir/host/gettext-0.19.6/build-aux/missing aclocal-1.15 -I m4
cp -fpR /root/openwrt/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libuClibc-*.so /root/openwrt/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libcrypt-*.so /root/openwrt/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libm-*.so /root/openwrt/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libpthread-*.so /root/openwrt/build_dir/target-mips_mips32_uClibc-0.9.33.2/toolchain/
cp: cannot stat '/root/openwrt/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libuClibc-*.so': No such file or directory
cp: cannot stat '/root/openwrt/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libcrypt-*.so': No such file or directory
cp: cannot stat '/root/openwrt/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libm-*.so': No such file or directory
cp: cannot stat '/root/openwrt/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libpthread-*.so': No such file or directory
Makefile:636: recipe for target '/root/openwrt/build_dir/target-mips_mips32_uClibc-0.9.33.2/toolchain/.built' failed
make[3]: *** [/root/openwrt/build_dir/target-mips_mips32_uClibc-0.9.33.2/toolchain/.built] Error 1
make[3]: Leaving directory '/root/openwrt/feeds/base/package/libs/toolchain'
package/Makefile:191: recipe for target 'package/feeds/base/toolchain/compile' failed
make[2]: *** [package/feeds/base/toolchain/compile] Error 2
make[2]: *** Waiting for unfinished jobs....
autom4te: need GNU m4 1.4 or later: /root/openwrt/staging_dir/host/bin/m4
aclocal-1.15: error: echo failed with exit status: 1
Makefile:370: recipe for target 'aclocal.m4' failed
make[4]: *** [aclocal.m4] Error 1
make[4]: Leaving directory '/root/openwrt/build_dir/host/gettext-0.19.6'
Makefile:78: recipe for target '/root/openwrt/build_dir/host/gettext-0.19.6/.built' failed
make[3]: *** [/root/openwrt/build_dir/host/gettext-0.19.6/.built] Error 2
[... cut ...]
package/Makefile:188: recipe for target '/root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/root/openwrt/staging_dir/target-mips_mips32_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openwrt'
/root/openwrt/include/toplevel.mk:174: recipe for target 'world' failed
make: *** [world] Error 2
```

Why?!? Maybe because
```
autom4te: need GNU m4 1.4 or later: /root/openwrt/staging_dir/host/bin/m4
aclocal-1.15: error: echo failed with exit status: 1
```
?

flafyrex, 2016-10-30:
Going to bed... Good night there in France(GMT +2:00). At us(RO) now is 19:53. Good night.

pgid69, 2016-11-01:
Hello
I'm also using Jessie 8 for compiling, so i don't understand where is the problem.
I suggest you to send me your .config and your feeds.conf so that i can try to compile OpenWrt for you.
My email is mazoyer.gilles@omega.ovh.
A last world : as the problem you have is not related to bcm63xx-phone i'm closing the issue : you have my email to communicate with me.
BR

pgid69, 2016-11-07:
Issue closed because not related to bcm63xx-phone, but on how to compile custom OpenWrt kernel

closed 2016-11-07 by pgid69