pgid69/bcm63xx-phone #9: bcm63xx-phone kernel module fail to build on lede-project HEAD

dpeddi, 2016-11-06:
Hi,

it seems that with the transition to linux-4.4.28  on bm63xx platform, disappeared some essentials headers.

/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c:185:28: error: 'SPI_CLK_CFG' undeclared (first use in this function)
    reg = bcm_spi_readb(bs, SPI_CLK_CFG);
                            ^
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c:185:28: note: each undeclared identifier is reported only once for each function it appears in
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c: In function 'bcm_mpi_set_fill_byte':
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c:198:34: error: 'SPI_FILL_BYTE' undeclared (first use in this function)
    bcm_spi_writeb(bs, fill_byte, SPI_FILL_BYTE);
                                  ^
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c: In function 'bcm_mpi_rw_buf':
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c:229:26: error: 'SPI_INT_MASK' undeclared (first use in this function)
    bcm_spi_writeb(bs, 0, SPI_INT_MASK);
                          ^
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c:273:38: error: 'SPI_MSG_CTL' undeclared (first use in this function)
          bcm_spi_writeb(bs, msg_ctl, SPI_MSG_CTL);
                                      ^
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c:283:43: error: 'SPI_INT_STATUS' undeclared (first use in this function)
    bcm_spi_writeb(bs, SPI_INTR_CLEAR_ALL, SPI_INT_STATUS);
                                           ^
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c:290:28: error: 'SPI_CMD' undeclared (first use in this function)
    bcm_spi_writew(bs, cmd, SPI_CMD);
                            ^
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c:306:35: error: 'SPI_RX_TAIL' undeclared (first use in this function)
       rx_tail = bcm_spi_readb(bs, SPI_RX_TAIL);
                                   ^
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c: In function 'bcm63xx_spi_interrupt':
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c:327:29: error: 'SPI_INT_STATUS' undeclared (first use in this function)
    intr = bcm_spi_readb(bs, SPI_INT_STATUS);
                             ^
/data/devel/router/lede/source/build_dir/target-mips_mips32_musl-1.1.15/bcm63xx-phone/mpi/mpi.c:329:26: error: 'SPI_INT_MASK' undeclared (first use in this function)
    bcm_spi_writeb(bs, 0, SPI_INT_MASK);

pgid69, 2016-11-07:
Hi,

Thank you for your feedback.
I will try to resolve the problem as soon as possible.

pgid69, 2016-11-07:
I fixed the problem

dpeddi, 2016-11-07:
wow super fast :+1:  

I'll try early!

closed 2016-11-20 by pgid69