pgid69/bcm63xx-phone #11: can compile with sip stack and then register to sip servers for voip?

cubieb, 2016-11-17:
Thanks for your opensource project, that is really a good job. 
the hg556a origin firmware support a sip client then we can register to sip server on the internet

can we port some sip stack to impliment that functions?

thanks,
bob

pgid69, 2016-11-17:
bcm63xx-phone is a kernel module that is intended to be used with the Asterisk channel i developped. The SIP stack is in Asterisk. It's Asterisk that register to a SIP server.
I don't understand what you mean by "can we port some sip stack" ? What do you want to do ?

cubieb, 2016-11-18:
I know， Asterisk at here is just like a sip proxy， if i do not install asterisk， can we setup the sip call   success？

pgid69, 2016-11-18:
No without Asterisk, no SIP stack, so no SIP call.

closed 2016-11-20 by pgid69