pgid69/bcm63xx-phone #13: bcm63xx-phone compiled but not working

redocili, 2017-05-01:
hello
i'm using an huawei hg553 with lede 4.4.50
Bcm63xx-phone compiled corrcetly (kernel compiling without bcm63xx-spi)
when insmod bcm63xx-phone i get this (with dmesg):

[  224.361392] bcm63xx-phone: bcm63xx-phone: v0.1
[  224.366025] bcm63xx-phone bcm_drv_init() 1442: bcm_drv_init()
[  224.372004] bcm63xx-phone board_init() 706: board_init()
[  224.377509] bcm63xx-phone: CPU is Broadcom 6358
[  224.382208] bcm63xx-phone: Board is HW553
[  224.386380] bcm63xx-phone board_get_desc() 764: board_get_desc()
[  224.392660] bcm63xx-phone bcm_drv_init() 1457: Initializing mutex
[  224.398980] bcm63xx-phone bcm_drv_init() 1460: Initializing RX wait queue
[  224.406009] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
[  224.413042] bcm63xx-phone bcm_drv_init() 1462: Initializing TX wait queue
[  224.420074] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
[  224.427114] bcm63xx-phone bcm_drv_init() 1477: Registering device bcm63xx-phone
[  224.435053] bcm63xx-phone pcm_init() 2482: pcm_init()
[  224.440344] bcm63xx-phone pcm_init() 2550: Using 25 buffers of 256 bytes for RX
[  224.447948] bcm63xx-phone pcm_init() 2586: Using 25 buffers of 256 bytes for TX
[  224.455514] bcm63xx-phone pcm_get_shared_res() 1950: pcm_get_shared_res()
[  224.462557] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm, addr=0xfffe1600, size=256)
[  224.472033] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dma, addr=0xfffe1800, size=256)
[  224.481867] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmac, addr=0xfffe1900, size=32)
[  224.491697] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmas, addr=0xfffe1a00, size=32)
[  224.977033] bcm63xx-phone pcm6358_pll_init() 970: pcm6358_pll_init()
[  224.983692] bcm63xx-phone pcm6358_pll_init() 975:  pll_ctrl1=f0010001, pll_ctrl2=53d47, pll_ctrl3=a85
[  225.041014] bcm63xx-phone pcm6358_pll_init() 1028:  pll_ctrl1=20001, pll_ctrl2=8492b, pll_ctrl3=1e1c
[  225.050567] bcm63xx-phone pcm_regs_init() 1247: pcm_regs_init()
[  225.056726] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
[  225.063049] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
[  225.070263] bcm63xx-phone pcm_dma_init() 1502: pcm_dma_init()
[  225.076225] bcm63xx-phone pcm_dma_init() 1535: Allocating 200 bytes of DMA memory for RX descs
[  225.085212] bcm63xx-phone pcm_dma_init() 1546: 200 bytes of memory allocated, starting at address 0xa0d50000
[  225.095396] bcm63xx-phone pcm_dma_init() 1552: Allocating 200 bytes of DMA memory for TX descs
[  225.104439] bcm63xx-phone pcm_dma_init() 1563: 200 bytes of memory allocated, starting at address 0xa0d2b000
[  225.114630] bcm63xx-phone pcm_dma_init() 1571: Allocating 12800 bytes of memory for RX and TX data buffers
[  225.124643] bcm63xx-phone pcm_dma_init() 1578: 12800 bytes of memory allocated, starting at address 0x80d20000
[  225.135004] bcm63xx-phone pcm_dma_init() 1604: Allocating 224 bytes of memory for DMA addresses
[  225.144020] bcm63xx-phone pcm_dma_init() 1611: 224 bytes of memory allocated, starting at address 0x82da1700
[  225.154189] bcm63xx-phone pcm_dma_init() 1618: Mapping DMA addresses of RX descs
[  225.161871] bcm63xx-phone pcm_dma_init() 1635: Mapping DMA addresses of TX descs
[  225.169551] bcm63xx-phone pcm_dma_rx_descs_reset() 1342: pcm_dma_rx_descs_reset()
[  225.177387] bcm63xx-phone pcm_dma_tx_descs_reset() 1373: pcm_dma_tx_descs_reset()
[  225.185241] bcm63xx-phone bcm_drv_init() 1527: t->mm_desc.ioctl_param_off = 0, t->mm_desc.ioctl_param_size = 208
[  225.195796] bcm63xx-phone bcm_drv_init() 1537: t->mm_desc.rbs[0].rx_ring_buf_desc_off = 208, t->mm_desc.rbs[0].rx_ring_buf_d                                                  esc_size = 12
[  225.208578] bcm63xx-phone bcm_drv_init() 1545: t->mm_desc.rbs[0].tx_ring_buf_desc_off = 224, t->mm_desc.rbs[0].tx_ring_buf_d                                                  esc_size = 12
[  225.221355] bcm63xx-phone bcm_drv_init() 1537: t->mm_desc.rbs[1].rx_ring_buf_desc_off = 240, t->mm_desc.rbs[1].rx_ring_buf_d                                                  esc_size = 12
[  225.234131] bcm63xx-phone bcm_drv_init() 1545: t->mm_desc.rbs[1].tx_ring_buf_desc_off = 256, t->mm_desc.rbs[1].tx_ring_buf_d                                                  esc_size = 12
[  225.247025] bcm63xx-phone bcm_drv_init() 1567: 8192 bytes of memory allocated, starting at address 0x80d2e000
[  225.257306] bcm63xx-phone bcm_phone_mgr_init() 851: bcm_phone_mgr_init()
[  225.264254] bcm63xx-phone bcm_phone_mgr_devs_init() 725: bcm_phone_mgr_devs_init()
[  225.272098] bcm63xx-phone phone_device_alloc() 55: phone_device_alloc()
[  225.278995] bcm63xx-phone phone_device_alloc() 66: 2020 bytes of memory allocated, starting at address 0x82d93000
[  225.289612] bcm63xx-phone phone_dev_le88221_init() 54: phone_dev_le88221_init()
[  225.297184] bcm63xx-phone phone_dev_zarlink_ve880_init() 1948: phone_dev_zarlink_ve880_init()
[  225.306006] bcm63xx-phone bcm_mpi_init() 919: bcm_mpi_init()
[  225.312349] bcm63xx-phone bcm_mpi_init() 983: condition '((ret) && (0 == bcm_mpi_dev_data.ref_count)) || ((!ret) && (1 == bc                                                  m_mpi_dev_data.ref_count))' is false
[  225.327220] bcm63xx-phone bcm_mpi_init() 990: condition 'bcm_mpi_dev_data.ref_count > 0' is false
[  225.336503] CPU 0 Unable to handle kernel paging request at virtual address 00000054, epc == 801b4628, ra == 801b4ae8
[  225.347490] Oops[#1]:
[  225.349796] CPU: 0 PID: 4704 Comm: insmod Not tainted 4.4.50 #0
[  225.355882] task: 82ec5e30 ti: 80cf6000 task.ti: 80cf6000
[  225.361423] $ 0   : 00000000 00000001 80cf7b7c 00000001
[  225.366797] $ 4   : 00000010 00000010 80cf7b80 00000002
[  225.372173] $ 8   : 616c7365 6e74203e 20302720 69732066
[  225.377548] $12   : 00000000 ffffff80 00000000 00000000
[  225.382925] $16   : 00000010 00000033 80cf7b80 00000000
[  225.388301] $20   : 82d9310c 82d9310c 0000000a 80de2c34
[  225.393676] $24   : 00000000 8005141c
[... 11 lines cut ...]
[  225.588288] Stack : 0000000a 80de2c34 80de0000 80059260 80de0000 80de7380 80de0000 80de705c
[  225.588288]    80de0000 801b4c78 80de0000 80057164 80de0000 80059260 80526d62 80cf7b98
[  225.588288]    80de0810 80cf7b7c 80de705c 80d98508 80de7050 80ddd4ec 80ddd4ec 00000002
[  225.588288]    80de7050 80de0000 82d93000 80de7410 0000000a 80d9a124 00000000 80dddf5c
[  225.588288]    0000079c 0000000a 00000000 80057164 80dddb10 82d93000 80de7050 00000002
[  225.588288]    ...
[  225.624844] Call Trace:
[  225.627399] [<801b4628>] dev_driver_string+0x0/0x38
[  225.632402] [<801b4ae8>] __dev_printk+0x30/0x88
[  225.637065] [<801b4c78>] dev_err+0x38/0x44
[  225.641637] [<80d98508>] bcm_mpi_init+0x204/0x260 [bcm63xx_phone]
[  225.648444] [<80d9a124>] phone_dev_zarlink_ve880_init+0x9c/0x100 [bcm63xx_phone]
[  225.656598] [<80d99b3c>] phone_dev_le88221_init+0x110/0x288 [bcm63xx_phone]
[  225.664302] [<80d993f8>] phone_device_alloc+0xac/0x154 [bcm63xx_phone]
[  225.671550] [<80d99658>] bcm_phone_mgr_init+0x1b8/0x58c [bcm63xx_phone]
[  225.678889] [<80d9a6c4>] init_module+0x48c/0x594 [bcm63xx_phone]
[  225.685345] [<80019ad0>] do_one_initcall+0x200/0x22c
[  225.690464] [<80075e78>] do_init_module+0x74/0x1c8
[  225.695376] [<80077710>] load_module+0x16c8/0x1b70
[  225.700304] [<80077cd0>] SyS_init_module+0x118/0x168
[  225.705418] [<800129d8>] syscall_common+0x30/0x54
[  225.710224]
[  225.711740]
[  225.711740] Code: 8fb00018  03e00008  27bd0030 <8c820044> 14400007  00000000  8c820040  14400004  00000000
[  225.722091] ---[ end trace a991f10ae33ecc3d ]---

pgid69, 2017-05-02:
Hello

From the log i think the kernel panic is caused because in bcm_mpi_init() when platform_driver_register() is called, the function bcm63xx_spi_probe() is not called, so the driver is half initialized.
I will try to find out why, but it will be difficult because my router is not running lede and i don't want to upgrade it.
So if i find the reason, and i you're ok, i will ask you to test.

redocili, 2017-05-02:
Sure i will.

pgid69, 2017-05-03:
Hello

I think i found the problem, why bcm63xx_spi_probe() is not called.
Before i try to make a patch can you make a modification in file bcm63xx-phone/src/mpi/mpi.c at line 559 : replace line
    .name = "bcm63xx-spi",
by line
    .name = "bcm6358-spi",
Then recompile and test.
Thank you for your help.

Gilles

pgid69, 2017-05-03:
Hello

I think i found the problem
Before i try to make a patch can you test the following modification : in file bcm63xx-phone/src/mpi/mpi.c at line 559 replace
      .name = "bcm63xx-spi",
with
      .name = "bcm6358-spi",
Then recompile and test.
Thank you for your help.

redocili, 2017-05-04:
It seems ok, ko loaded succesfully, i will test soon voip functionality..

`[  320.938413] bcm63xx-phone: bcm63xx-phone: v0.1
[  320.943071] bcm63xx-phone bcm_drv_init() 1442: bcm_drv_init()
[  320.949127] bcm63xx-phone board_init() 706: board_init()
[  320.954649] bcm63xx-phone: CPU is Broadcom 6358
[  320.959367] bcm63xx-phone: Board is HW553
[  320.963548] bcm63xx-phone board_get_desc() 764: board_get_desc()
[  320.969791] bcm63xx-phone bcm_drv_init() 1457: Initializing mutex
[  320.976115] bcm63xx-phone bcm_drv_init() 1460: Initializing RX wait queue
[  320.983158] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
[  320.990202] bcm63xx-phone bcm_drv_init() 1462: Initializing TX wait queue
[  320.997275] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
[  321.004323] bcm63xx-phone bcm_drv_init() 1477: Registering device bcm63xx-phone
[  321.012286] bcm63xx-phone pcm_init() 2482: pcm_init()
[  321.017597] bcm63xx-phone pcm_init() 2550: Using 25 buffers of 256 bytes for RX
[  321.025208] bcm63xx-phone pcm_init() 2586: Using 25 buffers of 256 bytes for TX
[  321.032868] bcm63xx-phone pcm_get_shared_res() 1950: pcm_get_shared_res()
[  321.039937] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm, addr=0xfffe1600, size=256)
[  321.049426] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dma, addr=0xfffe1800, size=256)
[  321.059271] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmac, addr=0xfffe1900, size=32)
[  321.069121] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmas, addr=0xfffe1a00, size=32)
[  321.165347] bcm63xx-phone pcm6358_pll_init() 970: pcm6358_pll_init()
[  321.171972] bcm63xx-phone pcm6358_pll_init() 975:  pll_ctrl1=f0010001, pll_ctrl2=53d47, pll_ctrl3=a85
[  321.213375] bcm63xx-phone pcm6358_pll_init() 1028:  pll_ctrl1=20001, pll_ctrl2=8492b, pll_ctrl3=1e1c
[  321.222873] bcm63xx-phone pcm_regs_init() 1247: pcm_regs_init()
[  321.229050] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
[  321.235384] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
[  321.242608] bcm63xx-phone pcm_dma_init() 1502: pcm_dma_init()
[  321.248588] bcm63xx-phone pcm_dma_init() 1535: Allocating 200 bytes of DMA memory for RX descs
[  321.257657] bcm63xx-phone pcm_dma_init() 1546: 200 bytes of memory allocated, starting at address 0xa2f9c000
[  321.267858] bcm63xx-phone pcm_dma_init() 1552: Allocating 200 bytes of DMA memory for TX descs
[  321.276859] bcm63xx-phone pcm_dma_init() 1563: 200 bytes of memory allocated, starting at address 0xa2eec000
[  321.287060] bcm63xx-phone pcm_dma_init() 1571: Allocating 12800 bytes of memory for RX and TX data buffers
[  321.297437] bcm63xx-phone pcm_dma_init() 1578: 12800 bytes of memory allocated, starting at address 0x80c04000
[  321.307833] bcm63xx-phone pcm_dma_init() 1604: Allocating 224 bytes of memory for DMA addresses
[  321.316851] bcm63xx-phone pcm_dma_init() 1611: 224 bytes of memory allocated, starting at address 0x82fe9300
[  321.327025] bcm63xx-phone pcm_dma_init() 1618: Mapping DMA addresses of RX descs
[  321.334724] bcm63xx-phone pcm_dma_init() 1635: Mapping DMA addresses of TX descs
[  321.342417] bcm63xx-phone pcm_dma_rx_descs_reset() 1342: pcm_dma_rx_descs_reset()
[  321.350256] bcm63xx-phone pcm_dma_tx_descs_reset() 1373: pcm_dma_tx_descs_reset()
[  321.358114] bcm63xx-phone bcm_drv_init() 1527: t->mm_desc.ioctl_param_off = 0, t->mm_desc.ioctl_param_size = 208
[  321.368665] bcm63xx-phone bcm_drv_init() 1537: t->mm_desc.rbs[0].rx_ring_buf_desc_off = 208, t->mm_desc.rbs[0].rx_ring_buf_desc_size = 12
[  321.381451] bcm63xx-phone bcm_drv_init() 1545: t->mm_desc.rbs[0].tx_ring_buf_desc_off = 224, t->mm_desc.rbs[0].tx_ring_buf_desc_size = 12
[  321.394237] bcm63xx-phone bcm_drv_init() 1537: t->mm_desc.rbs[1].rx_ring_buf_desc_off = 240, t->mm_desc.rbs[1].rx_ring_buf_desc_size = 12
[  321.407021] bcm63xx-phone bcm_drv_init() 1545: t->mm_desc.rbs[1].tx_ring_buf_desc_off = 256, t->mm_desc.rbs[1].tx_ring_buf_desc_size = 12
[  321.419926] bcm63xx-phone bcm_drv_init() 1567: 8192 bytes of memory allocated, starting at address 0x838a8000
[  321.430206] bcm63xx-phone bcm_phone_mgr_init() 851: bcm_phone_mgr_init()
[  321.437170] bcm63xx-phone bcm_phone_mgr_devs_init() 725: bcm_phone_mgr_devs_init()
[  321.445018] bcm63xx-phone phone_device_alloc() 55: phone_device_alloc()
[  321.451916] bcm63xx-phone phone_device_alloc() 66: 2020 bytes of memory allocated, starting at address 0x80c08800
[  321.462540] bcm63xx-phone phone_dev_le88221_init() 54: phone_dev_le88221_init()
[  321.470119] bcm63xx-phone phone_dev_zarlink_ve880_init() 1948: phone_dev_zarlink_ve880_init()
[  321.478951] bcm63xx-phone bcm_mpi_init() 919: bcm_mpi_init()
[  321.484987] bcm63xx-phone bcm63xx_spi_probe() 412: bcm63xx_spi_probe()
[  321.491872] bcm6358-spi bcm6358-spi: at 0xfffe0800 (irq 9, FIFOs size 542)
[  321.499450] bcm63xx-phone phone_dev_zarlink_init() 674: phone_dev_zarlink_init()
[  321.507154] bcm63xx-phone zarlink_reset() 164: zarlink_reset(gpio_num=0x8018)
[  321.581397] bcm63xx-phone phone_device_init() 30: phone_device_init()
[  321.588101] bcm63xx-phone zarlink_device_id_init() 458: zarlink_device_id_init()
[  321.595826] bcm63xx-phone: 1 device(s) provides 2 FXS line(s)
[  321.601800] bcm63xx-phone bcm_phone_mgr_init() 869: Initializing event wait queue
[  321.609560] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
[  321.618482] bcm63xx-phone bcm_timer_init() 135: bcm_timer_init()
[  321.624749] bcm63xx-phone bcm_drv_init() 1581: bcm63xx-phone successfully initialized
`

redocili, 2017-05-05:
voip test in progress... used only one SIP provider to make and receive calls.
i have found some diffculty in configuring asterisk (i'm a noob here)

sip calls work great
no echo
don't display caller id when receiving calls, id caller works on outgoing calls

great works, many thanks
========

my test settings (maybe needs somebody)
=====
sip.conf
=====
[general]
context=default
disallow=all
allow=ulaw
allow=alaw

register => 12345678:password@sip_address:port/12345678

[provider]
type=friend
defaultuser=12345678
secret=password
insecure=port,invite
host=sip_address
context=from-voip
fromuser=12345678
user=12345678
disallow=all
allow=ulaw
allow=alaw

=========
extensions.conf
=========
[from-voip]
exten => 12345678,1,Dial(Bcm63xxPhone/1,120,t)
  same => n,Hangup(16)

[bcmph-line-1]
exten => _X.,1,Dial(SIP/provider/${EXTEN})
  same => n,Hangup(16)

===========

bcm63xx_phone.conf at default

pgid69, 2017-05-10:
Hello

Thank you for your feedback.
I will make a patch as soon as possible.

BR

redocili, 2017-05-10:
thanks
any chance to fix tho most important issue (echo cancellation)?
i have seen there are some implementation of some algos for bcm63xx here:
https://github.com/openwrt/bcm63xx-next/tree/master/drivers/staging/echo
have you ever tried to adapt code to your kernel driver?
Thanks

pgid69, 2017-05-30:
Hello

When i have time i will try to look at the code, understand it and use it.
Another possibility is to convert bcm63xx-phone driver into a DAHDI driver : then echo cancellation will be made by DAHDI as handling of tones...
Would be smarter but requires more work because i have to understand DAHDI architecture.

BR

closed 2017-06-06 by pgid69