pgid69/bcm63xx-phone #14: issue using bcm63xx-phone device on Pirelli DRG A226M

ivan1970, 2017-05-24:
Hello,
I'm using lede snapshot and I compiled my own image loading bcm63xx-phone kernel module and removing bcm63xx-spi kernel module. When I load the kernel module it is correctly loaded:

> root@LEDE:~# insmod bcm63xx-phone
> root@LEDE:~# lsmod | grep bcm
> bcm63xx_phone         169334  0

as you can see from the log:

> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.546182] bcm63xx-phone: bcm63xx-phone: v0.1
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.550853] bcm63xx-phone bcm_drv_init() 1442: bcm_drv_init()
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.556803] bcm63xx-phone board_init() 706: board_init()
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.562272] bcm63xx-phone: CPU is Broadcom 6358
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.566940] bcm63xx-phone: Board is DWV-S0
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.571201] bcm63xx-phone board_get_desc() 764: board_get_desc()
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.577394] bcm63xx-phone bcm_drv_init() 1457: Initializing mutex
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.583674] bcm63xx-phone bcm_drv_init() 1460: Initializing RX wait queue
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.590672] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.597668] bcm63xx-phone bcm_drv_init() 1462: Initializing TX wait queue
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.604666] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.611667] bcm63xx-phone bcm_drv_init() 1477: Registering device bcm63xx-phone
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.619530] bcm63xx-phone pcm_init() 2482: pcm_init()
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.624777] bcm63xx-phone pcm_init() 2550: Using 25 buffers of 256 bytes for RX
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.632332] bcm63xx-phone pcm_init() 2586: Using 25 buffers of 256 bytes for TX
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.639864] bcm63xx-phone pcm_get_shared_res() 1950: pcm_get_shared_res()
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.646870] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm, addr=0xfffe1600, size=256)
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.656310] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dma, addr=0xfffe1800, size=256)
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.666110] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmac, addr=0xfffe1900, size=32)
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.675904] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmas, addr=0xfffe1a00, size=32)
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.770580] bcm63xx-phone pcm6358_pll_init() 970: pcm6358_pll_init()
> Wed May 24 14:44:52 2017 kern.info kernel: [ 7499.777150] bcm63xx-phone pcm6358_pll_init() 975:  pll_ctrl1=f0010001, pll_ctrl2=53d47, pll_ctrl3=a85
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.818566] bcm63xx-phone pcm6358_pll_init() 1028:  pll_ctrl1=20001, pll_ctrl2=8492b, pll_ctrl3=1e1c
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.828005] bcm63xx-phone pcm_regs_init() 1247: pcm_regs_init()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.834117] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.840404] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.847590] bcm63xx-phone pcm_dma_init() 1502: pcm_dma_init()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.853526] bcm63xx-phone pcm_dma_init() 1535: Allocating 200 bytes of DMA memory for RX descs
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.862483] bcm63xx-phone pcm_dma_init() 1546: 200 bytes of memory allocated, starting at address 0xa0cfe000
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.872633] bcm63xx-phone pcm_dma_init() 1552: Allocating 200 bytes of DMA memory for TX descs
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.881588] bcm63xx-phone pcm_dma_init() 1563: 200 bytes of memory allocated, starting at address 0xa0c74000
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.891739] bcm63xx-phone pcm_dma_init() 1571: Allocating 12800 bytes of memory for RX and TX data buffers
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.901716] bcm63xx-phone pcm_dma_init() 1578: 12800 bytes of memory allocated, starting at address 0x80d40000
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.912046] bcm63xx-phone pcm_dma_init() 1604: Allocating 224 bytes of memory for DMA addresses
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.921033] bcm63xx-phone pcm_dma_init() 1611: 224 bytes of memory allocated, starting at address 0x81b42100
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.931179] bcm63xx-phone pcm_dma_init() 1618: Mapping DMA addresses of RX descs
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.938834] bcm63xx-phone pcm_dma_init() 1635: Mapping DMA addresses of TX descs
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.946483] bcm63xx-phone pcm_dma_rx_descs_reset() 1342: pcm_dma_rx_descs_reset()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.954290] bcm63xx-phone pcm_dma_tx_descs_reset() 1373: pcm_dma_tx_descs_reset()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.962117] bcm63xx-phone bcm_drv_init() 1527: t->mm_desc.ioctl_param_off = 0, t->mm_desc.ioctl_param_size = 208
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.972637] bcm63xx-phone bcm_drv_init() 1537: t->mm_desc.rbs[0].rx_ring_buf_desc_off = 208, t->mm_desc.rbs[0].rx_ring_buf_desc_size = 12
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.985391] bcm63xx-phone bcm_drv_init() 1545: t->mm_desc.rbs[0].tx_ring_buf_desc_off = 224, t->mm_desc.rbs[0].tx_ring_buf_desc_size = 12
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7499.998145] bcm63xx-phone bcm_drv_init() 1537: t->mm_desc.rbs[1].rx_ring_buf_desc_off = 240, t->mm_desc.rbs[1].rx_ring_buf_desc_size = 12
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.010895] bcm63xx-phone bcm_drv_init() 1545: t->mm_desc.rbs[1].tx_ring_buf_desc_off = 256, t->mm_desc.rbs[1].tx_ring_buf_desc_size = 12
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.023755] bcm63xx-phone bcm_drv_init() 1567: 8192 bytes of memory allocated, starting at address 0x80cfc000
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.034007] bcm63xx-phone bcm_phone_mgr_init() 851: bcm_phone_mgr_init()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.040937] bcm63xx-phone bcm_phone_mgr_devs_init() 725: bcm_phone_mgr_devs_init()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.048754] bcm63xx-phone phone_device_alloc() 55: phone_device_alloc()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.055611] bcm63xx-phone phone_device_alloc() 84: 2020 bytes of memory allocated, starting at address 0x80fa9000
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.066207] bcm63xx-phone phone_dev_le88266_init() 54: phone_dev_le88266_init()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.073750] bcm63xx-phone phone_dev_zarlink_ve880_init() 1948: phone_dev_zarlink_ve880_init()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.082554] bcm63xx-phone bcm_mpi_init() 919: bcm_mpi_init()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.088537] bcm63xx-phone bcm63xx_spi_probe() 412: bcm63xx_spi_probe()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.095392] bcm6358-spi bcm6358-spi: at 0xfffe0800 (irq 9, FIFOs size 542)
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.102887] bcm63xx-phone phone_dev_zarlink_init() 674: phone_dev_zarlink_init()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.110573] bcm63xx-phone zarlink_reset() 164: zarlink_reset(gpio_num=0x8018)
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.182561] bcm63xx-phone phone_device_init() 30: phone_device_init()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.189215] bcm63xx-phone zarlink_device_id_init() 458: zarlink_device_id_init()
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.196889] bcm63xx-phone: 1 device(s) provides 2 FXS line(s)
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.202822] bcm63xx-phone bcm_phone_mgr_init() 869: Initializing event wait queue
> Wed May 24 14:44:53 2017 kern.info kernel: [ 7500.210548] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
[... 54 lines cut ...]
> dma_channels[0].dma_flowcl = 0x0
[... cut ...]
> dma_channels[0].dmas_sram3 = 0x0
> dma_channels[0].dmas_sram4 = 0x0
> dma_channels[1].dma_flowcl = 0x0
> dma_channels[1].dma_flowch = 0x0
> dma_channels[1].dma_bufalloc = 0x0
> dma_channels[1].dmac_chancfg = 0x0
> dma_channels[1].dmac_ir = 0x0
> dma_channels[1].dmac_irmask = 0x0
> dma_channels[1].dmac_maxburst = 0x8
> dma_channels[1].dmas_rstart = 0xc74000
> dma_channels[1].dmas_sram2 = 0x0
> dma_channels[1].dmas_sram3 = 0x0
> dma_channels[1].dmas_sram4 = 0x0
> ioctl(2149074188) failed. errno=1
> Error starting
>

pgid69, 2017-05-27:
Hello 
Sorry for the late answer.
I can't reproduce your problem because my router runs OpenWrt 15.05 and i don't want to upgrade to an unstable version of Lede. 
So i need your help.
The errno from both ioctl that failed, is the same and is EPERM : operation not permitted.
I don't understand why.
If you send me the kernel log (the log where appears the bcm63xx-phone messages) before and after launching `bcm63xx-phone-test test=loopback`, perhaps i could have more details about the problem.
Thank you for your help.

ivan1970, 2017-05-29:
thanks for your answer.
this is the kernel log after loading the module (**insmod bcm63xx-phone**):

> [ 4093.717337] bcm63xx-phone: bcm63xx-phone: v0.1
> [ 4093.721977] bcm63xx-phone bcm_drv_init() 1442: bcm_drv_init()
> [ 4093.727926] bcm63xx-phone board_init() 706: board_init()
> [ 4093.733394] bcm63xx-phone: CPU is Broadcom 6358
> [ 4093.738059] bcm63xx-phone: Board is DWV-S0
> [ 4093.742285] bcm63xx-phone board_get_desc() 764: board_get_desc()
> [ 4093.748482] bcm63xx-phone bcm_drv_init() 1457: Initializing mutex
> [ 4093.754762] bcm63xx-phone bcm_drv_init() 1460: Initializing RX wait queue
> [ 4093.761761] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
> [ 4093.768760] bcm63xx-phone bcm_drv_init() 1462: Initializing TX wait queue
> [ 4093.775755] bcm63xx-phone bcm_wait_queue_init() 32: bcm_wait_queue_init()
> [ 4093.782756] bcm63xx-phone bcm_drv_init() 1477: Registering device bcm63xx-phone
> [ 4093.790687] bcm63xx-phone pcm_init() 2482: pcm_init()
> [ 4093.795921] bcm63xx-phone pcm_init() 2550: Using 25 buffers of 256 bytes for RX
> [ 4093.803482] bcm63xx-phone pcm_init() 2586: Using 25 buffers of 256 bytes for TX
> [ 4093.811024] bcm63xx-phone pcm_get_shared_res() 1950: pcm_get_shared_res()
> [ 4093.818029] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm, addr=0xfffe1600, size=256)
> [ 4093.827479] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dma, addr=0xfffe1800, size=256)
> [ 4093.837283] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmac, addr=0xfffe1900, size=32)
> [ 4093.847083] bcm63xx-phone pcm_get_mem_res() 1906: pcm_get_mem_res(id=pcm_dmas, addr=0xfffe1a00, size=32)
> [ 4093.946586] bcm63xx-phone pcm6358_pll_init() 970: pcm6358_pll_init()
> [ 4093.953189] bcm63xx-phone pcm6358_pll_init() 975:  pll_ctrl1=f0010001, pll_ctrl2=53d47, pll_ctrl3=a85
> [ 4093.994632] bcm63xx-phone pcm6358_pll_init() 1028:  pll_ctrl1=20001, pll_ctrl2=8492b, pll_ctrl3=1e1c
> [ 4094.004078] bcm63xx-phone pcm_regs_init() 1247: pcm_regs_init()
> [ 4094.010205] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
> [ 4094.016492] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
> [ 4094.023674] bcm63xx-phone pcm_dma_init() 1502: pcm_dma_init()
> [ 4094.029611] bcm63xx-phone pcm_dma_init() 1535: Allocating 200 bytes of DMA memory for RX descs
> [ 4094.038610] bcm63xx-phone pcm_dma_init() 1546: 200 bytes of memory allocated, starting at address 0xa0daa000
> [ 4094.048752] bcm63xx-phone pcm_dma_init() 1552: Allocating 200 bytes of DMA memory for TX descs
> [ 4094.057725] bcm63xx-phone pcm_dma_init() 1563: 200 bytes of memory allocated, starting at address 0xa0dab000
> [ 4094.067901] bcm63xx-phone pcm_dma_init() 1571: Allocating 12800 bytes of memory for RX and TX data buffers
> [ 4094.077885] bcm63xx-phone pcm_dma_init() 1578: 12800 bytes of memory allocated, starting at address 0x80e54000
> [ 4094.088244] bcm63xx-phone pcm_dma_init() 1604: Allocating 224 bytes of memory for DMA addresses
> [ 4094.097224] bcm63xx-phone pcm_dma_init() 1611: 224 bytes of memory allocated, starting at address 0x80d67500
> [ 4094.107365] bcm63xx-phone pcm_dma_init() 1618: Mapping DMA addresses of RX descs
> [ 4094.115009] bcm63xx-phone pcm_dma_init() 1635: Mapping DMA addresses of TX descs
> [ 4094.122654] bcm63xx-phone pcm_dma_rx_descs_reset() 1342: pcm_dma_rx_descs_reset()
> [ 4094.130455] bcm63xx-phone pcm_dma_tx_descs_reset() 1373: pcm_dma_tx_descs_reset()
> [ 4094.138269] bcm63xx-phone bcm_drv_init() 1527: t->mm_desc.ioctl_param_off = 0, t->mm_desc.ioctl_param_size = 208
> [ 4094.148791] bcm63xx-phone bcm_drv_init() 1537: t->mm_desc.rbs[0].rx_ring_buf_desc_off = 208, t->mm_desc.rbs[0].rx_ring_buf_desc_size = 12
> [ 4094.161539] bcm63xx-phone bcm_drv_init() 1545: t->mm_desc.rbs[0].tx_ring_buf_desc_off = 224, t->mm_desc.rbs[0].tx_ring_buf_desc_size = 12
> [ 4094.174286] bcm63xx-phone bcm_drv_init() 1537: t->mm_desc.rbs[1].rx_ring_buf_desc_off = 240, t->mm_desc.rbs[1].rx_ring_buf_desc_size = 12
> [ 4094.187028] bcm63xx-phone bcm_drv_init() 1545: t->mm_desc.rbs[1].tx_ring_buf_desc_off = 256, t->mm_desc.rbs[1].tx_ring_buf_desc_size = 12
> [ 4094.199884] bcm63xx-phone bcm_drv_init() 1567: 8192 bytes of memory allocated, starting at address 0x80e14000
> [ 4094.210125] bcm63xx-phone bcm_phone_mgr_init() 851: bcm_phone_mgr_init()
> [ 4094.217035] bcm63xx-phone bcm_phone_mgr_devs_init() 725: bcm_phone_mgr_devs_init()
> [ 4094.224842] bcm63xx-phone phone_device_alloc() 55: phone_device_alloc()
> [ 4094.231700] bcm63xx-phone phone_device_alloc() 84: 2020 bytes of memory allocated, starting at address 0x80944800
> [ 4094.242285] bcm63xx-phone phone_dev_le88266_init() 54: phone_dev_le88266_init()
> [ 4094.249821] bcm63xx-phone phone_dev_zarlink_ve880_init() 1948: phone_dev_zarlink_ve880_init()
> [ 4094.258607] bcm63xx-phone bcm_mpi_init() 919: bcm_mpi_init()
> [ 4094.264591] bcm63xx-phone bcm63xx_spi_probe() 412: bcm63xx_spi_probe()
> [ 4094.271439] bcm6358-spi bcm6358-spi: at 0xfffe0800 (irq 9, FIFOs size 542)
> [ 4094.278907] bcm63xx-phone phone_dev_zarlink_init() 674: phone_dev_zarlink_init()
> [ 4094.286589] bcm63xx-phone zarlink_reset() 164: zarlink_reset(gpio_num=0x8018)
> [ 4094.358578] bcm63xx-phone phone_device_init() 30: phone_device_init()
> [ 4094.365236] bcm63xx-phone zarlink_device_id_init() 458: zarlink_device_id_init()
> [ 4094.372906] bcm63xx-phone: 1 device(s) provides 2 FXS line(s)
[... cut ...]
> [ 4371.223612] bcm63xx-phone pcm_pr_stats() 2060: tx_good              = 0
> [ 4371.230429] bcm63xx-phone pcm_pr_stats() 2061: tx_bytes             = 0
> [ 4371.237251] bcm63xx-phone pcm_pr_stats() 2062: cnt_irq_rx           = 0
> [ 4371.244076] bcm63xx-phone pcm_pr_stats() 2063: cnt_irq_tx           = 0
> [ 4371.250897] bcm63xx-phone bcm_phone_mgr_stop() 595: bcm_phone_mgr_stop()
> [ 4371.257812] bcm63xx-phone bcm_timer_stop() 116: bcm_timer_stop()
> [ 4371.264014] bcm63xx-phone phone_dev_zarlink_ve880_stop() 2102: phone_dev_zarlink_ve880_stop()
>

pgid69, 2017-05-29:
As the log shows the problem comes from the function VpInitDebug() that returns 1 : VP_STATUS_FAILURE. 
I think there's a problem with SPI communication but to be sure can you makes the following modifications in the sources :
- In file bcm63xx-phone/src/extern/zarlink/2.24.0.lite/api_lib/includes/vp_api_cfg.h change line 42 from `#undef VP_DEBUG` to `#define VP_DEBUG`.
- In file bcm63xx-phone/src/mpi/mpi.c, remove the prefix dd_ before bcm_pr_debug at lines 287, 688 and 784.
- In file bcm63xx-phone/src/inc/config.h change line 16 from `#undef BCMPH_DEBUG_MPI` to `#define BCMPH_DEBUG_MPI`

Then make a full recompilation, run again the test program and send me the kernel log.

Thank you

ivan1970, 2017-05-29:
I'm sorry but I'm not able to recompile after these changes, could you suggest me why?

```
make -f ./scripts/Makefile.build obj=/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone
  mips-openwrt-linux-musl-gcc -Wp,-MD,/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/mpi/.mpi.o.d  -nostdinc -isystem /data/lede/source/staging_dir/toolchain-mips_mips32_gcc-5.4.0_musl/lib/gcc/mips-openwrt-linux-musl/5.4.0/include -I./arch/mips/include -Iarch/mips/include/generated/uapi -Iarch/mips/include/generated  -Iinclude -I./arch/mips/include/uapi -Iarch/mips/include/generated/uapi -I./include/uapi -Iinclude/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -DVMLINUX_LOAD_ADDRESS=0xffffffff80010000 -DDATAOFFSET=0 -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -mno-check-zero-division -mabi=32 -G 0 -mno-abicalls -fno-pic -pipe -mno-branch-likely -msoft-float -DGAS_HAS_SET_HARDFLOAT -Wa,-msoft-float -ffreestanding -march=mips32 -Wa,-mips32 -Wa,--trap -I./arch/mips/include/asm/mach-bcm63xx/ -I./arch/mips/include/asm/mach-generic -fno-delete-null-pointer-checks -Wno-maybe-uninitialized -Os -fno-caller-saves --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector -Wno-unused-but-set-variable -fomit-frame-pointer -fno-var-tracking-assignments -g -femit-struct-debug-baseonly -fno-var-tracking -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -DCC_HAVE_ASM_GOTO -Wall -DBCMPH_DEBUG -I/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/inc -I/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/extern/zarlink/arch/bcm63xx -I/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/includes -I/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/vp880_api  -DMODULE -mno-long-calls  -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(mpi)"  -D"KBUILD_MODNAME=KBUILD_STR(bcm63xx_phone)" -c -o /data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/mpi/mpi.o /data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/mpi/mpi.c
In file included from /data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/inc/bcm63xx_line_state.h:18:0,
                 from /data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/inc/bcm63xx_phone.h:14,
                 from /data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/inc/board.h:19,
                 from /data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/inc/mpi.h:17,
                 from /data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/mpi/mpi.c:29:
/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/mpi/mpi.c: In function 'bcm_mpi_rw_buf':
/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/mpi/mpi.c:288:75: error: 'prepend_len' undeclared (first use in this function)
       (unsigned int)(buf_len), (int)(do_tx), (int)(do_rx), (unsigned int)(prepend_len));
                                                                           ^
/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/inc/bcm63xx_log.h:26:129: note: in definition of macro 'bcm_pr_debug'
 #define bcm_pr_debug(fmt, args...) printk(KERN_INFO "bcm63xx-phone %s() %lu: " fmt, __FUNCTION__, (unsigned long)(__LINE__), ## args)
                                                                                                                                 ^
/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/mpi/mpi.c:288:75: note: each undeclared identifier is reported only once for each function it appears in
       (unsigned int)(buf_len), (int)(do_tx), (int)(do_rx), (unsigned int)(prepend_len));
                                                                           ^
/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/inc/bcm63xx_log.h:26:129: note: in definition of macro 'bcm_pr_debug'
 #define bcm_pr_debug(fmt, args...) printk(KERN_INFO "bcm63xx-phone %s() %lu: " fmt, __FUNCTION__, (unsigned long)(__LINE__), ## args)
                                                                                                                                 ^
scripts/Makefile.build:258: recipe for target '/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/mpi/mpi.o' failed
make[5]: *** [/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/mpi/mpi.o] Error 1
Makefile:1400: recipe for target '_module_/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone' failed
make[4]: *** [_module_/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone] Error 2
make[4]: Leaving directory '/data/lede/source/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.4.69'
Makefile:46: recipe for target '/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/.built' failed
make[3]: *** [/data/lede/source/build_dir/target-mips_mips32_musl/bcm63xx-phone/.built] Error 2
make[3]: Leaving directory '/data/lede/custom-feed/bcm63xx-phone/bcm63xx-phone'
package/Makefile:109: recipe for target 'package/feeds/custom/bcm63xx-phone/compile' failed
make[2]: *** [package/feeds/custom/bcm63xx-phone/compile] Error 2
make[2]: Leaving directory '/data/lede/source'
package/Makefile:105: recipe for target '/data/lede/source/staging_dir/target-mips_mips32_musl/stamp/.package_compile' failed
make[1]: *** [/data/lede/source/staging_dir/target-mips_mips32_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/data/lede/source'
/data/lede/source/include/toplevel.mk:206: set di istruzioni per l'obiettivo "world" non riuscito
make: *** [world] Errore 2

```

ivan1970, 2017-05-29:
here new logs:

> [28913.673601] bcm63xx-phone bcm_phone_mgr_deinit() 906: bcm_phone_mgr_deinit()
> [28913.680936] bcm63xx-phone bcm_timer_deinit() 154: bcm_timer_deinit()
> [28913.687516] bcm63xx-phone bcm_timer_stop() 116: bcm_timer_stop()
> [28913.693741] bcm63xx-phone bcm_phone_mgr_deinit() 909: Flushing timer workqueue and destroying it
> [28913.702839] bcm63xx-phone bcm_wait_queue_deinit() 39: bcm_wait_queue_deinit()
> [28913.710212] bcm63xx-phone bcm_phone_mgr_devs_deinit() 833: bcm_phone_mgr_devs_deinit()
> [28913.718422] bcm63xx-phone phone_device_free() 107: phone_device_free()
> [28913.725156] bcm63xx-phone le88266_deinit() 24: le88266_deinit()
> [28913.731265] bcm63xx-phone phone_dev_zarlink_ve880_stop() 2102: phone_dev_zarlink_ve880_stop()
> [28913.740070] bcm63xx-phone phone_dev_zarlink_ve880_deinit() 2015: phone_dev_zarlink_ve880_deinit()
> [28913.749223] bcm63xx-phone phone_dev_zarlink_deinit() 699: phone_dev_zarlink_deinit()
> [28913.757212] bcm63xx-phone zarlink_device_id_deinit() 465: zarlink_device_id_deinit()
> [28913.765204] bcm63xx-phone phone_device_deinit() 47: phone_device_deinit()
> [28913.772205] bcm63xx-phone zarlink_reset() 164: zarlink_reset(gpio_num=0x8018)
> [28913.844018] bcm63xx-phone bcm_mpi_deinit() 1013: bcm_mpi_deinit()
> [28913.850376] bcm63xx-phone bcm63xx_spi_remove() 503: bcm63xx_spi_remove()
> [28913.857615] bcm63xx-phone bcm_drv_exit() 1613: Freeing mmap buffer used for RX, TX...
> [28913.865762] bcm63xx-phone pcm_deinit() 2684: pcm_deinit()
> [28913.871344] bcm63xx-phone pcm_stop() 2321: pcm_stop()
> [28913.876555] bcm63xx-phone pcm_dma_stop() 1470: pcm_dma_stop()
> [28913.882484] bcm63xx-phone pcm_stop() 2333: Halting TX DMA
> [28913.898222] bcm63xx-phone pcm_stop() 2340: TX DMA halted
> [28913.903752] bcm63xx-phone pcm_stop() 2355: Halting RX DMA
> [28913.916074] bcm63xx-phone pcm_stop() 2362: RX DMA halted
> [28913.921567] bcm63xx-phone pcm_stop() 2373: Disabling all PCM channels
> [28913.928227] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
> [28913.934519] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
> [28913.941717] bcm63xx-phone pcm_dma_tx_descs_reset() 1373: pcm_dma_tx_descs_reset()
> [28913.949532] bcm63xx-phone pcm_pr_stats() 2053: ======= Stats =======
> [28913.956103] bcm63xx-phone pcm_pr_stats() 2054: rx_errors            = 0
> [28913.962928] bcm63xx-phone pcm_pr_stats() 2055: rx_length_errors     = 0
> [28913.969757] bcm63xx-phone pcm_pr_stats() 2056: rx_empty_errors      = 0
> [28913.976584] bcm63xx-phone pcm_pr_stats() 2057: rx_good              = 0
> [28913.983411] bcm63xx-phone pcm_pr_stats() 2058: rx_bytes             = 0
> [28913.990239] bcm63xx-phone pcm_pr_stats() 2059: tx_errors            = 0
> [28913.997066] bcm63xx-phone pcm_pr_stats() 2060: tx_good              = 0
> [28914.003893] bcm63xx-phone pcm_pr_stats() 2061: tx_bytes             = 0
> [28914.010723] bcm63xx-phone pcm_pr_stats() 2062: cnt_irq_rx           = 0
> [28914.017549] bcm63xx-phone pcm_pr_stats() 2063: cnt_irq_tx           = 0
> [28914.024377] bcm63xx-phone pcm_dma_deinit() 1798: pcm_dma_deinit()
> [28914.030667] bcm63xx-phone pcm_dma_deinit() 1820: Unmapping DMA addresses of TX descs
> [28914.038662] bcm63xx-phone pcm_dma_deinit() 1825: Unmapping DMA addresses of RX descs
> [28914.046783] bcm63xx-phone pcm_dma_deinit() 1872: Freeing memory of RX and TX data buffers
> [28914.055263] bcm63xx-phone pcm_dma_deinit() 1876: Freeing memory for DMA adresses
> [28914.062902] bcm63xx-phone pcm_dma_deinit() 1881: Freeing DMA memory of RX descs
> [28914.070462] bcm63xx-phone pcm_dma_deinit() 1890: Freeing DMA memory of TX descs
> [28914.078023] bcm63xx-phone pcm_regs_deinit() 1259: pcm_regs_deinit()
> [28914.084494] bcm63xx-phone pcm_regs_reset() 1192: pcm_regs_reset()
> [28914.090790] bcm63xx-phone pcm_regs_reset() 1228: PCM_CTRL_REG <- 0x23200101
> [28914.097979] bcm63xx-phone pcm6358_pll_deinit() 1037: pcm6358_pll_deinit()
> [28914.192024] bcm63xx-phone pcm_release_shared_res() 2001: pcm_release_shared_res()
> [28914.199764] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1a00, size=32)
> [28914.209146] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1900, size=32)
> [28914.218502] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1800, size=256)
> [28914.227945] bcm63xx-phone pcm_release_mem_res() 1936: pcm_release_mem_res(addr=0xfffe1600, size=256)
> [28914.237403] bcm63xx-phone bcm_drv_exit() 1616: Unregistering device
> [28914.244245] bcm63xx-phone bcm_wait_queue_deinit() 39: bcm_wait_queue_deinit()
> [28914.251636] bcm63xx-phone bcm_wait_queue_deinit() 39: bcm_wait_queue_deinit()
> [28914.259044] bcm63xx-phone board_deinit() 771: board_deinit()
> [28914.264892] bcm63xx-phone bcm_drv_exit() 1624: bcm63xx-phone removed
> [28942.303871] bcm63xx-phone: bcm63xx-phone: v0.1
> [28942.308522] bcm63xx-phone bcm_drv_init() 1442: bcm_drv_init()
> [28942.314476] bcm63xx-phone board_init() 706: board_init()
> [28942.319955] bcm63xx-phone: CPU is Broadcom 6358
> [28942.324628] bcm63xx-phone: Board is DWV-S0
> [28942.328854] bcm63xx-phone board_get_desc() 764: board_get_desc()
> [28942.335049] bcm63xx-phone bcm_drv_init() 1457: Initializing mutex
> [28942.341333] bcm63xx-phone bcm_drv_init() 1460: Initializing RX wait queue
[... 155 lines cut ...]
> [29083.169446] bcm63xx-phone pcm_stop() 2321: pcm_stop()
> [29083.174677] bcm63xx-phone pcm_dma_stop() 1470: pcm_dma_stop()
[... cut ...]
> [29083.288887] bcm63xx-phone pcm_pr_stats() 2060: tx_good              = 0
> [29083.295713] bcm63xx-phone pcm_pr_stats() 2061: tx_bytes             = 0
> [29083.302543] bcm63xx-phone pcm_pr_stats() 2062: cnt_irq_rx           = 0
> [29083.309369] bcm63xx-phone pcm_pr_stats() 2063: cnt_irq_tx           = 0
> [29083.316199] bcm63xx-phone bcm_phone_mgr_stop() 595: bcm_phone_mgr_stop()
> [29083.323114] bcm63xx-phone bcm_timer_stop() 116: bcm_timer_stop()
> [29083.329324] bcm63xx-phone phone_dev_zarlink_ve880_stop() 2102: phone_dev_zarlink_ve880_stop()
>

pgid69, 2017-05-30:
Hello 
The problem is that the Legerity device (Le88266) is not detected.
But i don't know why.
- The Wiki page of OpenWrt  Pirelli DRG A226m  mentions 8 or 16 MB for flash size and 32 or 64 MB for RAM size so that's means there are several versions of the router.
So can you verify that your router has a Legerity 88266 by opening your device and look for the component ?
- Perhaps it's because the Chip Select of the device is different : it's not 2 but 1 or 0.
To test that possibility, in the file bcm63xx-phone/src/board.c, your could change line 639 from `.cs = 2,` to `.cs = 1,` or `.cs = 0,` to test CS 1 and 0.

Just for information can you send me the bootlog also.

Thank you

ivan1970, 2017-05-30:
hello,
Unfortunately I don't know where I can find the component. My router have 8MB for flash and 32GB for RAM. It is like that one that you can see into wiki: [https://wiki.openwrt.org/toh/pirelli/drg_a226m](url)
I tried settin the .cs to both 1 and 0 but I have always the same issue.
attached you can find the required log:
[bootlog.txt](https://github.com/pgid69/bcm63xx-phone/files/1037765/bootlog.txt)

pgid69, 2017-05-30:
I don't understand what's happen and i don't know what other test i could ask you to run.
Perhaps you should try an older version of OpenWrt (and because LEDE prerequisites are 64 MB of RAM), as it is reported to work with versions 12.09 and 14.07 (https://github.com/pgid69/bcm63xx-phone/issues/5). My router (Huawei HG553) runs OpenWrt 15.05.1.
Sorry i can't help you more.

closed 2017-12-27 by pgid69