pgid69/bcm63xx-phone #15: Huawei HG556A works fine

YPKEH, 2017-07-24:
OpenWRT Barrier Breaker 14.07
HG55VDFA VER.C Router version B (see https://wiki.openwrt.org/toh/huawei/hg556a)

Patched for disable SPI - **its very important**.:
-CONFIG_SPI_BCM63XX=y
+CONFIG_SPI_BCM63XX=n

You have to load bcm63xx-phone.ko **before start** asterisk:
my rc.local starting script:

insmod bcm63xx-phone.ko
sleep 10
/etc/init.d/asterisk start
exit 0

Asterisk 11, excelent voice quality.

**Thanks for pgid69**

pgid69, 2017-07-24:
Thank you very much for your feedback.

closed 2017-07-24 by pgid69

talnacialex, 2017-09-07:
I have to say that it also works on my router. I finally managed to compile it. Works very good.

crdlight, 2018-02-16:
I have a Huawei HG556A can anyone share the config files to enable fxs on it.
thank you