pgid69/bcm63xx-phone #16: Possible full support for Voice FXS Chip?

talnacialex, 2017-09-07:
Hello! I recently found the firmware of the Vodafone Huawei Echo-Life Hg556 Leaked on github. It has some C files out there waiting for exploration. I found a couple of them that gives me confidence that there could be the zarlink drivers. Here is the repository. https://github.com/cubieb/hg556a_source Thank you and have a good day! Au revoir!

talnacialex, 2017-09-07:
Also I found this: https://github.com/ahmadrasyidsalim/Huawei-EchoLife-HG556a which looks like a recent version

pgid69, 2017-09-07:
Hello

What do you mean exactly when you say "Possible full support for Voice FXS Chip?" ?
Else thank you for these links. 
I quickly browse the files and found no source related to zarlink driver. 
In which directory do you found something interesting ?
Anyway bcm63xx-phone driver is already able to control the Zarlink chip.
What is really missing is the echo canceller and handling of caller id but that's can be done in software and so is independant of hardware.
I recently discovered that Asterisk can generate sound samples that encode caller id.
So what i really need is time to complete bcm63xx-phone driver and/or Asterisk channel.

BR
Best regards

talnacialex, 2017-09-11:
Hello! This file looks interesting: https://github.com/ahmadrasyidsalim/Huawei-EchoLife-HG556a/blob/master/bcmdrivers/broadcom/include/bcm963xx/si3215drv.h

Says about something RINGING

And si3215 is the FXS Chip.

talnacialex, 2017-09-11:
And also this file looks interesting: https://github.com/ahmadrasyidsalim/Huawei-EchoLife-HG556a/tree/master/userapps/broadcom/tr69c/inc

pgid69, 2017-09-11:
Hello

I think the file https://github.com/ahmadrasyidsalim/Huawei-EchoLife-HG556a/blob/master/bcmdrivers/broadcom/include/bcm963xx/si3215drv.h is related to chip SiLabs SI3215, not any Zarlink/Microsemi chip.
And folder https://github.com/ahmadrasyidsalim/Huawei-EchoLife-HG556a/tree/master/userapps/broadcom/tr69c/inc contains files about TR069 implementation https://en.wikipedia.org/wiki/TR-069.
So sorry but nothing useful.
Thanks

talnacialex, 2017-09-11:
When I searched the SI3215 datasheet, it said that it's a FXS Chip(like the zarlink and microsemi ones).

talnacialex, 2017-09-11:
I am waiting for possible Caller ID Implementation. If you can do this, I will Use my HW556 Spain everyday as a PBX. I used it already as a application server, but it's time for upgrades. Looking forward to hear from you.

Regards,
Talnaci Alexandru. Au revoir!

pgid69, 2017-12-27:
I finally implement Caller ID, so i close the issue. Reopen another one if you have problem.

Gilles

closed 2017-12-27 by pgid69