pgid69/bcm63xx-phone #17: Errors...

talnacialex, 2017-12-28:
Hi. I am back. I'm trying to compile on LEDE and seems that I get a error related to spi. Here it is:

make[3]: Entering directory '/home/alex/Desktop/openwrt-build/lede/feeds/fxs/bcm63xx-phone/bcm63xx-phone'
make -C "/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/linux-4.4.107" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" SUBDIRS="/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone" BCMPH_EFLAGS="-DBCMPH_MODULE_NAME=bcm63xx_phone -DBCMPH_MODULE_VERSION=0.2 -DBCMPH_EXPORT_DEV_FILE -DBCMPH_DEBUG" modules V=1
make[4]: Entering directory '/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/linux-4.4.107'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/.tmp_versions ; rm -f /home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/.tmp_versions/*
make -f ./scripts/Makefile.build obj=/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone
  mips-openwrt-linux-musl-gcc -Wp,-MD,/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/board/.board.o.d  -nostdinc -isystem /home/alex/Desktop/openwrt-build/lede/staging_dir/toolchain-mips_mips32_gcc-5.4.0_musl-1.1.16/lib/gcc/mips-openwrt-linux-musl/5.4.0/include -I./arch/mips/include -Iarch/mips/include/generated/uapi -Iarch/mips/include/generated  -Iinclude -I./arch/mips/include/uapi -Iarch/mips/include/generated/uapi -I./include/uapi -Iinclude/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -DVMLINUX_LOAD_ADDRESS=0xffffffff80010000 -DDATAOFFSET=0 -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -mno-check-zero-division -mabi=32 -G 0 -mno-abicalls -fno-pic -pipe -mno-branch-likely -msoft-float -DGAS_HAS_SET_HARDFLOAT -Wa,-msoft-float -ffreestanding -march=mips32 -Wa,-mips32 -Wa,--trap -I./arch/mips/include/asm/mach-bcm63xx/ -I./arch/mips/include/asm/mach-generic -fno-delete-null-pointer-checks -Wno-maybe-uninitialized -Os -fno-caller-saves --param=allow-store-data-races=0 -DCC_HAVE_ASM_GOTO -Wframe-larger-than=1024 -fstack-protector -Wno-unused-but-set-variable -fomit-frame-pointer -fno-var-tracking-assignments -g -femit-struct-debug-baseonly -fno-var-tracking -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -Wall -I/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/inc -DBCMPH_MODULE_NAME=bcm63xx_phone -DBCMPH_MODULE_VERSION=0.2 -DBCMPH_EXPORT_DEV_FILE -DBCMPH_DEBUG -I/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/arch/bcm63xx -I/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/includes -I/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/vp880_api  -DMODULE -mno-long-calls  -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(board)"  -D"KBUILD_MODNAME=KBUILD_STR(bcm63xx_phone)" -c -o /home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/board/board.o /home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/board/board.c
In file included from /home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/arch/bcm63xx/vp_api_types.h:12:0,
                 from /home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/includes/vp_api.h:43,
                 from /home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/board/board.c:31:
/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/inc/mpi.h:17:42: fatal error: ../drivers/spi/spi-bcm63xx.h: No such file or directory
compilation terminated.
scripts/Makefile.build:258: recipe for target '/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/board/board.o' failed
make[5]: *** [/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/board/board.o] Error 1
Makefile:1405: recipe for target '_module_/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone' failed
make[4]: *** [_module_/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone] Error 2
make[4]: Leaving directory '/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/linux-4.4.107'
Makefile:49: recipe for target '/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/.built' failed
make[3]: *** [/home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/.built] Error 2
make[3]: Leaving directory '/home/alex/Desktop/openwrt-build/lede/feeds/fxs/bcm63xx-phone/bcm63xx-phone'
package/Makefile:105: recipe for target 'package/feeds/fxs/bcm63xx-phone/compile' failed
make[2]: *** [package/feeds/fxs/bcm63xx-phone/compile] Error 2
make[2]: Leaving directory '/home/alex/Desktop/openwrt-build/lede'
package/Makefile:101: recipe for target '/home/alex/Desktop/openwrt-build/lede/staging_dir/target-mips_mips32_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/alex/Desktop/openwrt-build/lede/staging_dir/target-mips_mips32_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/alex/Desktop/openwrt-build/lede'
/home/alex/Desktop/openwrt-build/lede/include/toplevel.mk:205: recipe for target 'world' failed
make: *** [world] Error 2

I applied the provided patches. I'm using LEDE 17.01 STABLE. Where could be wrong?
Compiling Environment: Ubuntu 17.10(64-bit), using 4gb RAM in VMWare Virtual Machine on a i5-2450M.
It seems that it can't find /home/alex/Desktop/openwrt-build/lede/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_generic/bcm63xx-phone/drivers/spi/spi-bcm63xx.h
SPI is disabled.

talnacialex, 2017-12-28:
So... I think I found a workaround... 

I edited the inc/mpi.h:
Line 17: 
Before: ../drivers/spi/spi-bcm63xx.h
After: ../../linux-4.4.107/drivers/spi/spi-bcm63xx.h

This fixed my error...

pgid69, 2017-12-28:
Hi

I've just pushed the correction.

BR

talnacialex, 2017-12-28:
Wow... that was fast! Let me try to compile it again...

Best regards,
Alex.

talnacialex, 2017-12-28:
Kmod-echo it's used for echo canceller function, am I right?

pgid69, 2017-12-28:
Yes, you're right.

pgid69, 2017-12-30:
Hi, do you succeed compiling the driver ? Can i close the issue ?

talnacialex, 2017-12-30:
Hello. Yes, now it compiles fine. Need to build now asterisk. You can close
the issue.

closed 2017-12-30 by pgid69