pgid69/bcm63xx-phone #18: Enable fxs on hg556a

crdlight, 2018-02-20:
I have a Huawei HG556A can anyone share the config files to enable fxs on it.
thank you

closed 2018-03-04 by pgid69

crdlight, 2018-03-05:
Yes config files for LEDE firmware I've installed Asterisk

pgid69, 2018-03-07:
Hello,

I emphasize the fact that bcm63xx-phone kernel module can be used only with a custom firmware because of a conflict with the default SPI kernel module built-in in the standard firmware (see README and issue #20)

My router is a Huawei HG553. As it is different than yours, I can't give you my config files used to compile the firmware.
The build system of LEDE/OpenWrt is explained [here](https://openwrt.org/docs/guide-developer/build-system/start)
Follow the steps of this [page](https://openwrt.org/docs/guide-developer/build-system/use-buildsystem).

Before step 2 "Updating Feeds" :
- You must apply patch `<dir>/bcm63xx-phone/bcm63xx-phone/patches/17.01/000-disable-spi-bcm63xx-module.patch` that disable spi driver (you must adapt dir to the path where you clone the project bcm63xx-phone).
- You have to create file feeds.conf : just copy feeds.conf.default, enable the line `src-git telephony ...` and add the line `src-link custom <dir>/bcm63xx-phone/` .

At step 3, image configuration, when you run `make menuconfig` for the second time, you must at least enable package asterisk 13 (menu Network/Telephony) and package asterisk13-chan-bcm63xx-phone.

Then follow the others instructions to build your own firmware.

BR