pgid69/bcm63xx-phone #19: compiling for openwrt 15.05.1 fails

openwr, 2018-02-28:
Hello, I have this error, any suggestion?

mkdir -p /opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/.tmp_versions ; rm -f /opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/.tmp_versions/*
make -f ./scripts/Makefile.build obj=/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone
  mips-openwrt-linux-uclibc-gcc -Wp,-MD,/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/mpi/.mpi.o.d  -nostdinc -isystem /opt/chaos_calmer/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/lib/gcc/mips-openwrt-linux-uclibc/4.8.3/include -I./arch/mips/include -Iarch/mips/include/generated  -Iinclude -I./arch/mips/include/uapi -Iarch/mips/include/generated/uapi -I./include/uapi -Iinclude/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -DVMLINUX_LOAD_ADDRESS=0xffffffff80010000 -DDATAOFFSET=0 -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -mno-check-zero-division -mabi=32 -G 0 -mno-abicalls -fno-pic -pipe -mno-branch-likely -msoft-float -DGAS_HAS_SET_HARDFLOAT -Wa,-msoft-float -ffreestanding -march=mips32 -Wa,-mips32 -Wa,--trap -I./arch/mips/include/asm/mach-bcm63xx/ -I./arch/mips/include/asm/mach-generic -fno-delete-null-pointer-checks -fno-PIE -Os -fno-caller-saves -Wno-maybe-uninitialized --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fno-stack-protector -Wno-unused-but-set-variable -fomit-frame-pointer -fno-var-tracking-assignments -g -femit-struct-debug-baseonly -fno-var-tracking -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -DCC_HAVE_ASM_GOTO -Wall -I/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/inc -DBCMPH_MODULE_NAME=bcm63xx_phone -DBCMPH_MODULE_VERSION=0.2 -DBCMPH_EXPORT_DEV_FILE -DBCMPH_DEBUG -I/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/arch/bcm63xx -I/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/includes -I/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/vp880_api  -DMODULE -mno-long-calls  -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(mpi)"  -D"KBUILD_MODNAME=KBUILD_STR(bcm63xx_phone)" -c -o /opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/mpi/mpi.o /opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/mpi/mpi.c
/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/mpi/mpi.c:566:7: error: unknown field 'probe_type' specified in initializer
       .probe_type = PROBE_FORCE_SYNCHRONOUS,
       ^
/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/mpi/mpi.c:566:21: error: 'PROBE_FORCE_SYNCHRONOUS' undeclared here (not in a function)
       .probe_type = PROBE_FORCE_SYNCHRONOUS,
                     ^
make[5]: *** [/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/mpi/mpi.o] Error 1
make[4]: *** [_module_/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone] Error 2
make[4]: Leaving directory `/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/linux-3.18.92'
make[3]: *** [/opt/chaos_calmer/build_dir/target-mips_mips32_uClibc-0.9.33.2/linux-brcm63xx_generic/bcm63xx-phone/.built] Error 2
make[3]: Leaving directory `/opt/custom-feed/bcm63xx-phone/bcm63xx-phone/bcm63xx-phone'
make[2]: *** [package/feeds/customfeed/bcm63xx-phone/compile] Error 2
make[2]: Leaving directory `/opt/chaos_calmer'
make[1]: *** [/opt/chaos_calmer/staging_dir/target-mips_mips32_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/opt/chaos_calmer'
make: *** [world] Errore 2

pgid69, 2018-03-01:
Hello

I've just made a commit which fixes the problem.

BR

openwr, 2018-03-01:
Hello, now it works! thank you.

openwr, 2018-03-02:
I am using kernel 3.18.92 (the last in 15.05.1 and I have seg fault when I load the phone module

[  426.268000] bcm63xx_phone phone_dev_le88221_init() 56: phone_dev_le88221_init()
[  426.276000] bcm63xx_phone phone_dev_zarlink_ve880_init() 2017: phone_dev_zarlink_ve880_init()
[  426.284000] bcm63xx_phone bcm_mpi_init() 868: bcm_mpi_init()
[  426.288000] bcm63xx_phone bcm63xx_spi_probe() 414: bcm63xx_spi_probe()
[  426.296000] CPU 0 Unable to handle kernel paging request at virtual address 00000014, epc == 83380e40, ra == 83380e3c
[  426.308000] Oops[#1]:
[  426.308000] CPU: 0 PID: 926 Comm: insmod Not tainted 3.18.92 #3
[  426.308000] task: 83b0ea48 ti: 83282000 task.ti: 83282000
[  426.308000] $ 0   : 00000000 00000001 00000000 00000000
[  426.308000] $ 4   : 803179a4 803179a4 80317c60 801983a8
[  426.308000] $ 8   : 00000029 3a206263 6d363378 785f7370
[  426.308000] $12   : 00000000 0000000e 00000000 7465642c
[  426.308000] $16   : 8031aac8 833a7eb4 8031aac8 8031aab8
[  426.308000] $20   : 00000003 804a0000 0000000a 833a3618
[  426.308000] $24   : 00000001 00000000
[  426.308000] $28   : 83282000 83283b08 833a2b1c 83380e3c
[  426.308000] Hi    : 00000000
[  426.308000] Lo    : 00000095
[  426.308000] epc   : 83380e40 0x83380e40 [bcm63xx_phone@83380000+0x28bf0]
[  426.308000]     Not tainted
[  426.308000] ra    : 83380e3c 0x83380e3c [bcm63xx_phone@83380000+0x28bf0]
[  426.308000] Status: 10008c03 KERNEL EXL IE
[  426.308000] Cause : 00800008
[  426.308000] BadVA : 00000014
[  426.308000] PrId  : 0002a010 (Broadcom BMIPS4350)
[  426.308000] Modules linked in: bcm63xx_phone(+) echo pppoe ppp_async iptable_nat b43 pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mac80211 ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ipv6 arc4 crypto_blkcipher gpio_button_hotplug [last unloaded: echo]
[  426.308000] Process insmod (pid: 926, threadinfo=83282000, task=83b0ea48, tls=771ae440)
[  426.308000] Stack : 802dc3dc 8339dfe4 0000019e 8339dfe4 0000000a 833a3618 833a2b1c 80103b30
[  426.308000]    804a0000 8031aac8 833a7eb4 833a7eb4 804a0000 00000003 804a0000 0000000a
[  426.308000]    833a3618 801a32f8 83aca180 00000000 8380c038 8010385c 8031aac8 8031aac8
[  426.308000]    8031aafc 801a1c14 00000000 8015a7bc 83aca180 00000000 83aca180 8031aac8
[  426.308000]    8031aafc 833a7eb4 8032c190 833a0000 0000000c 801a1e40 833a0000 801a00a8
[  426.308000]    ...
[  426.308000] Call Trace:[<80103b30>] 0x80103b30
[  426.308000] [<801a32f8>] 0x801a32f8
[  426.308000] [<8010385c>] 0x8010385c
[  426.308000] [<801a1c14>] 0x801a1c14
[  426.308000] [<8015a7bc>] 0x8015a7bc
[  426.308000] [<801a1e40>] 0x801a1e40
[  426.308000] [<801a00a8>] 0x801a00a8
[  426.308000] [<801a1dc4>] 0x801a1dc4
[  426.308000] [<801a0160>] 0x801a0160
[  426.308000] [<802aff10>] 0x802aff10
[  426.308000] [<8019ff38>] 0x8019ff38
[  426.308000] [<801a1348>] 0x801a1348
[  426.308000] [<801a230c>] 0x801a230c
[  426.308000] [<833ac614>] 0x833ac614 [bcm63xx_phone@83380000+0x28bf0]
[  426.308000] [<833ae400>] 0x833ae400 [bcm63xx_phone@83380000+0x28bf0]
[  426.308000] [<833addcc>] 0x833addcc [bcm63xx_phone@83380000+0x28bf0]
[  426.308000] [<800550f8>] 0x800550f8
[  426.308000] [<833ad5d4>] 0x833ad5d4 [bcm63xx_phone@83380000+0x28bf0]
[  426.308000] [<833ae490>] 0x833ae490 [bcm63xx_phone@83380000+0x28bf0]
[  426.308000] [<833ad940>] 0x833ad940 [bcm63xx_phone@83380000+0x28bf0]
[  426.308000] [<800550f8>] 0x800550f8
[  426.308000] [<833ae8c0>] 0x833ae8c0 [bcm63xx_phone@83380000+0x28bf0]
[  426.308000] [<83398e3c>] 0x83398e3c [bcm63xx_phone@83380000+0x28bf0]
[  426.308000] [<833ae5d4>] 0x833ae5d4 [bcm63xx_phone@83380000+0x28bf0]
[  426.308000] [<800199e4>] 0x800199e4
[  426.308000] [<8007e568>] 0x8007e568
[  426.308000] [<800a3dc4>] 0x800a3dc4
[  426.308000] [<800a3ecc>] 0x800a3ecc
[  426.308000] [<80070848>] 0x80070848
[  426.308000] [<800a3b78>] 0x800a3b78
[  426.308000] [<8006e614>] 0x8006e614
[  426.308000] [<8002cd80>] 0x8002cd80
[  426.308000] [<80070b5c>] 0x80070b5c
[  426.308000] [<80070bd0>] 0x80070bd0
[  426.308000] [<80012e48>] 0x80012e48
[  426.308000]
[  426.308000]
[  426.308000] Code: 0c015430  afb00024  8e6200e0 <8c510014> 122000a2  3c14833b  268481a0  00002821  0c005470
[  426.680000] ---[ end trace 7aa674cfb7585174 ]---

openwr, 2018-03-02:
any suggestion?

pgid69, 2018-03-02:
Hello

I've just made a commit which should fix the problem.
I can't be sure because my router is running LEDE 17.01 so i can't test myself.
Thank you for your feedback.

BR

openwr, 2018-03-02:
Wow! now it works! thank you.

closed 2018-03-02 by pgid69