pgid69/bcm63xx-phone #22: Few questions - thanks in advance

kofec, 2018-03-19:
First of all - thanks a lot for your effort. I have Huawei EchoLife HG553. I have read your README and closed issues. I have few question.
1) As mention " it contains two kernel drivers " We should decide which one we will used and compile and install only one ? If we install e.g dahdi and we want to try bcm63xx-phone first we should uninstall kmod-bcm63xx-phone-dahdi and install kmod-bcm63xx-phone ? Am I right?
2) Regarding patches. Is it for speed optimization only "in a way more efficient" ? Or it is also needed for proper way of driver working? Is it applicable for both drivers? 
It was compile fine for snapshot openwrt (with patches):
```
kmod-bcm63xx-phone - 4.9.86+0.2-1
kmod-bcm63xx-phone-dahdi - 4.9.86+0.2-1
asterisk13-chan-bcm63xx-phone - 4.9.86-4.9.86
kmod-dahdi - 4.9.86+2.11.1-20180111-2
kmod-dahdi-dummy - 4.9.86+2.11.1-20180111-2
kmod-dahdi-echocan-oslec - 4.9.86+2.11.1-20180111-2
kmod-echo - 4.9.86-1
```
3) Do you have any simplest config files for both configurations - both drivers ? Which one you are using ?
I mean all files need for asterisk to make call between soft sip client and FXS lines. Maybe somewhere in the forum or any other link ?

4) As I understand FXO port is not usable and there is no driver to use it in asterisk ?

5) And last question not full related but as you have this hardware. Have you heard about "SMP" working version ? Maybe someone is working on it - to enable second core?
https://downloads.openwrt.org/snapshots/targets/brcm63xx/smp/

thanks in advance for your answers

pgid69, 2018-03-20:
Hi

Q1: Yes you are right, both driver uses the same resources so they can't be loaded at the same time
Q2: Standard SPI driver is very slow. With the last version of kmod-bcm63xx-phone and kmod-bcm63xx-phone-dahdi, it is no more possible to use it. 
At least one patch must be applied.
The patch 000-disable-spi-bcm63xx-module.patch
 changes the kernel configuration and must be applied if symbol BCMPH_USE_SPI_DRIVER in file inc/config.h, is undefined. 
The other patch 001-add-fn-to-spi-bcm63xx-for-quick-transfer.patch
 must be used if symbol BCMPH_USE_SPI_DRIVER in file inc/config.h, is defined.
You can apply both patches or only one of them.
Whatever your choice you have to recompile the kernel and flash the firmware (that contains the kernel)
Patches are required for both kmod-bcm63xx-phone and kmod-bcm63xx-phone-dahdi 
Q3: As i'm not an expert of Asterisk, i prefer not to provides config files. If you use kmod-bcm63xx-phone the example in README should be a good start.
Personnaly I use kmod-bcm63xx-phone because with dahdi caller id does not work with my phone (dahdi handles only Bell signaling with FXS lines)
Q4: Yes FXO ports are not handled. I don't intend to support FXO because i can't test (HG553 does not have any FXO port)
Q5: Yes i know that only one core is active with bcm6358. I've read that enabling the second core is [tricky](https://wiki.openwrt.org/doc/hardware/soc/soc.broadcom.bcm63xx/smp) and beyond my skills.

BR

kofec, 2018-03-20:
Thanks a lot for detailed answer. Regarding FXO port I think about ADSL port:
http://skupujemykomputery.pl/jucom24/blog/wp-content/uploads/2013/12/tyl2.png

pgid69, 2018-03-21:
ADSL port is not an FXO port !
And as there's no public documentation about Broadcom ADSL devices, it is a very difficult taks to make a driver for it !

closed 2018-03-21 by pgid69