pgid69/bcm63xx-phone #23: latest kernel (4.4 and 4.9) doesn't have fatal error: ../drivers/spi/spi-bcm63xx.h

kofec, 2018-03-20:
```
make -f ./scripts/Makefile.build obj=/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone
  mips-openwrt-linux-musl-gcc -Wp,-MD,/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/board/.board.o.d  -nostdinc -isystem /mnt/openwrt/OpenwrtHG553/staging_dir/toolchain-mips_mips32_gcc-7.3.0_musl/lib/gcc/mips-openwrt-linux-musl/7.3.0/include -I./arch/mips/include -Iarch/mips/include/generated/uapi -Iarch/mips/include/generated  -Iinclude -I./arch/mips/include/uapi -Iarch/mips/include/generated/uapi -I./include/uapi -Iinclude/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -DVMLINUX_LOAD_ADDRESS=0xffffffff80010000 -DDATAOFFSET=0 -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -mno-check-zero-division -mabi=32 -G 0 -mno-abicalls -fno-pic -pipe -mno-branch-likely -msoft-float -DGAS_HAS_SET_HARDFLOAT -Wa,-msoft-float -ffreestanding -march=mips32 -Wa,-mips32 -Wa,--trap -I./arch/mips/include/asm/mach-bcm63xx/ -I./arch/mips/include/asm/mach-generic -fno-delete-null-pointer-checks -Wno-maybe-uninitialized -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-int-in-bool-context -Os -fno-caller-saves --param=allow-store-data-races=0 -DCC_HAVE_ASM_GOTO -Wframe-larger-than=1024 -fstack-protector -Wno-unused-but-set-variable -Wno-unused-const-variable -fomit-frame-pointer -fno-var-tracking-assignments -g -femit-struct-debug-baseonly -fno-var-tracking -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -Wall -I/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/inc -DBCMPH_MODULE_NAME=bcm63xx_phone -DBCMPH_MODULE_VERSION=0.2 -DBCMPH_EXPORT_DEV_FILE -DBCMPH_DEBUG -I/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/arch/bcm63xx -I/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/includes -I/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/vp880_api  -DMODULE -mno-long-calls  -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(board)"  -D"KBUILD_MODNAME=KBUILD_STR(bcm63xx_phone)" -c -o /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/board/board.o /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/board/board.c
In file included from /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/arch/bcm63xx/vp_api_types.h:12:0,
                 from /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/includes/vp_api.h:43,
                 from /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/board/board.c:31:
/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/inc/mpi.h:18:13: fatal error: ../drivers/spi/spi-bcm63xx.h: No such file or directory
 #   include <../drivers/spi/spi-bcm63xx.h>
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
scripts/Makefile.build:269: recipe for target '/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/board/board.o' failed
make[5]: *** [/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/board/board.o] Error 1
Makefile:1408: recipe for target '_module_/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone' failed
make[4]: *** [_module_/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone] Error 2
make[4]: Leaving directory '/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.4.121'
Makefile:49: recipe for target '/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/.built' failed
make[3]: *** [/mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/.built] Error 2
make[3]: Leaving directory '/mnt/openwrt/OpenwrtHG553/custom-feed/bcm63xx-phone/bcm63xx-phone/bcm63xx-phone'
package/Makefile:107: recipe for target 'package/feeds/custom/bcm63xx-phone/compile' failed
make[2]: *** [package/feeds/custom/bcm63xx-phone/compile] Error 2
make[2]: Leaving directory '/mnt/openwrt/OpenwrtHG553'
package/Makefile:103: recipe for target '/mnt/openwrt/OpenwrtHG553/staging_dir/target-mips_mips32_musl/stamp/.package_compile' failed
make[1]: *** [/mnt/openwrt/OpenwrtHG553/staging_dir/target-mips_mips32_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/openwrt/OpenwrtHG553'
/mnt/openwrt/OpenwrtHG553/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

```

kofec, 2018-03-20:
I think I know. The pach was not executed:
https://github.com/pgid69/bcm63xx-phone/blob/master/bcm63xx-phone/patches/17.01/001-add-fn-to-spi-bcm63xx-for-quick-transfer.patch

kofec, 2018-03-21:
Some notes what is wrong:
Applying patch platform/001-add-fn-to-spi-bcm63xx-for-quick-transfer.patch
patching file target/linux/brcm63xx/patches-4.4/900-add_fn_to_spi_bcm63xx_for_quick_transfer.patch

generic/862-gpio_spi_driver.patch
generic/890-uart_optional_sysrq.patch
generic/901-debloat_sock_diag.patch
generic/902-debloat_proc.patch
generic/904-debloat_dma_buf.patch
generic/910-kobject_uevent.patch
generic/911-kobject_add_broadcast_uevent.p

kofec@E5420Mint /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.4.121 $ ls -al target/linux/brcm63xx/patches-4.4/900-add_fn_to_spi_bcm63xx_for_quick_transfer.patch
-rw-rw-r-- 1 kofec kofec 11371 Mar 21 09:16 target/linux/brcm63xx/patches-4.4/900-add_fn_to_spi_bcm63xx_for_quick_transfer.patch
kofec@E5420Mint /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.4.121 $ cat target/linux/brcm63xx/patches-4.4/900-add_fn_to_spi_bcm63xx_for_quick_transfer.patch
--- a/drivers/spi/spi-bcm63xx.c
+++ b/drivers/spi/spi-bcm63xx.c
@@ -26,6 +26,7 @@
 #include <linux/completion.h>
 #include <linux/err.h>
 #include <linux/pm_runtime.h>
+#include "spi-bcm63xx.h"

cd build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.4.121/

platform/804-bcm63xx_enet_63268_rgmii_ports.patch
platform/900-add_fn_to_spi_bcm63xx_for_quick_transfer.patch

Applying patch platform/900-add_fn_to_spi_bcm63xx_for_quick_transfer.patch
patching file drivers/spi/spi-bcm63xx.c
Hunk #1 FAILED at 26.
Hunk #2 succeeded at 137 (offset 1 line).
Hunk #3 succeeded at 156 (offset 1 line).
Hunk #4 succeeded at 204 (offset 1 line).
Hunk #5 succeeded at 282 (offset 1 line).
Hunk #6 succeeded at 293 (offset 1 line).
Hunk #7 succeeded at 311 (offset 1 line).
Hunk #8 succeeded at 333 (offset 1 line).
Hunk #9 succeeded at 369 (offset 1 line).
Hunk #10 succeeded at 384 (offset 1 line).
Hunk #11 succeeded at 407 (offset 1 line).
Hunk #12 succeeded at 446 (offset 1 line).
Hunk #13 succeeded at 470 (offset 1 line).
Hunk #14 succeeded at 498 (offset 1 line).
Hunk #15 succeeded at 668 (offset 38 lines).
1 out of 15 hunks FAILED -- rejects in file drivers/spi/spi-bcm63xx.c
patching file drivers/spi/spi-bcm63xx.h
Patch platform/900-add_fn_to_spi_bcm63xx_for_quick_transfer.patch does not apply (enforce with -f)

kofec@E5420Mint /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.4.121 $ cat drivers/spi/spi-bcm63xx.c.rej
--- drivers/spi/spi-bcm63xx.c
+++ drivers/spi/spi-bcm63xx.c
@@ -26,6 +26,7 @@
 #include <linux/completion.h>
 #include <linux/err.h>
 #include <linux/pm_runtime.h>
+#include "spi-bcm63xx.h"

 /* BCM 6338/6348 SPI core */
 #define SPI_6348_RSET_SIZE             64

kofec, 2018-03-21:
But it didn't compile:
`  CC [M]  drivers/spi/spi-bcm63xx.o
In file included from drivers/spi/spi-bcm63xx.c:30:0:
drivers/spi/spi-bcm63xx.h:1:0: error: unterminated #ifndef
 #ifndef SPI_BCM63XX_H

scripts/Makefile.build:275: recipe for target 'drivers/spi/spi-bcm63xx.o' failed
make[7]: *** [drivers/spi/spi-bcm63xx.o] Error 1
scripts/Makefile.build:476: recipe for target 'drivers/spi' failed
make[6]: *** [drivers/spi] Error 2
`

kofec, 2018-03-21:
For some reason it was not added: ++#endif /* SPI_BCM63XX_H */

kofec, 2018-03-21:
for kernel 4.9 we have:
```
kofec@E5420Mint /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.9.87 $ quilt push -f
Applying patch platform/900-add_fn_to_spi_bcm63xx_for_quick_transfer.patch
patching file drivers/spi/spi-bcm63xx.c
Hunk #4 FAILED at 205.
Hunk #5 succeeded at 253 (offset -1 lines).
Hunk #6 succeeded at 264 (offset -1 lines).
Hunk #7 succeeded at 282 (offset -1 lines).
Hunk #8 succeeded at 304 (offset -1 lines).
Hunk #9 succeeded at 340 (offset -1 lines).
Hunk #10 succeeded at 355 (offset -1 lines).
Hunk #11 succeeded at 378 (offset -1 lines).
Hunk #12 succeeded at 417 (offset -1 lines).
Hunk #13 succeeded at 441 (offset -1 lines).
Hunk #14 succeeded at 469 (offset -1 lines).
Hunk #15 succeeded at 639 (offset -1 lines).
1 out of 15 hunks FAILED -- saving rejects to file drivers/spi/spi-bcm63xx.c.rej
patching file drivers/spi/spi-bcm63xx.h
Applied patch platform/900-add_fn_to_spi_bcm63xx_for_quick_transfer.patch (forced; needs refresh)
kofec@E5420Mint /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.9.87 $ cat drivers/spi/spi-bcm63xx.c.rej
--- drivers/spi/spi-bcm63xx.c
+++ drivers/spi/spi-bcm63xx.c
@@ -205,40 +209,69 @@ static const unsigned bcm63xx_spi_freq_t
        {   391000, SPI_CLK_0_391MHZ }
 };

-static void bcm63xx_spi_setup_transfer(struct spi_device *spi,
-                                     struct spi_transfer *t)
+static inline u8 bcm63xx_spi_set_clk_cfg(struct bcm63xx_spi *bs,
+       u32 speed_hz, u8 cs_off_clk_cycles)
 {
-       struct bcm63xx_spi *bs = spi_master_get_devdata(spi->master);
        u8 clk_cfg, reg;
        int i;

        /* Find the closest clock configuration */
-       for (i = 0; i < SPI_CLK_MASK; i++) {
-               if (t->speed_hz >= bcm63xx_spi_freq_table[i][0]) {
+       for (i = 0; (i < ARRAY_SIZE(bcm63xx_spi_freq_table)); i++) {
+               if (speed_hz >= bcm63xx_spi_freq_table[i][0]) {
                        clk_cfg = bcm63xx_spi_freq_table[i][1];
                        break;
                }
        }

        /* No matching configuration found, default to lowest */
-       if (i == SPI_CLK_MASK)
+       if (i >= ARRAY_SIZE(bcm63xx_spi_freq_table)) {
                clk_cfg = SPI_CLK_0_391MHZ;
+       }
+
+       clk_cfg |= (cs_off_clk_cycles << SPI_SSOFFTIME_SHIFT);

        /* clear existing clock configuration bits of the register */
        reg = bcm_spi_readb(bs, SPI_CLK_CFG);
-       reg &= ~SPI_CLK_MASK;
+       reg &= ~(SPI_CLK_MASK | SPI_SSOFFTIME_MASK);
        reg |= clk_cfg;

        bcm_spi_writeb(bs, reg, SPI_CLK_CFG);
-       dev_dbg(&spi->dev, "Setting clock register to %02x (hz %d)\n",
-               clk_cfg, t->speed_hz);
+
+       bs->last_speed_hz = speed_hz;
+       bs->last_cs_off_clk_cycles = cs_off_clk_cycles;
+
+       return (clk_cfg);
+}
+
+static inline void bcm63xx_spi_set_fill_byte(struct bcm63xx_spi *bs, u8 fill_byte)
+{
+       bcm_spi_writeb(bs, fill_byte, SPI_FILL_BYTE);
+
+       bs->last_fill_byte = fill_byte;
+}
+
+static void bcm63xx_spi_setup_transfer(struct spi_device *spi,
+       u32 speed_hz, u8 cs_off_clk_cycles, u8 fill_byte)
+{
+       struct bcm63xx_spi *bs = spi_master_get_devdata(spi->master);
+
+       if ((speed_hz != bs->last_speed_hz)
+           || (cs_off_clk_cycles != bs->last_cs_off_clk_cycles)) {
+               u8 clk_cfg = bcm63xx_spi_set_clk_cfg(bs, speed_hz, cs_off_clk_cycles);
+               dev_dbg(&spi->dev, "Setting clock register to %02x (hz %d)\n",
+                       clk_cfg, speed_hz);
+       }
+       if (fill_byte != bs->last_fill_byte) {
+               bcm63xx_spi_set_fill_byte(bs, fill_byte);
+       }
 }

 /* the spi->mode bits understood by this driver: */
 #define MODEBITS (SPI_CPOL | SPI_CPHA)

-static int bcm63xx_txrx_bufs(struct spi_device *spi, struct spi_transfer *first,
-                               unsigned int num_transfers)
+static int bcm63xx_txrx_bufs(struct spi_device *spi,
+               struct spi_transfer *first, unsigned int num_transfers,
+               bool drop_cs_after_each_byte, bool wait_completion_with_irq)
 {
        struct bcm63xx_spi *bs = spi_master_get_devdata(spi->master);
        u16 msg_ctl;
kofec@E5420Mint /mnt/openwrt/OpenwrtHG553/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.9.87 $
```

pgid69, 2018-03-21:
The last time i tested the patch, it was on linux 4.4.89.
This is why it failed to apply with more recent kernel.
Only if you have special need, i would suggest to ignore patch 001... and only apply patch 000...,  the one that modify the kernel configuration to compile the spi driver as a module instead of a built-in driver, and of course, to undef BCMPH_USE_SPI_DRIVER in inc/config.h

kofec, 2018-03-21:
that what I did. If I coorectly understand I have to find spi-bcm63xx.ko module and added to my HG553 
```
kofec@E5420Mint /mnt/openwrt/OpenwrtHG553 $ find . | grep spi-bcm63xx.ko
./staging_dir/target-mips_mips32_musl/root-brcm63xx/lib/modules/4.4.121/spi-bcm63xx.ko
```
and modules which I have:

```
root@HG553:~# strings /lib/modules/4.4.121/bcm63xx-phone.ko | grep depends
depends=echo,spi-bcm63xx
root@HG553:~# insmod bcm63xx-phone

root@HG553:~# lsmod
bcm63xx_phone         171792  0
echo                    3490  1 bcm63xx_phone
spi_bcm63xx             4588  1 bcm63xx_phone
```
I will test today.

lulol, 2018-06-12:
I just ran into this unterminated #ifndef issue which is due to a bug in the line count of the patch that creates the final patch.

This may happen if a patch is hand edited after generated (don't do that!) to insert some blank line and the total line count is not updated accordingly.

Have submitted a pull request for your consideration that fixes the issue.

pgid69, 2018-06-13:
Thank you for your help. I think you're right : maybe i edited the patch.

pgid69, 2018-07-04:
Hi

Can you tell me what are the parameters you use tou launch bcm63xx-phone-test ?

Thanks

closed 2018-07-07 by pgid69