pgid69/bcm63xx-phone #26: Vodafone/Huawei HG556a HG56BZRB VER.A (VoIP chip Le88266DLC) Error starting

lulol, 2018-07-06:
### Vodafone/Huawei HG556a HG56BZRB VER.A (VoIP chip Le88266DLC)
**LEDE Reboot (17.01.4, r3560-79f57e422d)**

Module` bcm63xx-phone.ko` loads and initializes successfully but `bcm63xx-phone-test` fails with:
ioctl(2149598476) failed. errno=22
Error starting

The parameters I used with bcm63xx-phone-test and some additional debug output ithat may be useful are below.
Using different values for the defaults didn't notice any difference either.

`root@OpenWrt:~# modprobe /lib/modules/4.4.92/bcm63xx-phone.ko`

`root@OpenWrt:~# dmesg -c`
[  198.756476] bcm63xx_phone: bcm63xx_phone: v0.2
[  198.761153] bcm63xx_phone bcm63xx_phone_init() 1021: bcm63xx_phone_init()
[  198.768233] bcm63xx_phone board_init() 719: board_init()
[  198.773737] bcm63xx_phone: CPU is Broadcom 6358
[  198.778447] bcm63xx_phone: Board is HW556
[  198.782617] bcm63xx_phone board_get_desc() 778: board_get_desc()
[  198.788924] bcm63xx_phone bcm_periodic_timer_init() 249: bcm_periodic_timer_init()
[  198.796778] bcm63xx_phone bcm_timer_init() 191: bcm_timer_init()
[  198.803009] bcm63xx_phone bcm_period_init() 28: bcm_period_init()
[  198.809343] bcm63xx_phone bcm63xx_phone_init() 1081: t->mm_desc.ioctl_param_off = 0, t->mm_desc.ioctl_param_size = 220
[  198.820456] bcm63xx_phone bcm63xx_phone_init() 1090: t->mm_desc.rbs[0].rx_ring_buf_desc_off = 224, t->mm_desc.rbs[0].rx_ring_buf_desc_size = 12
[  198.833783] bcm63xx_phone bcm63xx_phone_init() 1098: t->mm_desc.rbs[0].tx_ring_buf_desc_off = 240, t->mm_desc.rbs[0].tx_ring_buf_desc_size = 12
[  198.847149] bcm63xx_phone bcm63xx_phone_init() 1090: t->mm_desc.rbs[1].rx_ring_buf_desc_off = 256, t->mm_desc.rbs[1].rx_ring_buf_desc_size = 12
[  198.860468] bcm63xx_phone bcm63xx_phone_init() 1098: t->mm_desc.rbs[1].tx_ring_buf_desc_off = 272, t->mm_desc.rbs[1].tx_ring_buf_desc_size = 12
[  198.873895] bcm63xx_phone bcm63xx_phone_init() 1128: 8192 bytes of memory allocated, starting at address 0x80c70000
[  198.884705] bcm63xx_phone bcm_phone_mgr_init() 1048: bcm_phone_mgr_init()
[  198.891745] bcm63xx_phone bcm_periodic_work_init() 92: bcm_periodic_work_init()
[  198.899314] bcm63xx_phone bcm_period_init() 28: bcm_period_init()
[  198.905633] bcm63xx_phone bcm_phone_mgr_devs_init() 907: bcm_phone_mgr_devs_init()
[  198.913477] bcm63xx_phone phone_device_alloc() 53: phone_device_alloc()
[  198.920372] bcm63xx_phone phone_device_alloc() 82: 2032 bytes of memory allocated, starting at address 0x82eb3000
[  198.931024] bcm63xx_phone phone_dev_le88266_init() 56: phone_dev_le88266_init()
[  198.938600] bcm63xx_phone phone_dev_zarlink_ve880_init() 2017: phone_dev_zarlink_ve880_init()
[  198.947430] bcm63xx_phone bcm_mpi_init() 878: bcm_mpi_init()
[  198.953472] bcm63xx_phone bcm63xx_spi_probe() 414: bcm63xx_spi_probe()
[  198.960350] bcm63xx-spi bcm6358-spi: at 0xfffe0800 (irq 9, FIFOs size 542)
[  198.967860] bcm63xx_phone phone_dev_zarlink_init() 678: phone_dev_zarlink_init()
[  198.975563] bcm63xx_phone zarlink_reset() 162: zarlink_reset(gpio_num=0x8018)
[  199.050732] bcm63xx_phone phone_device_init() 28: phone_device_init()
[  199.057432] bcm63xx_phone zarlink_device_id_init() 456: zarlink_device_id_init()
[  199.065151] bcm63xx_phone: 1 device(s) provides 2 FXS line(s)
[  199.071132] bcm63xx_phone bcm_phone_mgr_init() 1072: Initializing event wait queue
[  199.078984] bcm63xx_phone bcm_wait_queue_init() 31: bcm_wait_queue_init()
[  199.086022] bcm63xx_phone bcm_drv_init() 1173: bcm_drv_init()
[  199.091991] bcm63xx_phone bcm_drv_init() 1178: Initializing RX wait queue
[  199.099032] bcm63xx_phone bcm_wait_queue_init() 31: bcm_wait_queue_init()
[  199.106072] bcm63xx_phone bcm_drv_init() 1180: Initializing TX wait queue
[  199.113130] bcm63xx_phone bcm_wait_queue_init() 31: bcm_wait_queue_init()
[  199.120173] bcm63xx_phone bcm_drv_init() 1195: Registering device bcm63xx_phone
[  199.128180] bcm63xx_phone pcm_init() 2531: pcm_init()
[  199.133478] bcm63xx_phone pcm_init() 2607: Using 15 buffers of 256 bytes for RX
[  199.141060] bcm63xx_phone pcm_init() 2649: Using 15 buffers of 256 bytes for TX
[  199.148685] bcm63xx_phone pcm_get_shared_res() 1986: pcm_get_shared_res()
[  199.155757] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm, addr=0xfffe1600, size=256)
[  199.165236] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm_dma, addr=0xfffe1800, size=256)
[  199.175070] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm_dmac, addr=0xfffe1900, size=32)
[  199.184898] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm_dmas, addr=0xfffe1a00, size=32)
[  199.282733] bcm63xx_phone pcm6358_pll_init() 1001: pcm6358_pll_init()
[  199.289450] bcm63xx_phone pcm6358_pll_init() 1006:  pll_ctrl1=f0010001, pll_ctrl2=53d47, pll_ctrl3=a85
[  199.330741] bcm63xx_phone pcm6358_pll_init() 1059:  pll_ctrl1=20001, pll_ctrl2=8492b, pll_ctrl3=1e1c
[  199.340245] bcm63xx_phone pcm_regs_init() 1279: pcm_regs_init()
[  199.346410] bcm63xx_phone pcm_regs_reset() 1223: pcm_regs_reset()
[  199.352780] bcm63xx_phone pcm_regs_reset() 1260: PCM_CTRL_REG <- 0x23200101
[  199.360020] bcm63xx_phone pcm_dma_init() 1538: pcm_dma_init()
[  199.365986] bcm63xx_phone pcm_dma_init() 1571: Allocating 120 bytes of DMA memory for RX descs
[  199.374973] bcm63xx_phone pcm_dma_init() 1582: 120 bytes of memory allocated, starting at address 0xa0c82000
[  199.385153] bcm63xx_phone pcm_dma_init() 1588: Allocating 120 bytes of DMA memory for TX descs
[... 340 lines cut ...]
[  315.171409] bcm63xx_phone bcm_drv_stop() 850: bcm_drv_stop()
[  315.177299] bcm63xx_phone pcm_stop() 2370: pcm_stop()
[  315.182562] bcm63xx_phone pcm_dma_stop() 1506: pcm_dma_stop()
[  315.188527] bcm63xx_phone pcm_stop() 2382: Halting TX DMA
[  315.199002] bcm63xx_phone pcm_stop() 2389: TX DMA halted
[  315.204540] bcm63xx_phone pcm_stop() 2404: Halting RX DMA
[  315.215039] bcm63xx_phone pcm_stop() 2411: RX DMA halted
[  315.220585] bcm63xx_phone pcm_stop() 2422: Disabling all PCM channels
[  315.227306] bcm63xx_phone pcm_regs_reset() 1223: pcm_regs_reset()
[  315.233648] bcm63xx_phone pcm_regs_reset() 1260: PCM_CTRL_REG <- 0x23200101
[  315.240897] bcm63xx_phone pcm_dma_tx_descs_reset() 1405: pcm_dma_tx_descs_reset()
[  315.248715] bcm63xx_phone pcm_pr_stats() 2100: ======= Stats =======
[  315.255325] bcm63xx_phone pcm_pr_stats() 2101: rx_errors            = 0
[  315.262191] bcm63xx_phone pcm_pr_stats() 2102: rx_length_errors     = 0
[  315.269106] bcm63xx_phone pcm_pr_stats() 2103: rx_empty_errors      = 0
[  315.276010] bcm63xx_phone pcm_pr_stats() 2104: rx_good              = 0
[  315.282865] bcm63xx_phone pcm_pr_stats() 2105: rx_bytes             = 0
[  315.289768] bcm63xx_phone pcm_pr_stats() 2106: tx_errors            = 0
[  315.296628] bcm63xx_phone pcm_pr_stats() 2107: tx_good              = 0
[  315.303481] bcm63xx_phone pcm_pr_stats() 2108: tx_bytes             = 0
[  315.310334] bcm63xx_phone pcm_pr_stats() 2109: cnt_irq_rx           = 0
[  315.317190] bcm63xx_phone pcm_pr_stats() 2110: cnt_irq_tx           = 0
[  315.324062] bcm63xx_phone bcm_phone_mgr_stop() 773: bcm_phone_mgr_stop()
[  315.331011] bcm63xx_phone bcm_periodic_timer_del_work() 199: bcm_periodic_timer_del_work()
[  315.339583] bcm63xx_phone phone_dev_zarlink_ve880_stop() 2174: phone_dev_zarlink_ve880_stop()

pgid69, 2018-07-07:
Hello

Thanks for your logs.
I made a fix that should resolve your problem.
You can also add param codec0=slin if you don't want to apply the fix.

BR

lulol, 2018-07-07:
Using the slin codec instad of alaw or ulaw works. This solves my problem. Thank you very much!

`# bcm63xx-phone-test test=tone codec0=slin`
`# bcm63xx-phone-test test=echo codec0=slin output=/tmp/echo.dat`

Both tests work as expected. Also tested the channel driver with Asterisk 11 and has good sound.

For future reference, currently the line matching the labels above the RJ11 connectors for this router are:
- line0 is labeled as PHONE2
- line1 is labeled as PHONE1

Still seems to persist some issue with the other line (line1).

Without this patch:
`# bcm63xx-phone-test test=tone codec1=slin line1=1 default_line=1`
ioctl(2149598476) failed. errno=22
Error starting

`# bcm63xx-phone-test test=tone codec0=slin codec1=slin line1=1 default_line=1`
Line 1 should be ringing
Waiting for the line 1 to be off hook...
Waiting for the line 1 to be off hook...

But line1 (PHONE1) is not enabled (no line voltage present), instead line0 (PHONE2) is enabled (line voltage present) but doesn't ring nor is detected when off hook.

With this patch the codec parameters can be ommited but line1 still doesn't seem to work as expected:
`# bcm63xx-phone-test test=tone line1=1 default_line=1`
Line 1 should be ringing
Waiting for the line 1 to be off hook...
Waiting for the line 1 to be off hook...

Again line1 (PHONE1) is not enabled and shows the same behaviour as before.

Regards.

lulol, 2018-07-09:
@pgid69 sure, there you go. Thank you

`root@OpenWrt:~# bcm63xx-phone-test test=tone line1=1 default_line=1`
Line 1 should be ringing
Waiting for the line 1 to be off hook...
Waiting for the line 1 to be off hook...
^C

`root@OpenWrt:~# dmesg -c`
[131391.849622] bcm63xx_phone bcm_drv_open() 997: bcm_drv_open()
[131391.855678] bcm63xx_phone bcm_drv_unlocked_ioctl() 640: BCMPH_IOCTL_START
[131391.862860] bcm63xx_phone bcm_drv_start() 949: bcm_drv_start()
[131391.869009] bcm63xx_phone bcm_drv_stop() 850: bcm_drv_stop()
[131391.875045] bcm63xx_phone pcm_stop() 2370: pcm_stop()
[131391.880402] bcm63xx_phone pcm_dma_stop() 1506: pcm_dma_stop()
[131391.886451] bcm63xx_phone pcm_stop() 2382: Halting TX DMA
[131391.898999] bcm63xx_phone pcm_stop() 2389: TX DMA halted
[131391.904605] bcm63xx_phone pcm_stop() 2404: Halting RX DMA
[131391.915000] bcm63xx_phone pcm_stop() 2411: RX DMA halted
[131391.920608] bcm63xx_phone pcm_stop() 2422: Disabling all PCM channels
[131391.927398] bcm63xx_phone pcm_regs_reset() 1223: pcm_regs_reset()
[131391.933839] bcm63xx_phone pcm_regs_reset() 1260: PCM_CTRL_REG <- 0x23200101
[131391.941166] bcm63xx_phone pcm_dma_tx_descs_reset() 1405: pcm_dma_tx_descs_reset()
[131391.949069] bcm63xx_phone pcm_pr_stats() 2100: ======= Stats =======
[131391.955767] bcm63xx_phone pcm_pr_stats() 2101: rx_errors            = 0
[131391.962726] bcm63xx_phone pcm_pr_stats() 2102: rx_length_errors     = 0
[131391.969681] bcm63xx_phone pcm_pr_stats() 2103: rx_empty_errors      = 0
[131391.976701] bcm63xx_phone pcm_pr_stats() 2104: rx_good              = 0
[131391.983675] bcm63xx_phone pcm_pr_stats() 2105: rx_bytes             = 0
[131391.990621] bcm63xx_phone pcm_pr_stats() 2106: tx_errors            = 0
[131391.997566] bcm63xx_phone pcm_pr_stats() 2107: tx_good              = 0
[131392.004508] bcm63xx_phone pcm_pr_stats() 2108: tx_bytes             = 0
[131392.011453] bcm63xx_phone pcm_pr_stats() 2109: cnt_irq_rx           = 0
[131392.018396] bcm63xx_phone pcm_pr_stats() 2110: cnt_irq_tx           = 0
[131392.025343] bcm63xx_phone bcm_phone_mgr_stop() 773: bcm_phone_mgr_stop()
[131392.032376] bcm63xx_phone bcm_periodic_timer_del_work() 199: bcm_periodic_timer_del_work()
[131392.041030] bcm63xx_phone phone_dev_zarlink_ve880_stop() 2174: phone_dev_zarlink_ve880_stop()
[131392.049971] bcm63xx_phone bcm_phone_mgr_get_first_timeslot_line() 143: bcm_phone_mgr_get_first_timeslot_line(line=0)
[131392.060965] bcm63xx_phone bcm_phone_mgr_get_first_timeslot_line() 143: bcm_phone_mgr_get_first_timeslot_line(line=1)
[131392.071969] bcm63xx_phone bcm_drv_start() 1259: t->mm_rbs_location.rbs[0].rx_buffer_offset = 288, t->mm_rbs_location.rbs[0].rx_buffer_size = 1968
[131392.085576] bcm63xx_phone bcm_drv_start() 1267: t->mm_rbs_location.rbs[0].tx_buffer_offset = 2256, t->mm_rbs_location.rbs[0].tx_buffer_size = 1984
[131392.099265] bcm63xx_phone bcm_drv_start() 1259: t->mm_rbs_location.rbs[1].rx_buffer_offset = 4240, t->mm_rbs_location.rbs[1].rx_buffer_size = 1968
[131392.112946] bcm63xx_phone bcm_drv_start() 1267: t->mm_rbs_location.rbs[1].tx_buffer_offset = 6208, t->mm_rbs_location.rbs[1].tx_buffer_size = 1984
[131392.126626] bcm63xx_phone pcm_configure_channels() 2134: pcm_configure_channels()
[131392.134479] bcm63xx_phone pcm_start() 2212: pcm_start()
[131392.139992] bcm63xx_phone pcm_regs_reset() 1223: pcm_regs_reset()
[131392.146401] bcm63xx_phone pcm_regs_reset() 1260: PCM_CTRL_REG <- 0x23200101
[131392.153712] bcm63xx_phone pcm_timeslot_alloc() 1202: pcm_timeslot_alloc(chnum=0, timeslot=0)
[131392.162542] bcm63xx_phone pcm_timeslot_alloc() 1202: pcm_timeslot_alloc(chnum=1, timeslot=1)
[131392.171376] bcm63xx_phone pcm_timeslot_alloc() 1202: pcm_timeslot_alloc(chnum=2, timeslot=2)
[131392.180224] bcm63xx_phone pcm_timeslot_alloc() 1202: pcm_timeslot_alloc(chnum=3, timeslot=3)
[131392.189069] bcm63xx_phone pcm_dma_rx_descs_reset() 1374: pcm_dma_rx_descs_reset()
[131392.196964] bcm63xx_phone pcm_start() 2339: PCM_CHAN_CTRL_REG <- 0xf0f
[131392.307031] bcm63xx_phone bcm_phone_mgr_start() 800: bcm_phone_mgr_start()
[131392.314287] bcm63xx_phone phone_dev_zarlink_ve880_start() 2159: phone_dev_zarlink_ve880_start(country=14, line_count=2)
[131392.325572] bcm63xx_phone phone_dev_zarlink_ve880_stop() 2174: phone_dev_zarlink_ve880_stop()
[131392.334523] bcm63xx_phone phone_dev_zarlink_start() 1888: phone_dev_zarlink_start(country=14)
[131392.343457] bcm63xx_phone phone_dev_zarlink_stop() 1836: phone_dev_zarlink_stop()
[131392.351322] bcm63xx_phone phone_dev_zarlink_init_vp_objects() 748: phone_dev_zarlink_init_vp_objects(country=14)
[131392.361963] bcm63xx_phone zarlink_dev_make() 471: zarlink_dev_make(deviceType=4)
[131392.369771] bcm63xx_phone zarlink_slic_line_make() 541: zarlink_slic_line_make(term_type=0, internal_channel_id=0, external_channel_id=0)
[131392.382679] bcm63xx_phone zarlink_slic_line_make() 541: zarlink_slic_line_make(term_type=0, internal_channel_id=1, external_channel_id=1)
[131392.395638] bcm63xx_phone zarlink_dev_init() 498: zarlink_dev_init(country=14)
[131392.463369] bcm63xx_phone zarlink_wait_for_event() 316: zarlink_wait_for_event(evt_category=2, evt_id=1024)
[131396.967287] bcm63xx_phone zarlink_wait_for_event() 334: Received VP_DEV_EVID_DEV_INIT_CMP event (i = 2810)
[131396.977425] bcm63xx_phone zarlink_wait_for_event() 354: Found event: pEvent.eventCategory = 2 pEvent.eventId = 1024 
[... cut ...]
[131411.569649] bcm63xx_phone phone_dev_zarlink_deinit_vp_objects() 863: phone_dev_zarlink_deinit_vp_objects()
[131411.579737] bcm63xx_phone zarlink_slic_line_deinit() 665: zarlink_slic_line_deinit()
[131411.587865] bcm63xx_phone zarlink_slic_line_deinit() 665: zarlink_slic_line_deinit()
[131411.595992] bcm63xx_phone zarlink_dev_deinit() 529: zarlink_dev_deinit()
[131411.603050] bcm63xx_phone zarlink_slic_line_free() 567: zarlink_slic_line_free()
[131411.610811] bcm63xx_phone zarlink_slic_line_free() 567: zarlink_slic_line_free()
[131411.618576] bcm63xx_phone zarlink_dev_free() 488: zarlink_dev_free()

pgid69, 2018-07-09:
Weird ! 

Everything in the log seems correct : both lines are initialized (line0 is implicitly enabled if you don't call bcm-phone-test with param line0=0) so at least you should have voltages on both lines.
I will run again the tests with my modem but it will take me some time, so be patient.
On your side, if you have time, you can try to make a modification in file /home/mazoyer/mnt/dev/github/bcm63xx-phone/bcm63xx-phone/src/board/board.c at lines 480 and 485, to switch the ids so that line 0 is id 1 (PHONE 1 RJ11 connector) and line 1 is id 0 (PHONE 2 RJ11 connector), recompile bcm63xx-phone kernel module and rerun the tests.

BR

lulol, 2018-07-09:
With one line is enough for my normal use so no worries. Just it's nice to have both lines to test asterisk configurations more easily.

Anyway I'll post the results as soon as can test the changes you propose.

lulol, 2018-07-09:
Now with the ids swapped in the kernel module
`# bcm63xx-phone-test test=tone line1=1 default_line=1`
rings PHONE2, as expected.

Also have noticed that with `# bcm63xx-phone-test test=tone` (which now it should enable PHONE1 but still doesn't) when off hook, PHONE2 is enabled but just for a second (this phone terminal seems to store charge quickly and use very little power) so really looks more like some kind of glitch on PHONE2 when the driver tries to enable PHONE1 but actually none is enabled more than the few seconds the PHONE2 charge lasts.

lulol, 2018-07-09:
@pgid69 That is correct. Thank you for checking.

`# bcm63xx-phone-test test=tone`
With one or both phones plugged  there is no difference. Only PHONE2 works and when enabling PHONE1 there is this glitch in PHONE2.
PHONE1 doesn't seem to ever get any voltage or glitch.

I'll try to look for the original firmware and flash it to see if the hardware is ok. Until I can confirm that this it's not a hardware problem and unless someone else confirms the issue I think that it's not worth spending time guessing just in case.

pgid69, 2018-07-09:
I run the tests on my router (Huawei HG553) and i have no problem.
When running `# bcm63xx-phone-test test=tone` with the phone plugged into PHONE 1, it rings and when i hook off i can hear the tone.
When running `# bcm63xx-phone-test test=tone line1=1 default_line=1` with the phone plugged into PHONE 2, it rings and when i hook off i can hear the tone.
I have no idea what could be wrong with your router !
Perhaps you could also try to change the country in bcm63xx-phone-test/src/main.c at line 281.
I have hard coded BCMPH_COUNTRY_FR (for France where i live) but perhaps your phone is not compatible.
Other values possible are build by appending to BCMPH_COUNTRY_ , codes defined in bcm63xx-phone/src/inc/countryArchive.h 

BR

lulol, 2018-07-09:
Well, have flashed `45c8569283fa78cd5033c2f55651baca  HG556aV100R001C10B077.bin` and seems that there is no hardware problem at all. Both lines are enabled and in both a voice propmt ask to check the usb modem (which is missing).

So back to openwrt.

Don't think that the phone wouldn't be compatible just to sense the line voltage on one of the fxs port, and have used two different phone terminal devices. Now also is verified that they work with the original firmware.

Still, the HG556 uses a Le88266 while your HG553 uses a Le88221, but one would expect that the SDK handles the differences.

Also as you suggested have changed BCMPH_COUNTRY_FR in `bcm63xx-phone-test/src/main.c` but can't see any difference. Should the dial tone change accordingly? It always sounds like an US dial dual tone.

With BCMPH_COUNTRY_FR the dial tone doesn't sound like a 440Hz tone and neither BCMPH_COUNTRY_ES sounds like 425Hz.

pgid69, 2018-07-10:
Thank you for your feedback.
You're right differences between Le88221 and Le88266 are handled by SDK provided by Zarlink.
Country does not change tone generation, it only changes the profile loaded by SDK to configure voltage for ringing and other hardware parameters...
Now that we have excluded a hardware problem, it can only be a software problem. But i have no idea what other test to make ?

lulol, 2018-07-10:
No worries. You already did a very good job with this router and now the voip is usable with openwrt. Many thanks!

Since the logs doesn't seem to show anything wrong and the Pirelli A226m also uses the Le88266 and it's [reported](https://github.com/pgid69/bcm63xx-phone/issues/5#issuecomment-213593408) that both fxs ports work fine with basically the same line parameters defined than the hw556 so well may be some issue with some models of the HG556a like my HG56BZRB VER.A

Long ago @dotmark already [reported](https://github.com/pgid69/bcm63xx-phone/issues/3#issuecomment-110316831) that only could enable one line, but his/her HG556a exact model is unknown.

May be @YPKEH and @talnacialex can comment if both fxs ports are working with their HG556a, HG55VDFA VER.C?

I'm happy with one working port but I'll be glad to help with testing. Regards.

pgid69, 2018-07-10:
Can you make two more tests ?
Considering that test `# bcm63xx-phone-test test=tone` does not work can you add parameter `codec0=slin16` ?
If that still does not work can you try to modify the id of line 0 : yesterday in bcm63xx-phone/src/board/board.c at line 480 we set id to 1 : now can you set it to 2 ? 
Thank you

lulol, 2018-07-10:
Hmmm. Had just reverted that id swap change since I prefer to do not keep cosmetic custom changes.

So the test that I guess were equivalent to your request, were no different and line1 never was enabled:

`# bcm63xx-phone-test test=tone line0=0 line1=1 default_line=1`
Line 1 should be ringing
Waiting for the line 1 to be off hook...
Waiting for the line 1 to be off hook...
^C
`# bcm63xx-phone-test test=tone line0=0 line1=1 default_line=1 codec1=slin16`
Line 1 should be ringing
Waiting for the line 1 to be off hook...
Waiting for the line 1 to be off hook...
^C

And as I understand, without the swap, should change the id 1 to 2 for line 1 in line 485.

This night can recompile the kernel module and report back.

lulol, 2018-07-10:
Heh. Was able to sneak a quick compile, but no luck with the results.
```
--- board.c.orig 2018-07-10 03:27:25.956279650 +0200
+++ board.c 2018-07-10 16:16:57.885172750 +0200
@@ -482,7 +482,7 @@
 };
 
 static zarlink_line_parameters_t hw556_le88266_line1_params = {
-   .id = 1,
+   .id = 2,
    .type = VP_TERM_FXS_GENERIC,
 };
 
```

`# bcm63xx-phone-test test=tone line0=0 line1=1 default_line=1`
ioctl(2149598476) failed. errno=1
Error starting

`# dmesg -c`
[ 7478.226247] bcm63xx_phone bcm_drv_open() 997: bcm_drv_open()
[ 7478.232866] bcm63xx_phone bcm_drv_unlocked_ioctl() 640: BCMPH_IOCTL_START
[ 7478.239983] bcm63xx_phone bcm_drv_start() 949: bcm_drv_start()
[ 7478.246107] bcm63xx_phone bcm_drv_stop() 850: bcm_drv_stop()
[ 7478.252006] bcm63xx_phone pcm_stop() 2370: pcm_stop()
[ 7478.257245] bcm63xx_phone pcm_dma_stop() 1506: pcm_dma_stop()
[ 7478.263205] bcm63xx_phone pcm_stop() 2382: Halting TX DMA
[ 7478.273751] bcm63xx_phone pcm_stop() 2389: TX DMA halted
[ 7478.279590] bcm63xx_phone pcm_stop() 2404: Halting RX DMA
[ 7478.289812] bcm63xx_phone pcm_stop() 2411: RX DMA halted
[ 7478.295331] bcm63xx_phone pcm_stop() 2422: Disabling all PCM channels
[ 7478.302078] bcm63xx_phone pcm_regs_reset() 1223: pcm_regs_reset()
[ 7478.308406] bcm63xx_phone pcm_regs_reset() 1260: PCM_CTRL_REG <- 0x23200101
[ 7478.315627] bcm63xx_phone pcm_dma_tx_descs_reset() 1405: pcm_dma_tx_descs_reset()
[ 7478.323433] bcm63xx_phone pcm_pr_stats() 2100: ======= Stats =======
[ 7478.330022] bcm63xx_phone pcm_pr_stats() 2101: rx_errors            = 0
[ 7478.336874] bcm63xx_phone pcm_pr_stats() 2102: rx_length_errors     = 0
[ 7478.343729] bcm63xx_phone pcm_pr_stats() 2103: rx_empty_errors      = 0
[ 7478.350607] bcm63xx_phone pcm_pr_stats() 2104: rx_good              = 0
[ 7478.357485] bcm63xx_phone pcm_pr_stats() 2105: rx_bytes             = 0
[ 7478.364338] bcm63xx_phone pcm_pr_stats() 2106: tx_errors            = 0
[ 7478.371211] bcm63xx_phone pcm_pr_stats() 2107: tx_good              = 0
[ 7478.378066] bcm63xx_phone pcm_pr_stats() 2108: tx_bytes             = 0
[ 7478.384929] bcm63xx_phone pcm_pr_stats() 2109: cnt_irq_rx           = 0
[ 7478.391782] bcm63xx_phone pcm_pr_stats() 2110: cnt_irq_tx           = 0
[ 7478.398638] bcm63xx_phone bcm_phone_mgr_stop() 773: bcm_phone_mgr_stop()
[ 7478.405581] bcm63xx_phone bcm_periodic_timer_del_work() 199: bcm_periodic_timer_del_work()
[ 7478.414146] bcm63xx_phone phone_dev_zarlink_ve880_stop() 2174: phone_dev_zarlink_ve880_stop()
[ 7478.423015] bcm63xx_phone bcm_phone_mgr_get_first_timeslot_line() 143: bcm_phone_mgr_get_first_timeslot_line(line=1)
[ 7478.433931] bcm63xx_phone bcm_drv_start() 1276: t->mm_rbs_location.rbs[0].rx_buffer_offset = 288, t->mm_rbs_location.rbs[0].rx_buffer_size = 16
[ 7478.447257] bcm63xx_phone bcm_drv_start() 1281: t->mm_rbs_location.rbs[0].tx_buffer_offset = 288, t->mm_rbs_location.rbs[0].tx_buffer_size = 16
[ 7478.460584] bcm63xx_phone bcm_drv_start() 1259: t->mm_rbs_location.rbs[1].rx_buffer_offset = 304, t->mm_rbs_location.rbs[1].rx_buffer_size = 3936
[ 7478.474102] bcm63xx_phone bcm_drv_start() 1267: t->mm_rbs_location.rbs[1].tx_buffer_offset = 4240, t->mm_rbs_location.rbs[1].tx_buffer_size = 3952
[ 7478.487692] bcm63xx_phone pcm_configure_channels() 2134: pcm_configure_channels()
[ 7478.495453] bcm63xx_phone pcm_start() 2212: pcm_start()
[ 7478.500877] bcm63xx_phone pcm_regs_reset() 1223: pcm_regs_reset()
[ 7478.507199] bcm63xx_phone pcm_regs_reset() 1260: PCM_CTRL_REG <- 0x23200101
[ 7478.514427] bcm63xx_phone pcm_timeslot_alloc() 1202: pcm_timeslot_alloc(chnum=0, timeslot=2)
[ 7478.523180] bcm63xx_phone pcm_timeslot_alloc() 1202: pcm_timeslot_alloc(chnum=1, timeslot=3)
[ 7478.531919] bcm63xx_phone pcm_dma_rx_descs_reset() 1374: pcm_dma_rx_descs_reset()
[ 7478.539733] bcm63xx_phone pcm_start() 2339: PCM_CHAN_CTRL_REG <- 0x303
[ 7478.650013] bcm63xx_phone bcm_phone_mgr_start() 800: bcm_phone_mgr_start()
[ 7478.657240] bcm63xx_phone phone_dev_zarlink_ve880_start() 2159: phone_dev_zarlink_ve880_start(country=14, line_count=2)
[ 7478.668448] bcm63xx_phone phone_dev_zarlink_ve880_stop() 2174: phone_dev_zarlink_ve880_stop()
[ 7478.677299] bcm63xx_phone phone_dev_zarlink_start() 1888: phone_dev_zarlink_start(country=14)
[ 7478.686128] bcm63xx_phone phone_dev_zarlink_stop() 1836: phone_dev_zarlink_stop()
[ 7478.693886] bcm63xx_phone phone_dev_zarlink_init_vp_objects() 748: phone_dev_zarlink_init_vp_objects(country=14)
[ 7478.704421] bcm63xx_phone zarlink_dev_make() 471: zarlink_dev_make(deviceType=4)
[ 7478.712122] bcm63xx_phone zarlink_slic_line_make() 541: zarlink_slic_line_make(term_type=0, internal_channel_id=0, external_channel_id=0)
[ 7478.724929] bcm63xx_phone zarlink_slic_line_make() 541: zarlink_slic_line_make(term_type=0, internal_channel_id=2, external_channel_id=1)
[... 135 lines cut ...]
[ 7506.598114] bcm63xx_phone bcm_drv_stop() 850: bcm_drv_stop()
[ 7506.604028] bcm63xx_phone pcm_stop() 2370: pcm_stop()
[ 7506.609341] bcm63xx_phone pcm_dma_stop() 1506: pcm_dma_stop()
[ 7506.615326] bcm63xx_phone pcm_stop() 2382: Halting TX DMA
[ 7506.628558] bcm63xx_phone pcm_stop() 2389: TX DMA halted
[ 7506.634088] bcm63xx_phone pcm_stop() 2404: Halting RX DMA
[ 7506.644563] bcm63xx_phone pcm_stop() 2411: RX DMA halted
[ 7506.650090] bcm63xx_phone pcm_stop() 2422: Disabling all PCM channels
[ 7506.656792] bcm63xx_phone pcm_regs_reset() 1223: pcm_regs_reset()
[ 7506.663137] bcm63xx_phone pcm_regs_reset() 1260: PCM_CTRL_REG <- 0x23200101
[... cut ...]
[ 7506.726099] bcm63xx_phone pcm_pr_stats() 2107: tx_good              = 0
[ 7506.732954] bcm63xx_phone pcm_pr_stats() 2108: tx_bytes             = 0
[ 7506.739807] bcm63xx_phone pcm_pr_stats() 2109: cnt_irq_rx           = 0
[ 7506.746663] bcm63xx_phone pcm_pr_stats() 2110: cnt_irq_tx           = 0
[ 7506.753534] bcm63xx_phone bcm_phone_mgr_stop() 773: bcm_phone_mgr_stop()
[ 7506.760497] bcm63xx_phone bcm_periodic_timer_del_work() 199: bcm_periodic_timer_del_work()
[ 7506.769064] bcm63xx_phone phone_dev_zarlink_ve880_stop() 2174: phone_dev_zarlink_ve880_stop()

talnacialex, 2018-07-10:
My second FXS port also does not work, but when pgid69 compiled a special
openwrt image for my router a while ago, both FXS ports worked.

pgid69, 2018-07-10:
@lulol 
Ok so id can only be 1. It's logical.
Perhaps we can try to activate debug messages in Zarlink SDK
In file bcm63xx-phone/src/extern/zarlink/2.24.0.lite/api_lib/includes/vp_api_cfg.h change line 42 from `#undef VP_DEBUG` to `#define VP_DEBUG`, and change line 62 from `#define VP_OPTION_DEFAULT_DEBUG_SELECT (VP_DBG_ERROR | VP_DBG_WARNING)` to `#define VP_OPTION_DEFAULT_DEBUG_SELECT (VP_DBG_ERROR | VP_DBG_WARNING | VP_DBG_INFO | VP_DBG_API_FUNC | VP_DBG_API_FUNC_INT )`

Else i don't understand why the firmware i compiled for @talnacialex works : it was not a special image.

Once again thank you for your patience and help.

lulol, 2018-07-10:
With the above debug flags doesn't look any more verbose for the kernel module load or the test program.

`root@OpenWrt:/tmp# insmod ./bcm63xx-phone.ko `
`root@OpenWrt:/tmp# dmesg -c`
[28119.075746] bcm63xx_phone: bcm63xx_phone: v0.2
[28119.080408] bcm63xx_phone bcm63xx_phone_init() 1021: bcm63xx_phone_init()
[28119.087489] bcm63xx_phone board_init() 719: board_init()
[28119.093033] bcm63xx_phone: CPU is Broadcom 6358
[28119.097755] bcm63xx_phone: Board is HW556
[28119.101983] bcm63xx_phone board_get_desc() 778: board_get_desc()
[28119.108328] bcm63xx_phone bcm_periodic_timer_init() 249: bcm_periodic_timer_init()
[28119.116171] bcm63xx_phone bcm_timer_init() 191: bcm_timer_init()
[28119.122404] bcm63xx_phone bcm_period_init() 28: bcm_period_init()
[28119.128730] bcm63xx_phone bcm63xx_phone_init() 1081: t->mm_desc.ioctl_param_off = 0, t->mm_desc.ioctl_param_size = 220
[28119.139834] bcm63xx_phone bcm63xx_phone_init() 1090: t->mm_desc.rbs[0].rx_ring_buf_desc_off = 224, t->mm_desc.rbs[0].rx_ring_buf_desc_size = 12
[28119.153180] bcm63xx_phone bcm63xx_phone_init() 1098: t->mm_desc.rbs[0].tx_ring_buf_desc_off = 240, t->mm_desc.rbs[0].tx_ring_buf_desc_size = 12
[28119.166500] bcm63xx_phone bcm63xx_phone_init() 1090: t->mm_desc.rbs[1].rx_ring_buf_desc_off = 256, t->mm_desc.rbs[1].rx_ring_buf_desc_size = 12
[28119.179821] bcm63xx_phone bcm63xx_phone_init() 1098: t->mm_desc.rbs[1].tx_ring_buf_desc_off = 272, t->mm_desc.rbs[1].tx_ring_buf_desc_size = 12
[28119.193248] bcm63xx_phone bcm63xx_phone_init() 1128: 8192 bytes of memory allocated, starting at address 0x83984000
[28119.204058] bcm63xx_phone bcm_phone_mgr_init() 1048: bcm_phone_mgr_init()
[28119.211110] bcm63xx_phone bcm_periodic_work_init() 92: bcm_periodic_work_init()
[28119.218689] bcm63xx_phone bcm_period_init() 28: bcm_period_init()
[28119.225008] bcm63xx_phone bcm_phone_mgr_devs_init() 907: bcm_phone_mgr_devs_init()
[28119.232860] bcm63xx_phone phone_device_alloc() 53: phone_device_alloc()
[28119.239755] bcm63xx_phone phone_device_alloc() 82: 2032 bytes of memory allocated, starting at address 0x82431800
[28119.250387] bcm63xx_phone phone_dev_le88266_init() 56: phone_dev_le88266_init()
[28119.257964] bcm63xx_phone phone_dev_zarlink_ve880_init() 2017: phone_dev_zarlink_ve880_init()
[28119.266794] bcm63xx_phone bcm_mpi_init() 878: bcm_mpi_init()
[28119.272834] bcm63xx_phone bcm63xx_spi_probe() 414: bcm63xx_spi_probe()
[28119.279719] bcm63xx-spi bcm6358-spi: at 0xfffe0800 (irq 9, FIFOs size 542)
[28119.287238] bcm63xx_phone phone_dev_zarlink_init() 678: phone_dev_zarlink_init()
[28119.294943] bcm63xx_phone zarlink_reset() 162: zarlink_reset(gpio_num=0x8018)
[28119.368297] bcm63xx_phone phone_device_init() 28: phone_device_init()
[28119.374992] bcm63xx_phone zarlink_device_id_init() 456: zarlink_device_id_init()
[28119.382724] bcm63xx_phone: 1 device(s) provides 2 FXS line(s)
[28119.388701] bcm63xx_phone bcm_phone_mgr_init() 1072: Initializing event wait queue
[28119.396549] bcm63xx_phone bcm_wait_queue_init() 31: bcm_wait_queue_init()
[28119.403592] bcm63xx_phone bcm_drv_init() 1173: bcm_drv_init()
[28119.409612] bcm63xx_phone bcm_drv_init() 1178: Initializing RX wait queue
[28119.416674] bcm63xx_phone bcm_wait_queue_init() 31: bcm_wait_queue_init()
[28119.423710] bcm63xx_phone bcm_drv_init() 1180: Initializing TX wait queue
[28119.430744] bcm63xx_phone bcm_wait_queue_init() 31: bcm_wait_queue_init()
[28119.437781] bcm63xx_phone bcm_drv_init() 1195: Registering device bcm63xx_phone
[28119.445826] bcm63xx_phone pcm_init() 2531: pcm_init()
[28119.451131] bcm63xx_phone pcm_init() 2607: Using 15 buffers of 256 bytes for RX
[28119.458705] bcm63xx_phone pcm_init() 2649: Using 15 buffers of 256 bytes for TX
[28119.466279] bcm63xx_phone pcm_get_shared_res() 1986: pcm_get_shared_res()
[28119.473323] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm, addr=0xfffe1600, size=256)
[28119.482802] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm_dma, addr=0xfffe1800, size=256)
[28119.492635] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm_dmac, addr=0xfffe1900, size=32)
[28119.502471] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm_dmas, addr=0xfffe1a00, size=32)
[28119.600311] bcm63xx_phone pcm6358_pll_init() 1001: pcm6358_pll_init()
[28119.607009] bcm63xx_phone pcm6358_pll_init() 1006:  pll_ctrl1=f0010001, pll_ctrl2=53d47, pll_ctrl3=a85
[28119.648316] bcm63xx_phone pcm6358_pll_init() 1059:  pll_ctrl1=20001, pll_ctrl2=8492b, pll_ctrl3=1e1c
[28119.657840] bcm63xx_phone pcm_regs_init() 1279: pcm_regs_init()
[28119.663996] bcm63xx_phone pcm_regs_reset() 1223: pcm_regs_reset()
[28119.670322] bcm63xx_phone pcm_regs_reset() 1260: PCM_CTRL_REG <- 0x23200101
[28119.677540] bcm63xx_phone pcm_dma_init() 1538: pcm_dma_init()
[28119.683503] bcm63xx_phone pcm_dma_init() 1571: Allocating 120 bytes of DMA memory for RX descs
[28119.692493] bcm63xx_phone pcm_dma_init() 1582: 120 bytes of memory allocated, starting at address 0xa2579000
[28119.702682] bcm63xx_phone pcm_dma_init() 1588: Allocating 120 bytes of DMA memory for TX descs
[28119.711662] bcm63xx_phone pcm_dma_init() 1599: 120 bytes of memory allocated, starting at address 0xa2578000
[28119.721892] bcm63xx_phone pcm_dma_init() 1607: Allocating 7680 bytes of memory for RX and TX data buffers
[28119.731813] bcm63xx_phone pcm_dma_init() 1614: 7680 bytes of memory allocated, starting at address 0x82ea2000
[... cut ...]
[28182.983410] bcm63xx_phone phone_dev_zarlink_deinit_vp_objects() 863: phone_dev_zarlink_deinit_vp_objects()
[28182.993409] bcm63xx_phone zarlink_slic_line_deinit() 665: zarlink_slic_line_deinit()
[28183.001437] bcm63xx_phone zarlink_dev_deinit() 529: zarlink_dev_deinit()
[28183.008390] bcm63xx_phone zarlink_slic_line_free() 567: zarlink_slic_line_free()
[28183.016077] bcm63xx_phone zarlink_slic_line_free() 567: zarlink_slic_line_free()
[28183.023788] bcm63xx_phone zarlink_dev_free() 488: zarlink_dev_free()

lulol, 2018-07-10:
Sorry, have realized that failed to enable the `#define VP_DEBUG` part. I'll try again.

lulol, 2018-07-10:
Now that is more like it. It's quite a lot of text to post.

`root@OpenWrt:/tmp# insmod ./bcm63xx-phone.ko`
`root@OpenWrt:/tmp# dmesg -c`
[31186.385787] bcm63xx_phone: bcm63xx_phone: v0.2
[31186.390490] bcm63xx_phone bcm63xx_phone_init() 1021: bcm63xx_phone_init()
[31186.397574] bcm63xx_phone board_init() 719: board_init()
[31186.403082] bcm63xx_phone: CPU is Broadcom 6358
[31186.408348] bcm63xx_phone: Board is HW556
[31186.412524] bcm63xx_phone board_get_desc() 778: board_get_desc()
[31186.418821] bcm63xx_phone bcm_periodic_timer_init() 249: bcm_periodic_timer_init()
[31186.426693] bcm63xx_phone bcm_timer_init() 191: bcm_timer_init()
[31186.432949] bcm63xx_phone bcm_period_init() 28: bcm_period_init()
[31186.439300] bcm63xx_phone bcm63xx_phone_init() 1081: t->mm_desc.ioctl_param_off = 0, t->mm_desc.ioctl_param_size = 220
[31186.450385] bcm63xx_phone bcm63xx_phone_init() 1090: t->mm_desc.rbs[0].rx_ring_buf_desc_off = 224, t->mm_desc.rbs[0].rx_ring_buf_desc_size = 12
[31186.463706] bcm63xx_phone bcm63xx_phone_init() 1098: t->mm_desc.rbs[0].tx_ring_buf_desc_off = 240, t->mm_desc.rbs[0].tx_ring_buf_desc_size = 12
[31186.477028] bcm63xx_phone bcm63xx_phone_init() 1090: t->mm_desc.rbs[1].rx_ring_buf_desc_off = 256, t->mm_desc.rbs[1].rx_ring_buf_desc_size = 12
[31186.490352] bcm63xx_phone bcm63xx_phone_init() 1098: t->mm_desc.rbs[1].tx_ring_buf_desc_off = 272, t->mm_desc.rbs[1].tx_ring_buf_desc_size = 12
[31186.503778] bcm63xx_phone bcm63xx_phone_init() 1128: 8192 bytes of memory allocated, starting at address 0x824b2000
[31186.514619] bcm63xx_phone bcm_phone_mgr_init() 1048: bcm_phone_mgr_init()
[31186.521658] bcm63xx_phone bcm_periodic_work_init() 92: bcm_periodic_work_init()
[31186.529235] bcm63xx_phone bcm_period_init() 28: bcm_period_init()
[31186.535553] bcm63xx_phone bcm_phone_mgr_devs_init() 907: bcm_phone_mgr_devs_init()
[31186.543408] bcm63xx_phone phone_device_alloc() 53: phone_device_alloc()
[31186.550302] bcm63xx_phone phone_device_alloc() 82: 2044 bytes of memory allocated, starting at address 0x82430800
[31186.560928] bcm63xx_phone phone_dev_le88266_init() 56: phone_dev_le88266_init()
[31186.568508] bcm63xx_phone phone_dev_zarlink_ve880_init() 2017: phone_dev_zarlink_ve880_init()
[31186.577340] bcm63xx_phone bcm_mpi_init() 878: bcm_mpi_init()
[31186.583379] bcm63xx_phone bcm63xx_spi_probe() 414: bcm63xx_spi_probe()
[31186.590258] bcm63xx-spi bcm6358-spi: at 0xfffe0800 (irq 9, FIFOs size 542)
[31186.597769] bcm63xx_phone phone_dev_zarlink_init() 678: phone_dev_zarlink_init()
[31186.605470] bcm63xx_phone zarlink_reset() 162: zarlink_reset(gpio_num=0x8018)
[31186.680024] bcm63xx_phone phone_device_init() 28: phone_device_init()
[31186.686710] bcm63xx_phone zarlink_device_id_init() 456: zarlink_device_id_init()
[31186.694434] bcm63xx_phone: 1 device(s) provides 2 FXS line(s)
[31186.700415] bcm63xx_phone bcm_phone_mgr_init() 1072: Initializing event wait queue
[31186.708272] bcm63xx_phone bcm_wait_queue_init() 31: bcm_wait_queue_init()
[31186.715313] bcm63xx_phone bcm_drv_init() 1173: bcm_drv_init()
[31186.721310] bcm63xx_phone bcm_drv_init() 1178: Initializing RX wait queue
[31186.728353] bcm63xx_phone bcm_wait_queue_init() 31: bcm_wait_queue_init()
[31186.735391] bcm63xx_phone bcm_drv_init() 1180: Initializing TX wait queue
[31186.742478] bcm63xx_phone bcm_wait_queue_init() 31: bcm_wait_queue_init()
[31186.749523] bcm63xx_phone bcm_drv_init() 1195: Registering device bcm63xx_phone
[31186.757520] bcm63xx_phone pcm_init() 2531: pcm_init()
[31186.762802] bcm63xx_phone pcm_init() 2607: Using 15 buffers of 256 bytes for RX
[31186.770381] bcm63xx_phone pcm_init() 2649: Using 15 buffers of 256 bytes for TX
[31186.777955] bcm63xx_phone pcm_get_shared_res() 1986: pcm_get_shared_res()
[31186.785000] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm, addr=0xfffe1600, size=256)
[31186.794488] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm_dma, addr=0xfffe1800, size=256)
[31186.804356] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm_dmac, addr=0xfffe1900, size=32)
[31186.814208] bcm63xx_phone pcm_get_mem_res() 1942: pcm_get_mem_res(id=pcm_dmas, addr=0xfffe1a00, size=32)
[31186.912022] bcm63xx_phone pcm6358_pll_init() 1001: pcm6358_pll_init()
[31186.918720] bcm63xx_phone pcm6358_pll_init() 1006:  pll_ctrl1=f0010001, pll_ctrl2=53d47, pll_ctrl3=a85
[31186.960021] bcm63xx_phone pcm6358_pll_init() 1059:  pll_ctrl1=20001, pll_ctrl2=8492b, pll_ctrl3=1e1c
[31186.969506] bcm63xx_phone pcm_regs_init() 1279: pcm_regs_init()
[31186.975666] bcm63xx_phone pcm_regs_reset() 1223: pcm_regs_reset()
[31186.982002] bcm63xx_phone pcm_regs_reset() 1260: PCM_CTRL_REG <- 0x23200101
[31186.989223] bcm63xx_phone pcm_dma_init() 1538: pcm_dma_init()
[31186.995195] bcm63xx_phone pcm_dma_init() 1571: Allocating 120 bytes of DMA memory for RX descs
[31187.004245] bcm63xx_phone pcm_dma_init() 1582: 120 bytes of memory allocated, starting at address 0xa24ac000
[31187.014431] bcm63xx_phone pcm_dma_init() 1588: Allocating 120 bytes of DMA memory for TX descs
[31187.023421] bcm63xx_phone pcm_dma_init() 1599: 120 bytes of memory allocated, starting at address 0xa24ab000
[31187.033636] bcm63xx_phone pcm_dma_init() 1607: Allocating 7680 bytes of memory for RX and TX data buffers
[31187.043578] bcm63xx_phone pcm_dma_init() 1614: 7680 bytes of memory allocated, starting at address 0x82f28000
[31187.053872] bcm63xx_phone pcm_dma_init() 1640: Allocating 128 bytes of memory for DMA addresses
[... cut ...]
[31246.984501] [36;1mAPI_FUNC (line 0): VpSetLineTone() = SUCCESS[0m
[31246.990961] [36;1mAPI_FUNC (line 0): VpSetLineState():[0m
[31246.996696] [36;1mAPI_FUNC (line 0): Vp880SetLineState+[0m
[31247.002530] [36;1mAPI_FUNC (line 0): Vp880SetLineState-[0m
[31247.008353] [36;1mAPI_FUNC (line 0): VpSetLineState() = SUCCESS[0m
[31247.014902] [36;1mAPI_FUNC (line 0): VpVpFreeLineCtx():[0m
[31247.020726] [36;1mAPI_FUNC (line 0): VpVpFreeLineCtx() = SUCCESS[0m
[31247.027382] bcm63xx_phone zarlink_dev_free() 488: zarlink_dev_free()

pgid69, 2018-07-11:
Indeed, a lot of messages, but alas, i don't see anything wrong in the log :-(((
Sorry to have consumed so much of your time.

lulol, 2018-07-11:
There are some obvious differences that can be easily noticed with a bit of filtering of these dumps, but since don't really know much about the api can't assert how significant these could be.

The script below can filter these logs somehow, unfortunately it's not easy to restore back the timestamps but it's not much of a problem. The most useful results are in `filtered.log` and the previous parents `0dd0.log` and `1dd1.log`.

Anyway, many thanks for your effort and all the good work with this driver. The amount of code you developed for it is staggering.

Don't worry, if in the future you need to do some testing with this router I'll be glad to help in what I can, just ping me. Regards.

```
#!/bin/sh

# strip timestamp
sed -n "s#^[^ ]* \(.*\)#\1#p" 0.log > 0t.log
sed -n "s#^[^ ]* \(.*\)#\1#p" 1.log > 1t.log

# filter out common lines
diff -Naur 0t.log 1t.log > 01.log

# split logs
sed -n "s#^-\(.*\)#\1#p" 01.log > 0d.log
sed -n "s#^+\(.*\)#\1#p" 01.log > 1d.log

# set common token self line
sed "s#line 0#line self#" 0d.log > 0ts.log
sed "s#line 1#line self#" 1d.log > 1ts.log

# set common token other line
sed "s#line 1#line other#" 0ts.log > 0x.log
sed "s#line 0#line other#" 1ts.log > 1x.log

# filter out common lines
diff -Naur 0x.log 1x.log > 01.log

# split logs
sed -n "s#^-\(.*\)#\1#p" 01.log > 0dd.log
sed -n "s#^+\(.*\)#\1#p" 01.log > 1dd.log

# restore self line token to original
sed "s#line self#line 0#" 0dd.log > 0dd0.log
sed "s#line self#line 1#" 1dd.log > 1dd1.log

diff -Naur 0dd0.log 1dd1.log > filtered.log

```

pgid69, 2018-07-11:
I used your script but still don't see something wrong
Thank you