pgid69/bcm63xx-phone #28: Cannot compile on LEDE latest version

talnacialex, 2018-08-21:
Hi there. I am updating my LEDE firmware of the VoIP box(of course, it runs bcm63xx-phone), and it seems that the patch files aren't working anymore and also, it stopped compiling properly. It mentioned something about struct condition done in the mpi file.

How we can fix this?
BR,
Talnaci Alexandru

talnacialex, 2018-08-21:
Forgot to mention the OS I am using to compile on:

Linux Mint 19 Tara - it's Ubuntu-based OS, Ubuntu 18.04

talnacialex, 2018-08-22:
the VoIP box is HW556 ver C

pgid69, 2018-08-22:
Hello

I will put patches for 18.06 branch of OpenWrt as soon as possible.
But if you want to compile the firmware without the patch you just have to launch 
make kernel_menuconfig
and change state of "Device drivers/SPI Support/Broadcom 63xx SPI Controller" : new state should be M for module or N if you don't need the module

BR

pgid69, 2018-08-23:
Hello

I've just added the patches for 18.06 branch of OpenWrt.
Tell me if it's ok or not.

Thank you

BR

talnacialex, 2018-08-23:
I will compile right now and see if the problem persists.

Thank you!
Best Regards,

Talnaci Alexandru

talnacialex, 2018-08-23:
Ok, so the patches work, but now it remains that error I talked about earlier. struct completion done;

Log: https://pastebin.com/cEceyLk0

talnacialex, 2018-08-23:
Ok! I will try to compile the new version and tell you if it succeded or not.

BR.

talnacialex, 2018-08-23:
Yes, it compiled successfully now. Thank you for making this amaizing driver. I don't know why LEDE/OpenWRT team did not add support for this driver. 

BR, 
Talnaci Alexandru

closed 2018-08-23 by talnacialex

pgid69, 2018-08-24:
Hello

Just a small question related to issue #26 submitted by user @lulol : does both FXS ports works or just only one ?

Thanks

Gilles