pgid69/bcm63xx-phone #29: bcm63xx-phone crashes LEDE 18

talnacialex, 2018-08-24:
Hi there. After successfully compilation of this driver, I started the router, flashed the firmware and everything was ok. Then, I ssh-d into it, ran insmod bcm63xx-phone and once I ran that command, the router froze(so it crashed) and it rebooted.

Router is Huawei EchoLife HG556a Version C.

talnacialex, 2018-08-24:
I managed to get the kernel log and will add it ASAP.

talnacialex, 2018-08-24:
https://pastebin.com/wv8qge2j

pgid69, 2018-08-24:
Thank you for the log.
Can you try to add the three following line in file mpi/mpi.c after line 581
```
#if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 13, 0)
   .prevent_deferred_probe = true;
#endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(3, 13, 0) */
```
Recompile bcm63xx-phone.ko and test to see if it works.
But i'm doubtful.

Else you can also try to use the kernel spi driver : 
* replace the line `#undef BCMPH_USE_SPI_DRIVER` by `#define BCMPH_USE_SPI_DRIVER` in file inc/config.h
* and uncomment the line `#  FILES+=$(LINUX_DIR)/drivers/spi/spi-bcm63xx.$(LINUX_KMOD_SUFFIX) in Makefile

If you compile the OpenWrt firmware without appling the patch 001-add-fn-to-spi-bcm63xx-for-quick-transfer.patch, you have to apply it, clean the kernel to force it's recompilation (`make target/linux/clean`) and recompile the firmware.
Else you can simply recompile bcm63xx-phone.ko

Then insmod spi-bcm63xx.ko and insmod bcm63xx-phone.ko

BR

pgid69, 2018-08-24:
Just a question : as `insmod bcm63xx-phone.ko` makes the router crash freezes, how did you manage to get a log ?

talnacialex, 2018-08-24:
I managed to compile the kernel with Crash Logs enabled, and I found it in /sys/kernel/debug and there is a recent log

talnacialex, 2018-08-24:
btw... with .prevent_deferred_probe = true; it gave compilation error, I had to set it to .prevent_deferred_probe = true
Compiled successfully, will test it tomorrow, as right now it's night.

talnacialex, 2018-08-25:
Tested, same errors. Will try to use SPI.

talnacialex, 2018-08-25:
SPI works. Closing. Thank you!

closed 2018-08-25 by talnacialex

p-velasco, 2020-05-29:
Ok, so the kernel changed to yet another way of managing devices.

Solved with this quick, dirty and ugly patch:

```
diff --git a/bcm63xx-phone/src/mpi/mpi.c b/bcm63xx-phone/src/mpi/mpi.c
index fda9a47..e665de2 100644
--- a/bcm63xx-phone/src/mpi/mpi.c
+++ b/bcm63xx-phone/src/mpi/mpi.c
@@ -400,6 +400,13 @@ static const struct platform_device_id bcm63xx_spi_dev_match[] = {
 };
 #endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(4, 4, 0) */
 
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0)
+static const struct of_device_id bcm63xx_spi_of_match[] = {
+       { .compatible = "brcm,bcm6358-spi", .data = &bcm6358_spi_reg_offsets },
+       { },
+};
+#endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0) */
+
 static int bcm63xx_spi_probe(struct platform_device *pdev)
 {
    struct device *dev = &(pdev->dev);
@@ -411,6 +418,9 @@ static int bcm63xx_spi_probe(struct platform_device *pdev)
    bcm_pr_debug("%s()\n", __func__);
 
    if (
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0)
+       (!dev->of_node) &&
+#endif /* LINUX_VERSION_CODE < KERNEL_VERSION(4, 14, 0) */
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 4, 0)
        (!pdev->id_entry->driver_data)
 #else /* LINUX_VERSION_CODE < KERNEL_VERSION(4, 4, 0) */
@@ -421,6 +431,15 @@ static int bcm63xx_spi_probe(struct platform_device *pdev)
    }
 
    bs = &(bcm_mpi_dev_data);
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0)
+   if (dev->of_node) {
+      const struct of_device_id *match;
+      match = of_match_node(bcm63xx_spi_of_match, dev->of_node);
+      if (!match)
+         return -EINVAL;
+      bcm_mpi_dev_data_init(bs, match->data);
+   } else
+#endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0) */
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 4, 0)
    bcm_mpi_dev_data_init(bs, (const unsigned long *)pdev->id_entry->driver_data);
 #else /* LINUX_VERSION_CODE < KERNEL_VERSION(4, 4, 0) */
@@ -573,12 +592,18 @@ static struct platform_driver bcm63xx_spi_driver = {
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 2, 0)
       .probe_type = PROBE_FORCE_SYNCHRONOUS,
 #endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(4, 2, 0) */
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0)
+      .of_match_table = bcm63xx_spi_of_match,
+#endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0) */
    },
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 4, 0)
    .id_table   = bcm63xx_spi_dev_match,
 #endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(4, 4, 0) */
    .probe      = bcm63xx_spi_probe,
    .remove     = bcm63xx_spi_remove,
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 13, 0)
+   .prevent_deferred_probe = true,
+#endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(3, 13, 0) */
 };
 #endif // BCMPH_USE_SPI_DRIVER
 #endif // !BCMPH_NOHW

```

sturia, 2020-06-16:
Thanks a lot for the patch

p-velasco, 2020-06-16:
> Thanks a lot for the patch, working on Pirelli A226M, Openwrt 19.07.3, kernel 4.14.180

Maybe you also need my ast-chan patch. If you manage to set up the DAHDI driver, please let me know (Asterisk segfaults with it for me, and I don't have time just now to attach GDB)

sturia, 2020-06-17:
well I actually talked too fast. Driver loads properly, but when I goto to test it with asterisk I realized I have issues with asterisk on 19.07.3 - it basically doesn't load any modules. 
Which asterisk version are you using? I tweaked asterisk 13 source feed (Makefile etc.) from Chaos Chalmer as on 19.07.3 the only asterisk version available is 16, for which there is no 
bcm63xx-phone official channel. Is this the patch you're referring to?

thanks

p-velasco, 2020-06-17:
Asterisk 16, and yes, I was talking about that patch for ast-chan to make it work with it. Check my fork on the branch for 19.07.3 or the PR #35

sturia, 2020-06-18:
Excellent, it compiled and it eventually works on asterisk16. The only issue left is when asterisk loads the bcm63xx_chan module: i have to force it as "preload" in modules.conf, otherwise it messes up with something and crashes asterisk. With preload it works fine, but there is a side effect of bcm63xx_chan NOT loading properly country tone rules from bcm63xx_phone.conf - it defaults to US tones.
ever experienced anything similar?

p-velasco, 2020-06-18:
Didn't test tones, but I don't remember any problems loading the module. I just tested it with chan-dongle and was properly routing calls, but didn't have the time to complete my setup. Will try in the next days if I get some spare time

sturia, 2020-06-18:
I performed few other tests - now everything works fine. country tones are recognized and bcm63xx_chan module doesn't require to be preloaded in asterisk to work. Unfortunately I didn't perform tests systematically, what I did is to go back to a very standard compile conditions and removing all non strictly necessary modules from asterisk, which in my case are:

asterisk16 - 16.3.0-7
asterisk16-app-confbridge - 16.3.0-7
asterisk16-app-exec - 16.3.0-7
asterisk16-app-system - 16.3.0-7
asterisk16-bridge-builtin-features - 16.3.0-7
asterisk16-bridge-simple - 16.3.0-7
asterisk16-bridge-softmix - 16.3.0-7
asterisk16-cdr - 16.3.0-7
asterisk16-chan-bcm63xx-phone - 4.14.180-4.14.180
asterisk16-chan-rtp - 16.3.0-7
asterisk16-chan-sip - 16.3.0-7
asterisk16-codec-a-mu - 16.3.0-7
asterisk16-codec-alaw - 16.3.0-7
asterisk16-codec-ulaw - 16.3.0-7
asterisk16-format-gsm - 16.3.0-7
asterisk16-format-pcm - 16.3.0-7
asterisk16-format-sln - 16.3.0-7
asterisk16-format-wav - 16.3.0-7
asterisk16-func-shell - 16.3.0-7
asterisk16-res-adsi - 16.3.0-7
asterisk16-res-musiconhold - 16.3.0-7
asterisk16-res-pjproject - 16.3.0-7
asterisk16-res-rtp-asterisk - 16.3.0-7
asterisk16-res-rtp-multicast - 16.3.0-7
asterisk16-res-smdi - 16.3.0-7
asterisk16-res-sorcery - 16.3.0-7
asterisk16-voicemail - 16.3.0-7