pgid69/bcm63xx-phone #31: fatal error: ../drivers/spi/spi-bcm63xx.h: No such file or directory. While compiling

realpg, 2018-09-21:
Default compiling works fine, but crash while insmod
`[ 2217.812987] bcm63xx_phone: bcm63xx_phone: v0.2
[ 2217.817637] bcm63xx_phone bcm63xx_phone_init() 1021: bcm63xx_phone_init()
[ 2217.824747] bcm63xx_phone board_init() 719: board_init()
[ 2217.830250] bcm63xx_phone: CPU is Broadcom 6358
[ 2217.834979] bcm63xx_phone: Board is HW556
[ 2217.839158] bcm63xx_phone board_get_desc() 778: board_get_desc()
[ 2217.848527] bcm63xx_phone bcm_periodic_timer_init() 249: bcm_periodic_timer_init()
[ 2217.856431] bcm63xx_phone bcm_timer_init() 191: bcm_timer_init()
[ 2217.862682] bcm63xx_phone bcm_period_init() 28: bcm_period_init()
[ 2217.869075] bcm63xx_phone bcm63xx_phone_init() 1081: t->mm_desc.ioctl_param_off = 0, t->mm_desc.ioctl_param_size = 220
[ 2217.880201] bcm63xx_phone bcm63xx_phone_init() 1090: t->mm_desc.rbs[0].rx_ring_buf_desc_off = 224, t->mm_desc.rbs[0].rx_ring_buf_desc_size = 12
[ 2217.893550] bcm63xx_phone bcm63xx_phone_init() 1098: t->mm_desc.rbs[0].tx_ring_buf_desc_off = 240, t->mm_desc.rbs[0].tx_ring_buf_desc_size = 12
[ 2217.906909] bcm63xx_phone bcm63xx_phone_init() 1090: t->mm_desc.rbs[1].rx_ring_buf_desc_off = 256, t->mm_desc.rbs[1].rx_ring_buf_desc_size = 12
[ 2217.920512] bcm63xx_phone bcm63xx_phone_init() 1098: t->mm_desc.rbs[1].tx_ring_buf_desc_off = 272, t->mm_desc.rbs[1].tx_ring_buf_desc_size = 12
[ 2217.933949] bcm63xx_phone bcm63xx_phone_init() 1128: 8192 bytes of memory allocated, starting at address 0x82fa0000
[ 2217.944769] bcm63xx_phone bcm_phone_mgr_init() 1048: bcm_phone_mgr_init()
[ 2217.951806] bcm63xx_phone bcm_periodic_work_init() 92: bcm_periodic_work_init()
[ 2217.959376] bcm63xx_phone bcm_period_init() 28: bcm_period_init()
[ 2217.965694] bcm63xx_phone bcm_phone_mgr_devs_init() 907: bcm_phone_mgr_devs_init()
[ 2217.973557] bcm63xx_phone phone_device_alloc() 53: phone_device_alloc()
[ 2217.980450] bcm63xx_phone phone_device_alloc() 82: 2032 bytes of memory allocated, starting at address 0x82eca800
[ 2217.991073] bcm63xx_phone phone_dev_le88266_init() 56: phone_dev_le88266_init()
[ 2217.998651] bcm63xx_phone phone_dev_zarlink_ve880_init() 2021: phone_dev_zarlink_ve880_init()
[ 2218.007483] bcm63xx_phone bcm_mpi_init() 875: bcm_mpi_init()
[ 2218.013363] bcm63xx_phone: Error: Driver 'bcm63xx-spi' is already registered, aborting...
[ 2218.021847] bcm63xx_phone bcm_periodic_work_deinit() 103: bcm_periodic_work_deinit()
[ 2218.029865] bcm63xx_phone bcm_period_deinit() 33: bcm_period_deinit()
[ 2218.036546] bcm63xx_phone bcm63xx_phone_init() 1207: Freeing mmap buffer used for RX, TX...
[ 2218.045204] bcm63xx_phone bcm_periodic_timer_deinit() 265: bcm_periodic_timer_deinit()
[ 2218.053404] bcm63xx_phone bcm_timer_stop() 167: bcm_timer_stop()
[ 2218.059652] bcm63xx_phone board_deinit() 785: board_deinit()
`

After searching history issues, change it to SPI mode, but compile failed :

`In file included from /root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/arch/bcm63xx/vp_api_types.h:12:0,
                 from /root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/includes/vp_api.h:43,
                 from /root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/board/board.c:31:
/root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/inc/mpi.h:18:13: fatal error: ../drivers/spi/spi-bcm63xx.h: No such file or directory
 #   include <../drivers/spi/spi-bcm63xx.h>
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [/root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/board/board.o] Error 1
make[4]: *** [_module_/root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone] Error 2
make[4]: Leaving directory `/root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.14.70'
make[3]: *** [/root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/.built] Error 2
make[3]: Leaving directory `/root/openwrt-hg556/feeds/chan/bcm63xx-phone/bcm63xx-phone'
time: package/feeds/chan/bcm63xx-phone/compile#1.00#0.36#1.38
make[2]: *** [package/feeds/chan/bcm63xx-phone/compile] Error 2
make[2]: Leaving directory `/root/openwrt-hg556'
make[1]: *** [/root/openwrt-hg556/staging_dir/target-mips_mips32_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/root/openwrt-hg556'
make: *** [world] Error 2
`

realpg, 2018-09-21:
After searching history issues, change it to SPI mode, but compile failed :

`
In file included from /root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/arch/bcm63xx/vp_api_types.h:12:0,
                 from /root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/extern/zarlink/2.24.0.lite/api_lib/includes/vp_api.h:43,
                 from /root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/board/board.c:31:
/root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/inc/mpi.h:18:13: fatal error: ../drivers/spi/spi-bcm63xx.h: No such file or directory
 \#   include <../drivers/spi/spi-bcm63xx.h>
             \^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [/root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone/board/board.o] Error 1
make[4]: *** [_module_/root/openwrt-hg556/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/bcm63xx-phone] Error 2
`

realpg, 2018-09-21:
My computer: Ubuntu 14.04.5 LTS with kernel 4.5.12

Kernel version: 4.14.70

pgid69, 2018-09-21:
From the log it seems that you don't apply none of the patch.
Read the README.

realpg, 2018-09-21:
OK. I will read the README again, carefully.  Report later.

Thanks for this AMAZING work!

realpg, 2018-09-21:
After reading README again, I patched the 000 & 001 patch (version 18.06). And removed the build directory of the Linux kernel as mentioned.

And I recompiled, the problem still exists
https://gist.github.com/realpg/9fe632c9f5a83f1b9a48b9b976780354

After read the content of the patch, I find that the patch is based on linux kernel 4.9, but the newest openwrt has a kernel 4.14

Modify the patch, and compile successfully. Flash the firmware tomorrow and report.

pgid69, 2018-09-21:
Hi 

What version of OpenWrt are you trying to compile ?
18.06 for architecture bcm63xx uses kernel version 4.9.
If it's trunk i don't know which kernel version is used, but if it's 4.14 there's no patch available and i don't intend to make them : i only make patches to stable release because trying to follow the trunk is too time consuming.
If you still want to use trunk : 
- the first patch can be avoided if you run make kernel_menuconfig and change the state to "Broadcom BCM63XX SPI controller" to M or deselect it (search for symbol SPI_BCM63XX to know where the item is located in the menus)
- for the second patch you can try to copy the file target/linux/brcm63xx/patches-4.9/900-add_fn_to_spi_bcm63xx_for_quick_transfer.patch to directory target/linux/brcm63xx/patches-4.14 but i'm not sure the patch will apply to kernel 4.14

Gilles

realpg, 2018-09-21:
@pgid69 openwrt trunk version

Feedback: The patch works well with kernel 4.14, just change the version folder name / config filename.
I've checked both version, file contents referenced are the same.

realpg, 2018-09-21:
Maybe we can make a new folder [trunk-kernel_4.14] in patches folder.

closed 2018-09-21 by realpg