pgid69/bcm63xx-phone #32: [Documentation] Help to add support for SiLabs chips

hitech95, 2019-03-18:
Hi,
I would create a driver for the SiLabs chips.
I have read a bit of this driver source but I cannot fully understand the flow path.

Can you provide me a starting point?

pgid69, 2019-03-23:
Hello

Basic architecture : three components
- PCM bus manangement (file pcm/pcm.c)
- Phone management : look for change in status line (phone/phone_mgr.c)
- Supervisor : handle ioctl and data transfer in and out of the driver (common.c, main.c, main_driver.c)

The ioctl are handled in function bcm_drv_unlocked_ioctl (main_driver.c).
The most important ioctl is BCMPH_IOCTL_START : it's used to configure and start the supervision of the phone lines.
- it configures PCM and starts it to enable the PCM clock. The PCM clock must be enabled to be able to configure the phone lines
- and it configure the phone lines

At the end of the configuration of the phone lines in function bcm_phone_mgr_start() (phone/phone_mgr.c), a timer is started that run every 10ms : it polls phone line status, fills and empties DMA buffers used by PCM... (function bcm_phone_mgr_timer_work_fn() in file phone/phone_mgr.c))

To add a new board :
- In board.h add a new value in enum phone_dev_type_t
- In board.h extend union parameters of structs phone_desc_line_t and phone_desc_device_t to add data you need to handle the silab chip.
- In board.c add a new description of your router
- create a new device for silabs that must extends phone_device_t in phone.h (like struct phone_dev_zarlink_t in file zarlink_common.h)
- allocate it in the function phone_device_alloc() (file phone.c)
Of course your device should have a specific instance of struct vtbl_phone_device with functions that handle the silabs chip : start(), stop()...

I hope my answer will help you.

If you have other questions i will try to answer more quickly.

hitech95, 2019-03-24:
Hi,
Thanks for the infos! I'm making some progress!
So you're using ioctl to communicate between userspace to kernelspace.

Right now I'm focusing on the PCM bus!
I don't fully know how the microsemi chips works but are they: PCM + SPI for control right?

Does you know in what mode the PCM bus is configured?
The PCM bus is using DMA to take and put data right?
AFIK the init of the PCM bus is done outside the pcm.c file (register assertion are outside that file)

What is the: 
BCMPH_NOHW ?

pgid69, 2019-03-24:
> Hi,
Hi

> Thanks for the infos! I'm making some progress!
> So you're using ioctl to communicate between userspace to kernelspace.
> 
Yes ioctl and a memory mapped region for voice samples

> Right now I'm focusing on the PCM bus!
> I don't fully know how the microsemi chips works but are they: PCM + SPI for control right?
> 
Configuration and status query of the Microsemi chip is done via the SPI bus, and only the voice samples are sent or received with the PCM bus.

> Does you know in what mode the PCM bus is configured?
I don't understand what you mean by PCM mode.

> The PCM bus is using DMA to take and put data right?
Yes it is

> AFIK the init of the PCM bus is done outside the pcm.c file (register assertion are outside that file)
PCM initialization is done in pcm.c : in function pcm_init() and in function pcm_start()

> What is the:
> BCMPH_NOHW ?
It is just a flag I used for debugging the driver without running it on the router. Of course PCM is simulated as just a loop buffer where data read was just data written. And there is no SPI communication. It allows me to debug initialization, deinitialzation and handling of buffers.

What is your device exactly ?

hitech95, 2019-03-24:
> What is your device exactly ?

AGPF with SI3215 chips. As you see it's to Microsemi based so I have to edit a lot of stuff!

hitech95, 2019-03-27:
Could you explain to me what are thoose two values:
BCMPH_PCM_CHANNEL_WIDTH
BCMPH_PCM_MAX_FRAME_SIZE

To me they have no sense.

pgid69, 2019-03-28:
I developped the driver without any documentation, only with an include file that i found in an open source driver, that contains constants for accessing the register of the PCM controller [https://github.com/Noltari/cfe_bcm63xx/blob/master/shared/broadcom/include/bcm963xx/6358_map.h]().
The constants are in file inc/bcm63xx.h.
You can read the comment i added for register PCM_CHAN_CTRL_REG : it explains the values of BCMPH_PCM_CHANNEL_WIDTH and BCMPH_PCM_MAX_FRAME_SIZE.
If you find or have more informations please share them with me.

hitech95, 2019-03-30:
I also don't have any Source available. I'm using your work.

The point is the fact that the channel width in the PCM should be always 16bit or 8 bit.
I think that it is the DMA channel width but i'm not sure.

pgid69, 2019-05-04:
Hello

I'd like to know if your project has made some progress.
I'm going to acquire a router with a SI3216 chips and i will try to add support in my driver.
Of course it will take time to do this.
Can you tell me what is the documentation or source files you are using to add support to the SI3215, because from the documentation i read, both chips are very similar.

Thank you

hitech95, 2019-05-04:
Hi, I had low progress due to my low available time.
If you want we could collab to make the driver fully working.

My end goad is to use DTS to define the configs for bloth the slic and the PCM interfaces.
(So we don't have to define each time the device in the driver)